realsenseai/librealsense #6714: Ubuntu install location

GOBish, 2020-06-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Ubuntu   | 
| Kernel Version (Linux Only)    |  ()                                         | 
| Platform                                 | Jetson  |
| SDK Version                            |  { l }                          | 
| Language                            |  {python }                          | 
| Segment			|  {}                          | 

### Issue Description
I followed the instructions to install realsense on ubuntu:
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

I am using a Nvidia xavier nx, with jetpack 4.4.

The installation went fine (except sudo apt-get install librealsense2-dkms was not found).
I can run realsense-viewer and it works well with the D435.

I cannot find the install folder, though - I would like to use the examples and python implementations.
I expected it to be in /usr/local/lib but nothing is there.

this issue: https://github.com/IntelRealSense/librealsense/issues/6449
shows that the install is working, but I cannot find the files. I need to point my PYTHONPATH to them.

Thanks for the help!

ev-mp, 2020-06-28:
@GOBish, check the installation path of the debian packages can done with
`dpkg -L librealsense2`
`dpkg -L librealsense2-dev`
`dpkg -L librealsense2-utils`
I believe you'll find it under `/usr/bin` 
Duplicate of #5474

As for dkms - is [maintained for Ubuntu LTS](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#using-pre-build-packages) kernels, and therefore does not apply to Jetsons'.

ev-mp, 2020-07-02:
@GOBish, was the info useful ?

GOBish, 2020-07-02:
Hi @ev-mp , with the help of @MartyG-RealSense on this issue https://github.com/IntelRealSense/librealsense/issues/6449
I was able to solve it. I cloned all the files again to a specific folder and built it from source using cmake flags we discuss in that issue.
Thanks for your help.

closed 2020-07-02 by GOBish

MartyG-RealSense, 2020-07-02:
@GOBish  Thanks so much for the update!

AswinkarthikeyenAK, 2022-06-12:
Hi,

Have realsense-viewer version 2.48.0 installed. However, I cannot find the location of installation. I would like to upgrade from v2.48.0 to v2.50.0. 

Can someone help please?

`dpkg -L librealsense2-dev
dpkg-query: package 'librealsense2-dev' is not installed
Use dpkg --info (= dpkg-deb --info) to examine archive files,
and dpkg --contents (= dpkg-deb --contents) to list their contents.
user@USER:~$ dpkg -L librealsense2-utils
dpkg-query: package 'librealsense2-utils' is not installed
Use dpkg --info (= dpkg-deb --info) to examine archive files,
and dpkg --contents (= dpkg-deb --contents) to list their contents.
`

MartyG-RealSense, 2022-06-12:
Hi @AswinkarthikeyenAK  It sounds as though you want to **upgrade** your existing RealSense SDK packages.  Instructions for doing so can be found at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#upgrading-the-packages

The package containing the SDK examples and tools, including realsense-viewer, is **librealsense2-utils**

AswinkarthikeyenAK, 2022-06-12:
Hi @MartyG-RealSense , 
Thanks, that is correct, and I tried the setups from the link. However, there were few issues I was running into and therefore, I found the location of realsense viewer and manged to remove it. After, I was able to install the SDK 2.50.0.

However, if you can help with how to restore the d455 back to factory calibration settings in ubuntu, or provide me the factroy calibration files it will be really great? I have been trying to find this and its hard to find a way to do this. 

Thanks

MartyG-RealSense, 2022-06-13:
A simple way to reset the camera to factory-new calibration is in realsense-viewer using the procedure described at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487