realsenseai/librealsense #6716: L515 LiDAR Camera is not activate on android application.

dorazarashi, 2020-06-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   |01.04.01.00 | 
| Operating System & Version | android  | 
| android Version   |  7.0 | 
| Platform                                 |   android zenfoneAR |
| SDK Version                            |  2.35.2  | 

I tried to use following android application on ASUS Zenfone (android version is 7.0).

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/capture

However, above application with L515 is not executed.
(this application is compiled with librealsense SDK v2.35.2)

No error message showed.
Conneted with L515, the word "Connect a Realsense camera" is disappered and black screen keeped.
Next, it worked when change device to D435(depth and color image displayed.)
So Only the case that connected L515 is not work.

How should I solve this problem?
(In intel realsense viewer v2.35.2, L515 LiDAR camera is now well working.)

dorodnic, 2020-06-29:
Hi @dorazarashi 
At the moment, L515 does not work via USB2. Perhaps phone connector / cable is causing the camera to enumerate as USB2?

dorazarashi, 2020-06-30:
Hi @dorodnic 
Thank you for reply.  Connector and cable type is USB3 .
(I used some USB3 cable that includes attached cable of L515.)
I will check if it is not recognized as USB2 on the application.

dorazarashi, 2020-07-02:
I checked following code on android application , and usb_type was 2.1.
`String usb_type = dl.createDevice(0).getInfo(CameraInfo.USB_TYPE_DESCRIPTOR);`
Cable and usb type-c conversion connector is USB3.0 , but ZenfoneAR's connector was found to be USB2.0.

I will run application on the other android phone(USB3.0). Thank you for advice.

closed 2020-07-02 by dorazarashi