realsenseai/librealsense #6717: Adding opencv inside librealsense for using it's function in andorid

ankittecholution, 2020-06-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 | 
| Operating System & Version |   Android  | 
| Platform                                 | Android |
| SDK Version                            |  2.19 }                          | 
| Language                            |  Android java wrraper                           | 

### Issue Description
I want use some of the opencv function directly imbedded in librealsense so that i can use and leverage some of the opencv function please guide me how to do that it should be for android opencv.

MartyG-RealSense, 2020-06-29:
Hi @ankittecholution  The only reference that I could find about using OpenCV with the RealSense Android wrapper is one about converting a RealSense frame to OpenCV **Mat**.

https://github.com/IntelRealSense/librealsense/issues/5185

dorodnic, 2020-06-29:
Hi @ankittecholution 
Please take a look at https://opencv.org/android/

kafan1986, 2020-07-01:
@ankittecholution You can integrate opencv as per the official documentation as suggested by @dorodnic 

I would suggest for availability of all the functionality of librealsense SDK and OpenCV, it is better to call it natively via NDK.

ankittecholution, 2020-07-05:
Thanks, I will try to implement it.

dorodnic, 2020-07-06:
Let us know if you run into issues

closed 2020-07-06 by dorodnic