realsenseai/librealsense #6719: “Set ROI” button (for depth image) not available for D435i on RealSense Viewer?

lipeize-pku, 2020-06-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |   Ubuntu 19.04  | 
| Platform                                 | PC  |
| SDK Version                            |  2.35.2                          | 

### Issue Description
Might be a stupid question.
But I can't find the "Set ROI" button beside "Enable Auto Exposure" button of Stereo Module described in [documentation](https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance) anywhere.

Do I need to do something in order for the button to appear?

"Enable Auto Exposure" is on.

MartyG-RealSense, 2020-06-29:
@dorodnic   I can confirm that I do not have the Set ROI button on my 2.35.2 Viewer either with D435i.

ev-mp, 2020-06-29:
@lipeize-pku  hi,
This issue was addressed in https://github.com/IntelRealSense/librealsense/pull/6621, scheduled for the next version.
You can pull the fix from the `development` branch if needed.

dorodnic, 2020-07-19:
Should be resolved in 2.36.0

closed 2020-07-21 by dorodnic