realsenseai/librealsense #6724: How to increase the precision of T265 and D435i on objects near to the groud

juvinski, 2020-06-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 and D435i} | 
| Firmware Version                   | (lastest) - 0.2.0.951 | 
| Operating System & Version |   Linux Ubuntu 18.4.3  | 
| Kernel Version (Linux Only)    |   4.9.140                                         | 
| Platform                                 |  Raspberry Pi and NVIDIA Jetson |
| SDK Version                            |  2.35.2                           | 
| Language                            |  Python3                          | 
| Segment			|  Robot | 

Hi guys,

I'm working on a rover and the precision for t265 and d435, and I'm not satisfied with the performance of the cameras near to the ground. For instance. I have a neural network to detect objects and using the depth stream to analyze the distance between the camer and the object. I'm aligning the streams and then consult the distance the value is really bad - objects with around 2 metters are returned the distacen around 30, 25, 40 centimeters. If we look an the image we see the camera is confusing the object with the ground. I tried this with this little ball on the example as I tried with a beer can, a cup, a glass:
[image]

If I tried to tracking the position of the robot with the t265 I'm facing a similar problem, when the rover start to move, I can see a variance on the position, on a straight line I can see a long variance . This image sample, the robot was on a straight line but on the tracking there are a jump first and then a perpendicular line instead a straight line.

[image]

Could you please help me how to increase the precision/accuracy of the cameras?

thanks a lot.

RealSenseSupport, 2020-10-26:
Hi,

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-10-30 by RealSenseSupport

juvinski, 2020-10-30:
Hi, yes I still need help