realsenseai/librealsense #6725: make frames from image file

jongsun-nuvi, 2020-06-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.align.html#pyrealsense2.align
I want align color image and depth image
I already capture image and saved as numpyarray
but I noticed that those images are not fit, so I want to align those.
how I can do that?
I already read pyrealsense2.align document

MartyG-RealSense, 2020-06-30:
Hi @jongsun-nuvi   There are only a few pyrealsense2 cases I could find of data being saved into a bumpy array and then depth aligned to color.  The most useful of these cases is in the link below.

https://github.com/IntelRealSense/librealsense/issues/6114

The RealSense user in that case had problems.  Their script only worked if they did not do alignment, or if they aligned **color to depth** instead of depth to color.  I hope that it will provide you with some useful guidance though.

dorodnic, 2020-07-09:
I think this was discussed here many times. Frame != Image. Frame = Image+Calibration. If in addition to numpy array you saved depth and color intrinsics and extrinsic calibration between specific resolutions pair, there is a way (somewhat complicated) to perform alignment. Otherwise there is not enough data to do it.

closed 2020-07-26 by MartyG-RealSense