realsenseai/librealsense #6726: Need 1 channel grayscale image for finding contours in openCV

dcprime, 2020-06-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415                                                       |
| Firmware Version                   | 05.12.03.00                                             | 
| Operating System & Version |  Win (8.1/10)                                          | 
| Kernel Version (Linux Only)    |  N/A                                                       | 
| Platform                                 | PC                                                          |
| SDK Version                            |  2.0                                                        | 
| Language                               |  C++                                                      | 
| Segment			        |  Hardware/Software development        | 

### Issue Description
I'm working on a pipeline where I capture a depth frame of a gesture being performed, convert it to grayscale using "White to Black" in the colorizer, do some post-processing (decimation), and then use OpenCV to clean up image noise and isolate the gesture from the depth image. 

In order to isolate the gestures, I'm attempting to use OpenCV's findContours() function, but it can only operate on a 1 channel image (CV_8UC1). However, the only way to make my colorized image appear properly when I output the OpenCV matrix to the screen is to convert it using the CV_8UC3 type (3 channels). 

My digging so far has shown that OpenCV cannot convert a 3-channel image into a 1-channel image, so I'm wondering if I can get the colorizer output to be single-channel grayscale before converting it to an OpenCV Mat object using CV_8UC1 instead of CV_8UC3. 

Is that possible? Or is there a better way to do this?

MartyG-RealSense, 2020-06-30:
Hi @dcprime   The link below has C++ scripting for exporting a greyscale depth image that uses 'white to black'.

https://github.com/IntelRealSense/librealsense/issues/1831

dcprime, 2020-06-30:
Hi Marty,

I've actually already incorporated much of the code in the issue you linked to into my own codebase. My problem is that I need to use the following to get my image to appear properly when I display it:
`Mat image(Size(w, h), CV_8UC3, (void*)depth.get_data(), Mat::AUTO_STEP);`

...which gives me a 3-channel Mat object (instead of 1 channel)

When I try 
`Mat image(Size(w, h), CV_16UC1, (void*)depth.get_data(), Mat::AUTO_STEP);`

...I get an output which is almost entirely black, which won't work either. If there is some other intermediate conversion step that I'm missing in order to get a "white to black" output with 1 channel, that would be great.

MartyG-RealSense, 2020-06-30:
This information is quite old but as it relates to OpenCV commands rather than Librealsense, it may still work.  Someone else who wanted to convert cv:16uc1 to cv:8uc1 was recommended to use a command called **Mat::convertTo**

https://stackoverflow.com/questions/19042024/how-to-convert-cvmat-type-from-cv-16uc1-to-cv-8uc1

dcprime, 2020-06-30:
So I tried converting my colorized output to cv:16u1, and my output image came out like this:
[image]

When I then tried converting this to cv:8uc1, this was the result:
[image]

Here is the code I was using:

`Mat image(Size(w, h), CV_16UC1, (void*)depth.get_data(), Mat::AUTO_STEP);`
`image.convertTo(image, CV_8UC1);`

There must still be something missing in the conversion process here...

MartyG-RealSense, 2020-06-30:
Maybe you could try adding a scale factor like in the link:

image.convertTo(image, CV_8UC1, scalefactor);

Maybe try '1' as the scalefactor and see what happens.

dcprime, 2020-06-30:
Altered my code to the following:
`Mat image(Size(w, h), CV_16UC1, (void*)depth.get_data(), Mat::AUTO_STEP);`
`Mat result(Size(w, h), CV_8UC1);`
`image.convertTo(result, CV_8UC1, 1);`

Same results as in my previous message. The default value for scale factor is actually 1 already. Tried different values for scale factor (from 0.5 to 10.0) and they had no effect on the image output.

dcprime, 2020-06-30:
Is there a way to set the colorizer to output a 1 channel instead of 3 channel image?

MartyG-RealSense, 2020-06-30:
Any difference if you use **1/255.0** as the scalefactor?

image.convertTo(result, CV_8UC1, 1/255.0);

dcprime, 2020-06-30:
I've made a bit of progress actually, and experimenting with that scalefactor did help.

I turned off the colorizer altogether, grabbed the depth data, and ran it through the conversion code like this:
`Mat image(Size(w, h), CV_16UC1, (void*)depth.get_data(), Mat::AUTO_STEP);`

Then performed the conversion with a 1/10.0 scale factor:
`image.convertTo(image, CV_8UC1, 1/10.0);`

The result was this:
[image]

...which is a step forward. The double image is gone, and I have some differentiation in the depth data, but the remaining issues now are:

1. The image has a LOT of noise, especially in the distance. This noise does not appear when using the colorizer, and I'm not sure what it does to clean this up.
2. I need to have close data colorized as white and far data colorized as black. I think some sort of inversion operation could be done for this, but the noise in point 1 above is a much bigger concern.
3. The depth range is less than I get when using the colorizer (the range of greys in the image should be wider, ideally). Perhaps this can be adjusted in the sensor settings?

Also, if I run this code with the scalefactor as 1, I get this:
[image]

With scalefactor 1/255.0 I get this:
[image]

...which is the same as I get when attempting to view the image as CV_16UC1 without conversion.

MartyG-RealSense, 2020-06-30:
Congratulations on the progress!  OpenCV is not one of my specializations, so I appreciate your patience.   :)

Is the image still black if 1/32 is used instead of 1/255?

dcprime, 2020-06-30:
Not black, but too dark. It gets darker as the denominator increases.

MartyG-RealSense, 2020-06-30:
Ah yes, I see that your best result so far was with 1/10 scale.

Some OpenCV users have had success dealing with noise by applying a **median filter**.

https://docs.opencv.org/2.4/doc/tutorials/imgproc/gausian_median_blur_bilateral_filter/gausian_median_blur_bilateral_filter.html#median-filter

dcprime, 2020-06-30:
So, I've made more progress by doing some mathematical transformations on my Mat object. 

First, I applied the 1/10.0 scaling mentioned earlier in our conversation, resulting in this type of image:
[image]

You can see above that my hand (which is the closest object to the camera) is darker than the other objects further away in the image, but the areas of occlusion were even darker, giving the false reading that they were closer objects. This is not acceptable. 

So, after much experimentation, I shifted the pixel values so that my output was darkest for the closest objects, while the occluded areas were shifted more into the lighter grey range. The transformation was:

`image = abs(45 - image);`

And the resulting image was:
[image]

Still a bunch of noise in the image, but at least my depth output is consistent. 

Since I actually want to have a final output which is white for close values and dark for far values, I applied a simple mathematical inversion:

`bitwise_not(image, image);`

Giving me this result:
[image]

I then applied a thresholding operation and and erode/dilate operation to clean up the noise and isolate my hand (which was the whole goal of this exercise). Here is the code:

`threshold(image, image, 215, 255, THRESH_TOZERO);`
`Mat kernal = Mat::ones(4, 4, CV_8UC1);`
`morphologyEx(image, image, MORPH_OPEN, kernal);`

Which results in an image like this:
[image]

My only issue now is that I need to expand the usable depth range from the camera, since if I move more than 1.5 meters back, the threshold cutoff eliminates the hand altogether, and increasing the threshold range itself brings unwanted objects or artifacts back into the image.

However, that's a different problem than this issue is addressing, and I can work through that on my own. I should be able to use findContours() to isolate my gesture from the rest of the image now, since I have a 1-channel output image. Thank you so much for your help, Marty, even it some of it was just acting as my "rubber duck"!

MartyG-RealSense, 2020-06-30:
Thanks so much for the highly detailed feedback, which is sure to be of help to other RealSense users trying to accomplish the same goal.  I'm pleased that I could help you to quack the code!

kafan1986, 2020-07-01:
@dcprime I think you need to understand the simple math behind the scaling factor provides when converting from 16UC1 to 8UC1. 

The the depth data is 16UC1 format that is it can contain 0 to 2^16 - 1 values in single channel, which represent the actual depth in mm. So the depth camera can actually measure depth till 65 metre. When you are converting it 8UC1 then ideally you should give it scale value of 1/255.0, if you want to preserve the original depth information, although you will loose resolution of depth as 8UC1 will only store the integer values and not the decimal places. Earlier, each value can take different values in mm. Now each value can only be scaled between 0 to 255 different blocks of value, and each block now represent to (65535/255) mm, around 10 cm. So any granularity between the depth data is lost.

If your requirement is say would be sufficient in say 10 metre distance from camera. Then a scale value of 10/65.535 would be OK. In this case the highest depth that can be actually measured is around 10 metre (represented by 255 in 8UC1 format) and any farther distance will also be represented by 255.

dcprime, 2020-07-01:
Thanks for your insight, kafan1986. I think I have a decent theoretical grasp of the principles you describe, but doing the math to create my optimal output is still something I'm attempting to work out. I'm realistically interested in operating at distances between 30cm and 5m from the camera, so extending the full range of gradients to within that range would be best. 

However, my current thinking (which may be flawed - please let me know if it is) is that I'd like the gradient output of pixel values to operate on a non-linear scale, with less gradient differences closer to the camera, and more gradient differences farther from the camera. The reason for this is that I want to isolate a hand gesture held in front of the body, eliminating other elements behind the hand, but still providing enough space for the hand to move through so that portions of it don't disappear in the course of its movement.

If most (or all) of the hand-depth-values appear the same, that's fine. But I'd like the person performing the gesture to be able to stand anywhere in that 30cm to 5m range and still capture the hand data. The best situation would be to make the closest pixels to the camera always appear at full intensity (say, 255) while anything behind the arm (including the body of the actor) is thresholded out of the image.

Any ideas for making that happen?

kafan1986, 2020-07-01:
> Thanks for your insight, kafan1986. I think I have a decent theoretical grasp of the principles you describe, but doing the math to create my optimal output is still something I'm attempting to work [...]

By gesture data you mean the only the "hand + forearm till elbow" part right or do you want only the hand (finger till wrist)?  Also, you don't want the rest of the body to appear as a part of it. Can I assume the camera is stationary? Will you be holding your arm in front of the camera at a distance significantly closer than your body?

dcprime, 2020-07-01:
The answers to those questions, in order, are: 
1. Forearm and hand at most, entire hand at least.
2. Ideally no body, but I'm prepared to post-process to remove blobs of body pixels if necessary
3. Yes, camera is stationary.
4. Arm will be in front of body at a comfortable distance (imagine a semi-lazy man holding up his hand in a "stop" gesture, or an impatient piano teacher gesturing to "wrap it up" during a recital performance).

kafan1986, 2020-07-01:
> The answers to those questions, in order, are:
>
> 1. Forearm and hand at most, entire hand at least.
> 2. Ideally no body, but I'm prepared to post-process to remove blobs of body pixels if necessary
> 3. [...]

One more question, you are planning to work on camera stream data or standalone images?

dcprime, 2020-07-01:
The plan is to stream into a neural network using a sliding window of a fixed number of frames, but depending on if we can pull off the training or not, we may drop down to using still images as a simpler option.

kafan1986, 2020-07-01:
1) Convert the 16UC1 data to 8UC1 data by scale factor of 1.0f/14f. This will give you a lossless distance information till 5 metre approx.
2) Perform background subtraction, remember for the first frame, you better be not infront of camera as you will become the part of the background model, till continuous learning eventually decides that you are rather part of the foreground. Also, do control the learning rate, sensitivity of the background subtraction model for your FPS and relative movement infront of the camera, as you don't want to become part of the background model, if you are not making much movement yourself.
3) From the background subtraction step, you will get a foreground mask of 8UC1 values, where all the foreground part will be 255 (brightest) and rest will be 0 (dark). Foreground blob will ideally your entire body. You can perform the morphological operation (erosion and dilation) on the mask to remove small patches of noise.
4) If there are multiple people in front of the camera, make sure they are standing separately to form different blobs or additional steps are required, that I am not covering now. You separate the mask into different blobs for each person, and perform next set of steps, one blob at a time.
5) Multiply your actual depth data with this foreground mask, to get the depth values of only the actual body part of a single person and rest of the image is black (remains zero).
6) Now, using the depth data of the person, get a sort of histogram representation with say maximum 3-4 peaks, with each peak separated by each other with some minimum distance. Take the peak with the lowest distance and get all the pixels with exact lowest distance value and perform Moment to get centroid position and use it as a seed for region growing algorithm. Be strict in the region growing algorithm threshold and you should be able to grow the region that depicts only the arm part of person.

Hope I was clear. Let me know for any doubts.

MartyG-RealSense, 2020-07-01:
@dcprime   My two cents: if  I were looking to start a neural network project now then I would consider saving a lot of time and labor by basing it on **OpenVINO**, which is compatible with RealSense 400 Series cameras via a wrapper.

https://software.intel.com/content/www/us/en/develop/tools/openvino-toolkit.html
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openvino

dcprime, 2020-07-01:
@kafan1986 Thanks for the suggestions. I was able to follow your reasoning and I will take what you said into consideration. For now, I'm limiting the user space to including just a single person, and have put in a basic calculation that simply takes the closest values to the camera (within a range of 10 pixel value increments) and thresholds them to 255, while dropping everything else to 0. This assumes the gesture is always closest to the camera, but it seems to work fine for this stage of the process.

@MartyG-RealSense Thanks for pointing me toward the OpenVINO Toolkit. I don't know if we're ready to dive into a new framework at this stage of our project (we're just using modified PyTorch models for now) but I will check it out for future projects for sure.

closed 2020-07-02 by dcprime