realsenseai/librealsense #6728: Getting param of left camera(ir sensor) using SDK2.0 is different from using CustomRW tool.

LW-SCU, 2020-06-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D345i | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |  win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            | 2.33.1                     | 
| Language                            |  C++                        | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Getting param of left camera(ir sensor) using SDK2.0 is different from using CustomRW tool.
 I got param of left camera by SDK2.0,this is my code:
```c++
int main(){
    pipeline pipe;
    config cfg;
    cfg.enable_stream(RS2_STREAM_INFRARED,1,1280,800,RS2_FORMAT_Y8,15);
    pipe.start(cfg);
    {
        int i = 10;
        while(i--) pipe.wait_for_frames();
    }

    auto in = pipe.wait_for_frames().get_infrared_frame(1).get_profile().
            as<video_stream_profile>().get_intrinsics();
    cout<<"FocalLengthLeft:"<<in.fx<<"\t"<<in.fy<<endl;
    cout<<"PrincipalPointLeft:"<<in.ppx<<"\t"<<in.ppy<<endl;
    cout<<"DistortionLeft:"<<endl;
    for(int i = 0;i<5;i++){
        cout<<intrinsic.coeffs[i]<<endl;
    }
    pipe.stop();
    return 0;
}
output:
FocalLengthLeft:640.041	640.041
PrincipalPointLeft:635.598	399.094
DistortionLeft:
0
0
0
0
0
```
I got a param xml file using CustomRW tool.
```xml
<?xml version="1.0"?>
<Config>
    <param name = "ResolutionLeftRight">
        <value>1280</value>
        <value>800</value>
    </param>
    <param name = "FocalLengthLeft">
        <value>635.805</value>
        <value>636.461</value>
    </param>
    <param name = "PrincipalPointLeft">
        <value>634.809</value>
        <value>400.025</value>
    </param>
    <param name = "DistortionLeft">
        <value>-0.0520426</value>
        <value>0.0531889</value>
        <value>-0.0017655</value>
        <value>-0.000307534</value>
        <value>-0.0154513</value>
    </param>
    <param name = "FocalLengthRight">
        <value>635.986</value>
        <value>636.569</value>
    </param>
    <param name = "PrincipalPointRight">
        <value>632.48</value>
        <value>402.725</value>
    </param>
    <param name = "DistortionRight">
        <value>-0.0503853</value>
        <value>0.0503416</value>
        <value>-0.00142165</value>
        <value>-0.000518733</value>
        <value>-0.0139281</value>
    </param>
    <param name = "RotationLeftRight">
        <value>0.999944</value>
        <value>0.00105141</value>
        <value>-0.0104881</value>
        <value>-0.00102754</value>
        <value>0.999997</value>
        <value>0.00228133</value>
        <value>0.0104904</value>
        <value>-0.00227043</value>
        <value>0.999942</value>
    </param>
    <param name = "TranslationLeftRight">
        <value>-50.2541</value>
        <value>0.144023</value>
        <value>-0.153173</value>
    </param>
    <param name = "HasRGB">
        <value>1</value>
    </param>
    <param name = "ResolutionRGB">
        <value>1920</value>
        <value>1080</value>
    </param>
    <param name = "FocalLengthRGB">
        <value>1395.14</value>
        <value>1396.13</value>
    </param>
    <param name = "PrincipalPointRGB">
        <value>969.085</value>
        <value>552.345</value>
    </param>
    <param name = "DistortionRGB">
        <value>0</value>
        <value>0</value>
        <value>0</value>
        <value>0</value>
        <value>0</value>
    </param>
    <param name = "RotationLeftRGB">
        <value>0.999985</value>
        <value>0.00503999</value>
        <value>0.00220488</value>
        <value>-0.00503166</value>
        <value>0.99998</value>
        <value>-0.00376907</value>
        <value>-0.00222383</value>
        <value>0.00375792</value>
        <value>0.99999</value>
    </param>
    <param name = "TranslationLeftRGB">
        <value>15.0915</value>
        <value>-0.0578172</value>
        <value>-0.397592</value>
    </param>
</Config>
```
[image]
FocalLengthLeft, PrincipalPointLeft and DistortionLeft are different. Is my code wrong?

MartyG-RealSense, 2020-06-30:
I recalled that the intrinsics coefficients are meant to be zero.  The explanation for this is given in this discussion:

https://github.com/IntelRealSense/librealsense/issues/1430

LW-SCU, 2020-07-01:
Hi @MartyG-RealSense  
Which parameter should I use？SDK2 or CustomRW tool.

MartyG-RealSense, 2020-07-01:
My interpretation of the advice in the link would be that if the SDK sets the coefficients to zero because they are not used in the calculation of fx, fy, ppx and ppy then you should regard the coefficients as being zero too (i.e follow the SDK's way of doing things).

closed 2020-07-09 by LW-SCU

wangcongrobot, 2021-03-24:
When I rewrote different large parameters using **CustomRW tool**, the image was changed. So it means the coefficients are used to the image?

MartyG-RealSense, 2021-03-25:
@wangcongrobot   Y8 infrared is rectified.  Y16 is not rectified as it is used for camera calibration.