realsenseai/librealsense #6729: Errors while implementing the pointcloud functions

SiddharthBhurat, 2020-06-30:
### Issue Description

When I am trying to draw a pointcloud using the functions provided by librealsene and including the correct libraries, it is giving the errors related to function draw_pointcloud

These are included:-

[image]

These are the errors:-

[image]

Please can i get a reference which would help me to solve this problem

dorodnic, 2020-06-30:
Maybe missing `using namespace rs2;`?
Please also check if the provided examples compile (rs-pointcloud)

SiddharthBhurat, 2020-07-01:
@dorodnic 

`using namespace rs2;` was missing, I added that but still same errors are showing up

Also checked the provided examples, they are compiling correctly

SiddharthBhurat, 2020-07-01:
@dorodnic 

Do I also need to include imgui.h ???

ev-mp, 2020-07-02:
@SiddharthBhurat , you're trying to reuse parts of the Librealsense SDK's main tool - realsense-viewer rather that the SDK itself.
This is a heavy UI-oriented tool with multiple dependencies besides the SDK (i.e. imGui), and in general is not suited for Copy&Paste usage due to the coupling of the logic module with UI and rendering sub-system.

In case you're looking into Pointcloud generation and rendering I would recommend to check the Pointcloud demo from the [sample aplications list](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras) - it has more narrowed list of dependencies and classes and is better encapsulated for reuse.
If you're still interested to pursue the original path then you'll have to trace and copy the relevant classes and includes  till all the dependencies are satisfied.

MartyG-RealSense, 2020-07-05:
@SiddharthBhurat   If you are learning how to add dependencies into a librealsense project, the discussion in the link below may also be helpful. 

https://github.com/IntelRealSense/librealsense/issues/4277

RealSenseSupport, 2020-07-27:
@SiddharthBhurat With the reply from @ev-mp and @MartyG-RealSense ,  did you get it through? Any other questions?

RealSenseSupport, 2020-08-12:
@SiddharthBhurat Do you still need support on this? Looking forward to your update. Thanks!

SiddharthBhurat, 2020-08-12:
no
thanks a lot for your help

closed 2020-08-13 by RealSenseSupport