realsenseai/librealsense #6732: rs2::playback::seek is not instantaneous, returns some frames from previous position

ttencate, 2020-06-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | n/a
| Firmware Version                   | (Open RealSense Viewer --> Click info) | n/a
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | Linux
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 5.7.5-arch1-1
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  | PC
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 2.35.0
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | C++
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | AR

The `seek()` function of `rs2::playback` does not take effect instantly, even in fully synchronous code. After a call to `seek()`, `wait_for_frames()` will return several frames from the old playback location before it starts returning frames from the seeked location.

Here is a minimal example that reproduces the issue:

```c++
#include <librealsense2/rs.hpp>

#include <iostream>

int main(int argc, char **argv) {
  // Pass the name of a .bag file on the command line.
  rs2::config config;
  config.enable_device_from_file(argv[1], false);

  rs2::pipeline pipeline;
  rs2::pipeline_profile profile = pipeline.start(config);
  rs2::playback playback = profile.get_device().as<rs2::playback>();
  playback.set_real_time(false);

  // Timestamps are since Unix epoch, convert to since start of recording.
  double first_timestamp_ms = pipeline.wait_for_frames().get_timestamp();

  // Get (and discard) some frames.
  for (int i = 0; i < 10; i++) {
    rs2::frameset frameset = pipeline.wait_for_frames();
    double timestamp_ms = frameset.get_timestamp() - first_timestamp_ms;
    std::cout << "Got frame: " << timestamp_ms << '\n';
  }

  // Seek to new position.
  double seek_to_ms = 5000;
  std::cout << "Seeking to: " << seek_to_ms << '\n';
  playback.seek(std::chrono::duration_cast<std::chrono::nanoseconds>(std::chrono::duration<double, std::milli>(seek_to_ms)));

  // And get some more frames.
  for (int i = 0; i < 10; i++) {
    rs2::frameset frameset = pipeline.wait_for_frames();
    double timestamp_ms = frameset.get_timestamp() - first_timestamp_ms;
    std::cout << "Got frame: " << timestamp_ms << '\n';
  }
}
```

Output:

```
Got frame: 0
Got frame: 166.095
Got frame: 333.065
Got frame: 499.365
Got frame: 665.778
Got frame: 665.778  // I think there are dropped frames in our BAG file...
Got frame: 832.576
Got frame: 999.318
Got frame: 1165.74
Got frame: 1331.35
Seeking to: 5000
Got frame: 1331.35 // Getting stale frame once...
Got frame: 1331.35 // Twice...
Got frame: 4994.29 // Before we actually end up at the desired location.
Got frame: 5160.88
Got frame: 5328.02
Got frame: 5493.74
Got frame: 5660.33
Got frame: 5826.46
Got frame: 5993.01
Got frame: 6160.19
```

I assume there is some buffering going on. I searched through the issues and the documentation, and found [this wiki page](https://github.com/IntelRealSense/librealsense/wiki/Frame-Buffering-Management-in-RealSense-SDK-2.0) which suggests that `rs2::pipeline` has a hardcoded internal buffer of 1 frame, but I'm definitely getting 2 and sometimes more stale frames.

Interestingly, the timestamps on the stale frames seem to be always equal to each other, even if that timestamp is normally (when not seeking) printed only once (so it's not due to dropped frames in our BAG). This smells a bit like a bug to me.

Is there a way to turn off the buffering that causes this? If not, is there a way to determine how many frames are buffered, so I can simply drop them?

MartyG-RealSense, 2020-06-30:
In the link below, @dorodnic the RealSense SDK Manager explains that **set_real_time=false** does not help to provide random access to frames and the simplest approach to seeking a particular position is to reset to the start and skip forward to the desired frame.  This may be part of what you are experiencing.

https://github.com/IntelRealSense/librealsense/issues/3121#issuecomment-456476519

ttencate, 2020-06-30:
I read that comment by @dorodnic as a workaround for the "possibly additional bugs hidden there", so please consider this issue a report of one of those bugs :) Reading the entire stream from the beginning just to seek backwards one frame¹ is rather slow and awkward, so a proper fix would be nice.

Incidentally, seeking is very slow to begin with, and sometimes hits the 15000 milliseconds default timeout on `wait_for_frames`. This is for a 250 MB .bag file read from a hard drive (not SSD). I'm not sure how the BAG file format works; if there is no index, doesn't librealsense create an index when opening the file, and if not, should it?

--

¹ Just as an example; to allow seeking backwards exactly one frame, we could of course just buffer the previous frame. My use case is to have more general seeking ability.

MartyG-RealSense, 2020-07-01:
Bags undergo indexing when closing the pipeline at the end of recording.  If recording is improperly closed then it can result in a **bag unindexed** error when trying to play the file back.

I wonder if you would get greater stability if playback was paused before seeking a frame, with an instruction like this:

**device.as<rs2::playback>().pause()**

As you defined a variable for rs2::playback in your script, I think in your script, that would be written as **playback.pause()** ?

ttencate, 2020-07-01:
(Markdown tip: surround inline code fragments with backticks so your `<>` don't get misinterpreted as HTML.)

I'm manually requesting frames from the pipeline, so I don't see what the meaning of pausing and resuming would even be. However, it seems to have some effect! This is what I tried:

```c++
  playback.pause();
  playback.seek(std::chrono::duration_cast<std::chrono::nanoseconds>(std::chrono::duration<double, std::milli>(seek_to_ms)));
  playback.resume();
```

It's still not perfect; I occasionally get one stale frame still, but this is not reliably reproducible, and I haven't seen more than one. But it seems that whilst paused, the pipeline refuses to emit frames even when asked, so in this state we can read and discard all remaining stale frames from the buffer. Unfortunately, `poll_for_frames` returns `false`, but we can use `try_wait_for_frames` with a very short timeout, which seems to do the trick:

```c++
  playback.pause();
  playback.seek(std::chrono::duration_cast<std::chrono::nanoseconds>(std::chrono::duration<double, std::milli>(seek_to_ms)));
  rs2::frameset frameset;
  while (pipeline.try_wait_for_frames(&frameset, 10)) {
  }
  playback.resume();
```

It's a gross hack, but sufficient for my purposes. Better solutions are still welcome.

MartyG-RealSense, 2020-07-01:
The suggestion to pause before seek was based on the observation that sometimes the SDK likes to be given the chance to stop for breath and catch up.  Examples of this are auto-exposure taking several frames to settle down when a stream starts when auto-exposure is enabled, and getting a good balance between performance (fast but risk of frame drops) and latency (slower but stable) when processing frames.

I'm very glad that pausing made a positive difference to your program.

closed 2020-07-26 by MartyG-RealSense