realsenseai/librealsense #6735: Problem reading the Projector and Asic temperature in C++

malapatiravi, 2020-07-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415} | 
| Firmware Version                   | 5.12.5 | 
| Operating System & Version |   Ubuntu 18/20 | 
| Kernel Version (Linux Only)    |  5.3/5.4.0-39-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.35.2                         | 
| Language                            |  C++                          | 
| Segment			|  Robot                        | 

### Issue Description
I am writing the following code and trying to read the projector temperature. 

[image]

But what i see is that Asic and projector temperatures are not supported. But i did the exact same thing in Python. 
[image]

ev-mp, 2020-07-01:
@malapatiravi , the temperature reading for D400 can be acquired during streaming only.
Start streaming of the depth sensor, then poll the option till it becomes available.

malapatiravi, 2020-07-01:
@ev-mp 
Thanks for your reponse. 
I added the following code after wait_for_frames and still get that projector temperature is not supported. 

[image]

[image]

malapatiravi, 2020-07-01:
@MartyG-RealSense  Can you please help with this.

MartyG-RealSense, 2020-07-01:
I recalled that @dorodnic posted some workaround code for reading projector temperature in early 2018.

https://github.com/IntelRealSense/librealsense/issues/866#issuecomment-357461253

The whole long discussion is worth reading from the start too.

malapatiravi, 2020-07-01:
Thanks @MartyG-RealSense That workaround did the job for me. However i could not read Asic temperature. 

How is that this feature not working in C++ but working without any problem in python? Isn't Python using C++ APIs in the backend? Any idea when we will fix available in C++?

MartyG-RealSense, 2020-07-01:
I'm on the Support team rather than the Developer team, so @ev-mp will be better able to answer your new questions above.

Having said that, the link below has scripts for a couple of different approaches to reading ASIC temperature with C++ that you could try.

https://github.com/IntelRealSense/librealsense/issues/3888

I am off-shift now for the next 7 hours but will be happy to continue the support conversation when  return if you still require assistance.  Good luck!

ev-mp, 2020-07-02:
@malapatiravi , the temperature sensor requires certain warm-up time, so a single "wait_for_frames()" interval will not be sufficient. Therefore it is recommended to do it continuously as mentioned above
>Start streaming of the depth sensor, then poll the option till it becomes available.

You can check the way it is done in the Realsense-Viewer application (with controls polled every sec or so),

>How is that this feature not working in C++ but working without any problem in python? Isn't Python using C++ APIs in the backend? Any idea when we will fix available in C++?

In C++ things tend to happen little faster (not to mention the extra layers of indirection). It is likely that the first call made for the sensor temperature will arrive earlier than the sensor is ready to respond.

malapatiravi, 2020-07-02:
Thanks @ev-mp and @MartyG-RealSense 
I was able to solve the problem. I reduced the frequency of polling temperature.

closed 2020-07-02 by malapatiravi

MartyG-RealSense, 2020-07-02:
@malapatiravi   Thanks for the update - glad you found a solution!

dingbaiyin, 2020-10-11:
Sorry for not responding in time, I will close the issue.