realsenseai/librealsense #6737: Motion Module - Limited FPS

jc211, 2020-07-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version  |   Win 10 |  
| Platform                                 |   PC  |
| SDK Version                            |  2.34.0                          | 
| Language                            |  C++                        | 

### Issue Description
I can't seem to get the full 250 Hz on the motion sensor. I made a skeleton program based on the motion example and I don't understand why I'm being limited to 30-40Hz. Whats odd is that the realsense viewer shows the correct fps but my own code doesn't get the required fps. Am I configuring the camera in a wrong way?

```c++
#include <iostream>
#include <chrono>
#include <librealsense2/rs.hpp>
using namespace std::chrono;
using namespace std;

typedef std::chrono::duration<float, std::ratio<1, 1>> seconds_f;

bool check_imu_is_supported()
{
    bool found_gyro = false;
    bool found_accel = false;
    rs2::context ctx;
    for (auto dev : ctx.query_devices())
    {
        // The same device should support gyro and accel
        found_gyro = false;
        found_accel = false;
        for (auto sensor : dev.query_sensors())
        {
            for (auto profile : sensor.get_stream_profiles())
            {
                if (profile.stream_type() == RS2_STREAM_GYRO)
                    found_gyro = true;

                if (profile.stream_type() == RS2_STREAM_ACCEL)
                    found_accel = true;
            }
        }
        if (found_gyro && found_accel)
            break;
    }
    return found_gyro && found_accel;
}

float get_frequency() {
    static auto start = steady_clock::now();
    auto end = steady_clock::now();
    auto d = (seconds_f)(end - start);
    start = end;
    return 1.0f / d.count();
}

int main(int argc, char* argv[]) {
    if (!check_imu_is_supported())
    {
        std::cerr << "Device supporting IMU (D435i) not found";
        return EXIT_FAILURE;
    }

    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;
    // Create a configuration for configuring the pipeline with a non default profile
    rs2::config cfg;

    // Add streams of gyro and accelerometer to configuration
    cfg.enable_stream(RS2_STREAM_ACCEL, RS2_FORMAT_MOTION_XYZ32F);
    cfg.enable_stream(RS2_STREAM_GYRO, RS2_FORMAT_MOTION_XYZ32F);

    float fps = 0.0f;
    int i = 0;
    auto profile = pipe.start(cfg, [&](rs2::frame frame)
        {
            fps += 0.1*(get_frequency() - fps);
            i++;
            if (i % 10 == 0) {
                cout << fps << endl;
            }
            // Cast the frame that arrived to motion frame
            auto motion = frame.as<rs2::motion_frame>();
        });

    getchar();
    return EXIT_SUCCESS;
}

```

MartyG-RealSense, 2020-07-01:
Do you get the correct frequencies if you define them in your cfg statements, please?

**cfg.enable_stream(RS2_STREAM_ACCEL, RS2_FORMAT_MOTION_XYZ32F, 250);
cfg.enable_stream(RS2_STREAM_GYRO, RS2_FORMAT_MOTION_XYZ32F, 400);**

jc211, 2020-07-01:
> 
> 
> Do you get the correct frequencies if you define them in your cfg statements, please?
> 
> **cfg.enable_stream(RS2_STREAM_ACCEL, RS2_FORMAT_MOTION_XYZ32F, 250); cfg.enable_stream(RS2_STREAM_GYRO, RS2_FORMAT_MOTION_XYZ32F, 400);**

Yes the callback is still limited to around 40 Hz even after the above adjustments.

MartyG-RealSense, 2020-07-01:
Apologies for the delay in responding further, as I was carefully researching your case.

I got the most relevant research results from focusing on the **callback** aspect of the problem.  The link below is the one that emerged from my research that seems to be most similar to your own problem:

https://github.com/IntelRealSense/librealsense/issues/6424

jc211, 2020-07-01:
It seems I'm having the same problem as the #6424 

With the pipeline method, I can get the expected number of frames per seconds by enabling the depth stream.
With the device method, I can get the expected number of frames by setting the Global Time to 0. 

Is this a firmware problem or a problem with my version of the library?

In any case, thank you for referring me to the above issue. The above workarounds should get me by for now.

closed 2020-07-01 by jc211

MartyG-RealSense, 2020-07-01:
Thanks for the update!

My knowledge of programming callbacks is not extensive, but my past experience of cases involving them suggests that if problems occur,  it tends to be related to users' scripts rather than a particular SDK or firmware.  Callbacks are not the easiest mechanism to program.

Good luck!