realsenseai/librealsense #6738: 'poll_for_frames()' method doesnt work in matlab.

mk-kor, 2020-07-01:
----------------------------------------------------------------------------------------------------

| Required Info                         | 'poll_for_frames()' method doesnt work in matlab. |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415} | 
| Firmware Version                   |  | 
| Operating System & Version |   Win10 | 
| Kernel Version (Linux Only)    |    | 
| Platform                                 | PC |
| SDK Version                            |  2.35.2  | 
| Language                            |  Matlab }                          | 
| Segment			| Robot                     | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hi.

Im trying to use 'poll_for_frames()', but it doesnt work in matlab.

It seems matlab cant recognize what is 'res'.

Codes.
width = 424;
height = 240;
fps = 60;

pipe = realsense.pipeline();
cfg = realsense.config();
cfg.enable_stream(realsense.stream.infrared,width,height,realsense.format.rgb8,fps);
cfg.enable_stream(realsense.stream.depth,width,height,realsense.format.z16,fps);

profile = pipe.start(cfg);

for i = 1:5          %% 1:20
         fs = pipe.wait_for_frames();
end

while(1)
fs = pipe.poll_for_frames();
color=fs.get_color_frame();
data=color.get_data();
img = permute(reshape(data2,[3,width,height]),[3 2 1]);
imshow(img2)
end

Error message.

'res' is an unrecognized function or variable.
error : realsense.pipeline/poll_for_frames (line 59)
            res, out = realsense.librealsense_mex('rs2::pipeline', 'poll_for_frames', this.objectHandle);

MartyG-RealSense, 2020-07-01:
**poll_for_frames** support for the RealSense MATLAB wrapper was added to the SDK in version **2.34.0**.  So it should be accessible in 2.35.2.

https://github.com/IntelRealSense/librealsense/pull/5970
https://github.com/IntelRealSense/librealsense/pull/5970/commits/f516a8a6cd702672d8c002edb3cf5b75109dc48f

@lramati  Could you advise please?  Thanks!

mk-kor, 2020-07-13:
#6786

Fixed ! I'll close the case.

thanks.

closed 2020-07-13 by mk-kor

MartyG-RealSense, 2020-07-13:
Great news - thanks for the update!