realsenseai/librealsense #6739: Load JSON and record

ggeof, 2020-07-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515i | 
| Firmware Version                   | 2.35.2.0 | 
| Operating System & Version |  Ubuntu 20  | 
| Kernel Version (Linux Only)    |  5.4.0                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.35.2                          | 
| Language                            |  C++                       | 
| Segment			|  Robo| 

### Issue Description
Hello,
I search to launch a Record. But before, I want to load a JSON file for my device. But the code
<code>
>  rs2::config cfg;
> cfg.enable_record_to_file(fileBag);
> this->profile = this->p->start(cfg);
> auto advanced_mode_dev = serializable_device(this->p->get_active_profile().get_device());
> std::ifstream t(myFile);
> std::string preset_json((std::istreambuf_iterator<char>(t)), std::istreambuf_iterator<char>());
> advanced_mode_dev.load_json(preset_json);
</code>
Not working...
it turns me on : `null pointer passed for argument "dev"`
I suppose because `this->p->get_active_profile().get_device()` it's a `recorder`.
Do you have a solution for my problem ?
Or it's possible to record starting from a device that's already "started" ?

ev-mp, 2020-07-02:
@ggeof , The `serializable` extension interface requires dynamic rather that static cast, so adapting the answer from #4046 would result in 
```cpp
std::string preset_json((std::istreambuf_iterator<char>(t)), std::istreambuf_iterator<char>());
if (auto advanced_mode_dev = p->get_active_profile().get_device().as<serializable_device>();)
{
    advanced_mode_dev.load_json(preset_json);
    ....
```

ggeof, 2020-07-03:
I no longer have the mistake for your code. But my configuration is not applied to my camera...
So it's impossible in this time to load configuration in a record or in a playback ?

MartyG-RealSense, 2020-07-03:
Hi @ggeof   Are you trying to set the resolution or FPS speed with your json file before starting a recording, please?  If you are, this should be done with **cfg** statements, as json settings for resolution and FPS will get overridden by the pipeline when it starts.

ggeof, 2020-07-06:
Hi @MartyG-RealSense,
No I used the L515i and I want change the "Apd Temperature" and the "Min Distance" in my application. And I've set up unit tests thanks to the recorder and the playback. But since the sensor parameters are not optimized, I don't have a very good score....

MartyG-RealSense, 2020-07-06:
I believe that the "APD Temperature" refers to a camera component called the Avalanche Photodiode.

https://en.wikipedia.org/wiki/Avalanche_photodiode

Temperatures are usually read-only, and changing the value of a temperature with scripting would defeat the purpose of monitoring it.  Instead, you could use indirect ways to affect operating temperatures (for example, altering the **Laser Power** value).

Minimum distance can be affected by changing the value of the **Disparity Shift** option.  Increasing its value reduces minimum distance (MinZ) and enables the camera to get closer to an object, though it also lowers the maximum observable distance (MaxZ).

Another way of affecting the minimum distance of depth is to set a **threshold filter** post-processing filter that ignores data that is below a defined minimum distance.  The link below has a C++ script that includes code within it for setting the Threshold Filter

https://stackoverflow.com/questions/59054413/intel-realsense-depth-camera-d435i-noises-and-mounds

ggeof, 2020-07-06:
Oops I was wrong I was talking about laser power. I don't know why I mentioned the temperature. 
Yep I'm going to do it "by hand" to change the parameters that interests me but I would have preferred to do it with a JSON, more practical when changing parameters ...
For the distance, the threshold filter doesn't work for the L515... So I need modify the parameter

MartyG-RealSense, 2020-07-06:
Do you need some C++ code for changing the Disparity Shift value, please?

ggeof, 2020-07-06:
No thank you. I will play with the function
` void rs2_set_option(const rs2_sensor* sensor, rs2_option option, float value, rs2_error** error);`

RealSenseSupport, 2020-07-13:
@ggeof 

Do you need further assistance on this?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-07-21 by RealSenseSupport