realsenseai/librealsense #6740: Pi4 + SR305: Frame didn't arrived within 5000

eldonaldo, 2020-07-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR305  | 
| Firmware Version                   | 3.27.3.0 | 
| Operating System & Version |   Raspbian  | 
| Platform                                 | Raspberry Pi 4, powered over ethernet  |
| SDK Version                            |   2.0                           | 
| Language                            |  Python 3.7.3                          | 
| Segment			|  Robot                           | 

### Issue Description
Hi Intel Realsense team,

we are experiencing an issue with the SR305 which is mounted on an ethernet powered Raspberry Pi. We sometimes (bot not always) receive the following error message during the `pipeline.wait_for_frames()` call: **Frame didn't arrived within 5000 issue** and frames get dropped.

Do you have any hints for us? We assume that it MIGHT be related to powering issues, i.e. that the Pi can't deliver the required watts over usb 3.0 to the SR305. We already checked all git issue related to 5000er errors and search nearly the entire web, but no proposed solution has worked for us so far :(

Thank you for your help!
Cheers

MartyG-RealSense, 2020-07-01:
Hi @eldonaldo   If you are not experiencing apparent performance loss from the **frame didn't arrived within 50000** error then using **try_wait_for_frames()** instead might be a solution.  Like wait_for_frames, it blocks until a frame is received.  The difference though is that it does not throw on timeout but instead returns **false**.

If it is an occasional error rather than a constant one, I would also check for the possibility of whether the Pi board is overheating.

dorodnic, 2020-07-09:
Hi @eldonaldo 
Please check (and attach) `dmesg` log when starting streaming. 
SR305 creates peak power usage on start. Hence its likely the camera is detected correctly, but when you try to start streaming the Pi is unable to provide enough power causing the camera to disconnect (and hence wait_for_frame to time-out)

MartyG-RealSense, 2020-07-18:
Hi @eldonaldo   Do you still require assistance with this case, please or can it be closed?   Thanks!

eldonaldo, 2020-07-20:
Dear friends, thank you very much for your help and sorry for the delayed response. Since we had a hard deadline and couldn't add power supply so quickly we decided to add scheduled restarts which solved the problem temporarily. We didn't experience the issue again. However, @dorodnic answer sounds plausible and thus we will add a dedicated power supply for the SR305 to (hopefully) finally get rid of the issue for ever.

MartyG-RealSense, 2020-07-20:
Thanks so much for the update @eldonaldo   Please let us know the results of your dedicated power supply testing when you have them.

MartyG-RealSense, 2020-08-02:
Hi again @eldonaldo  Do you have test results from your dedicated power supply testing to provide yet?   Thanks!

MartyG-RealSense, 2020-08-08:
Case closed due to no further comments received after 3 weeks.  Please open a new issue if you wish to continue with support for your problem later.  Thanks!

closed 2020-08-08 by MartyG-RealSense