realsenseai/librealsense #6741: Unable to calibrate D435 using Dynamic Calibrator tool

mfaizanse, 2020-07-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 series (D435) } | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |   {Win 10  | 
| Kernel Version (Linux Only)    | -                                        | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.35.2 }                          | 
| Language                            |  python                        | 
| Segment			|  {AR/others }                          | 

### Issue Description
I am trying to calibrate my Intel realsense D435 using Intel Realsense Dynamic Calibrator Tool (v2.11.0.0). I get to the blue boxes screen but unable to disappear all the blue boxes, the problem is that if i disappear the bottom left blue boxes then the other side boxes reappears,then if i disappear them then the left side blue boxes appear again. I have tried using printed pattern as well as using my iphone 6s plus. 

[image]
[image]

The factory calibration is not working good for depth estimation, so I need to calibrate the camera. Next, I wanted to then calibrate it using Matlab or OpenCV. For that I tried to get the unrectified IR images for the format y16 but I get the error: 
```
RuntimeError: Couldn't resolve requests
```
My python script is:
```
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.infrared, 1, 1280, 800, rs.format.y16, 6)
config.enable_stream(rs.stream.infrared, 2, 1280, 800, rs.format.y16, 6)
pipeline_profile = pipeline.start(config)
try:
    while True:
        frames = pipeline.wait_for_frames()
        ir1_frame = frames.get_infrared_frame(1) # Left IR Camera, it allows 0, 1 or no input
        ir2_frame = frames.get_infrared_frame(2) # Right IR camera
        if not ir1_frame or not ir2_frame:
            continue
        ir1_image = np.asanyarray(ir1_frame.get_data())
        ir2_image = np.asanyarray(ir2_frame.get_data())

finally:
    pipeline.stop()
```
The camera is connected through USB 3.2 port. And I can get the IR y16 stream on RealSense Viewer.

Please let me know how can I calibrate it using the Dynamic Calibrator Tool? or how can I get the y16 stream for IR in my python script?

MartyG-RealSense, 2020-07-01:
I had a similar case to yours earlier this month on the Intel Support RealSense forum.

https://support.intelrealsense.com/hc/en-us/community/posts/360046848333/comments/360011596314

The advice that worked for them to improve their camera image was to use the newer **On-Chip Calibration** tool built into the RealSense Viewer instead of using the Dynamic Calibrator software.

You can find this feature by launching the RealSense Viewer, going to the **More** option at the top of the Viewer's options side-panel, and selecting **On-Chip Calibration** from the More option's drop-down menu.  It requires Librealsense SDK version **2.29.0** or newer and firmware version **5.11.15.0** or newer. 

The link below leads to an Intel guide to using On-Chip Calibration.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras

mfaizanse, 2020-07-01:
Hi, thanks for the answer.

I have already tried the On-Chip Calibration but its highly accurate. I get a lot of holes in the depth map even if I place it in-front of a flat wall. That's why I wanted to do the full calibration.

MartyG-RealSense, 2020-07-01:
You could try following it up with the **Tare** function from the same **More** menu, as that focuses on depth accuracy improvement.

You may also be able to fill hopes without calibration by using post-processing filters such as the **hole Filling Filter**.  These can be accessed in the RealSense Viewer's **Post Processing** options section or programmed into an application with scripting.

[image]

**BEFORE HOLE FILLING FILTER**

[image]

**AFTER HOLE FILLING FILTER**

[image]

mfaizanse, 2020-07-01:
I have again done  On-Chip Calibration with slow mode for intrinsics (Health check: 0.08) and extrinsics (Health check: 0.06), and these are my resuts:
[image]
[image]

Does these results look good to you? As your results in last comment looks so smooth even without post-filtering.

MartyG-RealSense, 2020-07-02:
The results look good to me, considering that there is no post processing filters at all being applied.  It is a bit rough at the far sides on the point cloud, though this may just be where the side walls of the room begin and the camera is having difficulty reading depth from the sharp change of wall angle.

mfaizanse, 2020-07-03:
The for the feedback. Do you have any suggestions for making depth quality better especially for the edges and corner?

MartyG-RealSense, 2020-07-03:
Aside from using the optimal depth accuracy resolution for the D435, which is **848x480**, you may get cleaner edges and reduced point-cloud spraying if you set the Viewer to use the **Default** custom preset instead of 'Custom'.

[image]

mfaizanse, 2020-07-04:
Can you also tell me how can I get the y16 stream for IR in my python script?

MartyG-RealSense, 2020-07-04:
Y16 is not an ideal mode to use, as it is an unrectified stream intended for use with calibration.  It only works if IR only is enabled (i.e depth is disabled).  Tests that I just conducted with the RealSense Viewer and a D435i provided only these stream / FPS combinations that it would work with:

**640x400 at 15 or 25 FPS
1280x800 at 15 or 25 FPS**

The link below has a script for displaying the IR streams only with pyrealsense2.

https://github.com/IntelRealSense/librealsense/issues/3294#issuecomment-466042259

Change the **config.enable_stream** to the resolution, format and FPS speed that you want to use.  For example, to get the left (1) and right (2) infrared channels at 1280x400 Y16 and 25 FPS:

**config.enable_stream(rs.stream.infrared, 1, 1280, 800, rs.format.y16, 25)
config.enable_stream(rs.stream.infrared, 2, 1280, 800, rs.format.y16, 25)**

mfaizanse, 2020-07-06:
Thanks very much for your help.

closed 2020-07-06 by mfaizanse