realsenseai/librealsense #6742: RS Camera on android

nawara72, 2020-07-02:
Hi
 I have just started introducing myself to android, so I apologize in advance for the simple questions/incorrect terms I use.

I have just compiled the android code that comes with realsense library  and installed it on my Samsung Note9. 
The RS recording detects the D435 and displays the depth image on my mobile, however the rs camera app doesnt detect the D435 and keeps displaying the message to connect the camera. I got the same behaviour when I installed the rs camera app from playstore, it didnt detect the D435 on my note9 however when installed the app on my Samsung S8, the D435 was detected and app worked. Any idea why the rs recording detects camera on Note 9 but the rs camera doesn't?

Thanks

MartyG-RealSense, 2020-07-02:
If your Samsung Note 9 is running Android 10 then it may have a conflict with the RealSense Android wrapper.  A fix is to set **targetSdkVersion** to an earlier version such as '27'.

nawara72, 2020-07-02:
Thanks Marty, I will look into it further but just can't see how on the same mobile Note9 with android 10, the RS Recording works fine, detects the D435 connected  and displays the depth and RGB image but the RS Camera runs but doesn't detect a D435 connected. Will try to see f I can resolve it. Thanks

nawara72, 2020-07-02:
Marty my apologies , you were right changing the targetsdkversion solved the problem. I didn't know/notice that the rs recording had a different version from the rs camera. It works now. Thanks

MartyG-RealSense, 2020-07-03:
No problem at all.  Thanks for the update about your success!

closed 2020-07-03 by nawara72