realsenseai/librealsense #6746: How can I spend less time exporting a ply file?

DreamLFairy, 2020-07-02:
|---------------------------------|------------------------------------------- |
| Camera Model                       |L515| 
| Operating System & Version |  raspberrypi aarch64  | 
| Kernel Version (Linux Only)    |  5.4.42                                  | 
| Platform                                 |Raspberry Pi |
| SDK Version                            |  master/lastest            | 
| Language                            | c++        | 

### Issue Description

I run with visual studio 2017 and release version on Windows i7,the function "points.export_ To_ Ply (cloudpath, depth)" takes 110-180 ms, but I run release version on raspberrypi ,the function "points.export_ To_ Ply (cloudpath, depth)" takes 4000-5000ms.I can't accept taking such a long time.
==1. Are there any other functions that generate point cloud that takes less time?
==2. How can I do to remove the color and texture when generating point cloud so that it should take less time?I just need the xyz distance of point cloud
==3. Or do you have other suggestions?

MartyG-RealSense, 2020-07-02:
Hi @DreamLFairy 

1.  Writing to formats other than **bag** (the most efficient RealSense recording format) can be a slow process.  If you are experiencing a large difference in write time between Windows and Pi, then it may be that the Pi's storage has a lower access speed than a PC hard drive (especially a fast SSD hard drive).

The SDK has a function called **keep()** that can store frames in memory instead of continually writing to storage and then save them all in a single action when the recording pipeline closes.  This may not be practical for the memory capacity of a Pi though, and a non-starter on a low-memory model such as Pi 3b.

A Pi can have a fast SSD drive attached to it, such as the example in the link below:

https://www.tomshardware.com/uk/news/raspberry-pi-4-ssd-test,39811.html

Google **raspberry pi ssd drive** for more information.

2.  The link below has a simple script without a colorizer for generating a point cloud from depth using the SDK's pointcloud processing block.  It has a possibility of having a memory leak, but you should be able to avoid it if you follow the guidance in the comments.

https://github.com/IntelRealSense/librealsense/issues/2860

DreamLFairy, 2020-07-02:
ok,I have solved the problem, I write a function for exporting to ply, it takes 600-750ms, we can accept the time

closed 2020-07-02 by DreamLFairy

MartyG-RealSense, 2020-07-02:
Thanks so much for the update - great news that you found a satisfactory solution!