realsenseai/librealsense #6748: I have hard time compiling even gradle assembleRelease

MarcWeber, 2020-07-02:
How to reproduce? on OSX
1) get latest AndroidSdk
2) get latest gradle (brew install gradle)
Newest gradle comes with it's own wrapper, right ?
Tried running gradle from shell and from within AndroidSDK running into different issues.
Shell: 
```
* What went wrong:
A problem occurred configuring project ':capture'.
> Failed to notify project evaluation listener.
   > 'org.gradle.api.file.RegularFileProperty org.gradle.api.file.ProjectLayout.fileProperty(org.gradle.api.provider.Provider)'
```

AndroidSDK:  The NDK is not configured. Yes maybe it's simple just adding some configuration lines. But maybe somebody has already spent the time. 

I also tried the development branch. Similar results. Have I missed something simple ?

MartyG-RealSense, 2020-07-02:
It may not be a problem related to BREW or to Librealsense.  The link below has advice on making sure that NDK is installed and set up.

https://stackoverflow.com/questions/40943764/gradle-sync-failed-ndk-not-configured-download-it-with-sdk-manager/41348944

kafan1986, 2020-07-02:
@MarcWeber There are 2 ways of installing NDK from android SDK manager. Old way and new way (side-by-side). If you are side by side ndk. Then just adding the ndk version number under the **android** block, in the build.gradle of the app should allow you to build it fine. My build.gradle looks like below.

```
apply plugin: 'com.android.application'

android {
    ndkVersion '21.3.6528147'
    compileSdkVersion 29
```

MarcWeber, 2020-07-14:
I still have issues. First, it's enough to cd into the wrappers/android dir?
Second how to run gradle ? gradle 7.0  has 'gradle wrapper' and --gradle-version. Using eg 7.0 after having added the ndkVersion to all build files having 'com.android.application' yields:

```
FAILURE: Build failed with an exception.

* Where:
Build file '/Users/marc/librealsense3/wrappers/android/examples/native_example/app/build.gradle' line: 3

* What went wrong:
A problem occurred evaluating project ':app'.
> No signature of method: build_2w0vqbxqy6go4s3xkyj45b12w.android() is applicable for argument types: (build_2w0vqbxqy6go4s3xkyj45b12w$_run_closure1) values: [build_2w0vqbxqy6go4s3xkyj45b12w$_run_closure1@1b2f6c85]

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

Deprecated Gradle features were used in this build, making it incompatible with Gradle 7.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/6.5.1/userguide/command_line_interface.html#sec:command_line_warnings
```

I installed gradle with homebrew. ./gradlew says ```Error: Could not find or load main class org.gradle.wrapper.GradleWrapperMain```

kafan1986, 2020-07-14:
I have used android studio to compile my android project and haven't used gradle to build project directly.

MarcWeber, 2020-07-19:
So how have you told Android Studio to use librealsense ? Native or Java version ?  The assembleRelease in the title fo this issue is a gradle build task within this git repository within wrappers/android. Basically I'd be happy just running the examples shipping within this git repository.

kafan1986, 2020-07-19:
> So how have you told Android Studio to use librealsense ? Native or Java version ? The assembleRelease in the title fo this issue is a gradle build task within this git repository within wrappers/android. Basically I'd be happy just running the examples shipping within this git repository.

I have used in Native as well Java. For simple Java usage with existing exposed java API, you can add the dependencies in your gradle, so that it downloads the respective libraries.

For Native use you need to add the "include" folder of the latest (or as appropriate) SDK source code. My cmakelist.txt for native build looks like this for the librealsense integration.

```
#################### librealsense dependencies begin ####################
include_directories("${CMAKE_CURRENT_SOURCE_DIR}/include")
add_library(realsense2 SHARED IMPORTED)
set_target_properties(
        realsense2
        PROPERTIES
        IMPORTED_LOCATION
        ${CMAKE_CURRENT_SOURCE_DIR}/../../../build/jni/${ANDROID_ABI}/librealsense2.so
)
####################  librealsense dependencies end  ####################
```

AMArostegui, 2020-07-21:
I was having a similar problem to build _java_example_, on Windows 10 using Android Studio 4.0

As I understand (I’m a newbie), we have three programs interacting in the build process:

- Gradle
- Android Studio Gradle Plugin
- Gradle Wrapper

As we clone the _Realsense_ repository and browse to _java_example_, there is no `/gradle/wrapper/` folder and no `gradle-wrapper.properties` file so I assume Intel has not configured the wrapper for this repository.

When importing the project with Android Studio, those folders are not found, so the IDE creates them automatically, using the versions it decides on its own.

Now, as gradle wrapper is a helper tool to ensure the building reproducibility, if Android Studio chooses on its own, it defeats the goal of the wrapper.

In my case, the settings Android Studio created were:

Gradle 6.1.1
Gradle plugin 3.2.1

From [this file](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/build.gradle) it seems like my gradle plugin version is ok, so the problem must be on gradle itself.

And so, in Android Studio, `File => Project Structure => Project`, changing gradle version from the default 6.1.1 to 4.10.2, the project syncs and builds.

MartyG-RealSense, 2020-08-08:
Hi @MarcWeber   Do you still require assistance with this cae, please, or can it be closed?   Thanks!

FreyWang, 2020-12-17:
> I was having a similar problem to build _java_example_, on Windows 10 using Android Studio 4.0
>
> As I understand (I’m a newbie), we have three programs interacting in the build process:
>
> * Gradle
> * [...]

you save my life! oh my god ！！！！！！！！！！！！！

zc08, 2020-12-18:
I'm having the same problem, downgrading gradle from 6.1.1 to 4.10.2 didn't help...