realsenseai/librealsense #6750: Realsense viewer depth and color alignment toggle button

tispratik, 2020-07-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.03.00| 
| Operating System & Version |   Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.15.0-88-generic                                         | 
| Platform                                 |    - |
| SDK Version                            |  2.33.1                          | 
| Language                            |  python                         | 
| Segment			|  Robot                          | 

### Issue Description
We recorded a bag file with the viewer, but the depth and color frames are not aligned. Can we have a toggle button in the viewer to align the frames before recording the bag files?

MartyG-RealSense, 2020-07-02:
Hi @tispratik  Bag files do not store alignment, so using a button to align the frames before recording would not save the frames in an aligned state.

https://github.com/IntelRealSense/librealsense/issues/4306#issuecomment-506426275

tispratik, 2020-07-02:
Thanks for the clarification @MartyG-RealSense

closed 2020-07-02 by tispratik