realsenseai/librealsense #6754: Strange distortion coefficients of L515

cedrusx, 2020-07-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.04.01.00 | 
| Operating System & Version |  Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |    | 
| Platform                                 | PC  |
| SDK Version                            |  2.35.2                          | 
| Language                            |  N/A                          | 
| Segment			|  Robot                          | 

### Issue Description

I have two L515 devices, both updated to the latest firmware. I recorded some data with realsense-viewer, and checked the depth camera intrinsics from the topic `/device_0/sensor_0/Depth_0/info/camera_info`. For one of the devices, there are strange distortion values:
```
distortion_model: "UNKNOWN"
D: [4.5902333795888033e-41, -1.1269932900418869e+30, 4.590093249742371e-41, 0.0, 0.0]
K: [463.21484375, 0.0, 348.6484375, 0.0, 463.4453125, 245.275390625, 0.0, 0.0, 1.0]
```

The other looks fine:
```
distortion_model: "None"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [466.015625, 0.0, 356.12890625, 0.0, 466.7890625, 241.3671875, 0.0, 0.0, 1.0]
```

Seems to be a small issue, but may affect some existing programs which parse intrinsics from the camera_info topic.

Should I expect no distortion for the depth image from any L515 devices?

Thank you.

MartyG-RealSense, 2020-07-04:
Hi @cedrusx   It looks as though your problem may be related to the case in the link below.

https://github.com/IntelRealSense/librealsense/issues/6609

A fix to initialize the L515 coefficients and set the distortion model to None was this week merged into the next SDK release.

https://github.com/IntelRealSense/librealsense/pull/6615

cedrusx, 2020-07-04:
Hi @MartyG-RealSense thank you for the quick response. I'll close this issue.

closed 2020-07-04 by cedrusx