realsenseai/librealsense #6756: Playback Function for android

ankittecholution, 2020-07-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Operating System & Version |  Android  | 
| Platform                                 | Android |
| SDK Version                            |  2.31.0                           | 
| Language                            |  Android java wrraper                         | 

### Issue Description
I wanted to create a application for playing the bag file in android but I saw that we have a class for that but it does not support most of the function that is required by me like:- 
    1. pipeline_profile.get_device().as<rs2::playback>().set_real_time( false );   
    2. pipeline_profile.get_device().as<rs2::playback>().get_position();

I thought i can get these information with native implantation example but when I tried there are some issues 
This the JNI code that I wrote in c++:-

```
extern "C"
JNIEXPORT void JNICALL
SPOOF_JNI_METHOD(playback)(JNIEnv *env, jclass clazz, jlong device) {
    rs2::pipeline_profile *pipelineProfile = reinterpret_cast<rs2::pipeline_profile *>(device);
    pipelineProfile->get_device().as<rs2::playback>().set_real_time(false);
}
extern "C"
JNIEXPORT jint JNICALL
SPOOF_JNI_METHOD(playbackPosition)(JNIEnv *env, jclass clazz, jlong device) {
    rs2::device *device1 = reinterpret_cast<rs2::device *>(device);
    return static_cast<jint>(device1->as<rs2::playback>().get_position());
}
```

Here is the code of JAVA how I am calling those JNI:-
 ```
try (Config config = new Config()) {
                config.enableDeviceFromFile(filePath);
                try (Pipeline pipeline = new Pipeline()) {
                    try {
                        pp = pipeline.start(config);
                        spoof1.playbacktest(pp.getDevice().getHandle());
                        while (!mStreaming.isInterrupted()) {
```

My requirement is to run the file frame br frame by my processing block and there are multiple bag file so i need to know the when a file is also ending so that i can start other file also.

MartyG-RealSense, 2020-07-05:
Hi @ankittecholution   Could you provide information please about the specific issues that you are encountering with your C++ JNI code, please?

ankittecholution, 2020-07-05:
Actually, I want to use 2 functions in playback which are not there is android JNI. so I tried to create a JNI in Native android example and send the pipeline profile handle to it and tried to access those function but the app was crashing and when I looked into it was crashing in handle error it was going till default and then the app was crashing. this function that I wan to use are:-
1. pipeline_profile.get_device().asrs2::playback().set_real_time( false );
2. pipeline_profile.get_device().asrs2::playback().get_position();

MartyG-RealSense, 2020-07-05:
Next, may I check with you that:

1.  You are using the version of the Android RealSense wrapper for **unrooted** Android devices.

https://dev.intelrealsense.com/docs/android-build-of-the-intel-realsense-sdk-20

2.  The example native code program provided in the wrapper is able to work (confirming whether or not the problem is with your playback code specifically or the setup of native access in general).

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/native_example/readme.md

ankittecholution, 2020-07-05:
1. yes, I am using for unrooted devices. The Android RealSense wrapper is 2.31.0 version. 
2. The playback is working fine I have tried that so I would say the playback wrapper is working.
I need access and call a function called as<rs2::playback>().set_real_time(false); from java so that I can run the playback as realtime false. So please guide me to do so.

ankittecholution, 2020-07-05:
I am not able to cast the pipelineprofile in c++ JNI when I sending the handle of it in JNI.
Code where i am sending the handle of pipelineprofile in java
```
 PipelineProfile pp = pipeline.start(config);
testJNI.playbacktest(pp.getHandle());
```

JNI function that I created:-
```
extern "C"
JNIEXPORT void JNICALL
test_JNI_METHOD(playback)(JNIEnv *env, jclass clazz, jlong device) {
    rs2::pipeline_profile *pipelineProfile = reinterpret_cast<rs2::pipeline_profile *>(device);
    pipelineProfile->get_device().as<rs2::playback>().set_real_time(false);
}
```

MartyG-RealSense, 2020-07-05:
BTW, the link below has an interesting example of a RealSense user's code for finding the end of a bag file.  It may be a helpful C++ reference if you have not seen it already.

https://support.intelrealsense.com/hc/en-us/community/posts/360031847974/comments/360007849114

MartyG-RealSense, 2020-07-05:
May I also confirm that your code above is able to successfully call other functions using C++ but not set_real_time (which would suggest that the structure of the code is okay and it is the calling of set_real_time that is the problem).

ankittecholution, 2020-07-12:
I agree, but I think the cast has some problem because when I am sending the PipelineProfile handle and trying to cast it to rs2::pipeline_profile in c++ and trying to access any function it is crashing.

MartyG-RealSense, 2020-07-12:
I thought about the above question and looked into it but I do not have the Android to C++ communication knowledge to answer it.  

@ev-mp  Might you be able to advise @ankittecholution on the above query, please?

MartyG-RealSense, 2020-07-18:
@ankittecholution  Do you still require assistance with this case, please?   Thanks!

ankittecholution, 2020-07-18:
Yes I am still not able to do it. I need more assistance

MartyG-RealSense, 2020-07-18:
If **set_real_time support** is not currently provided in the Android wrapper then the best way to deal with your request would be for me to file an official feature request with Intel to consider adding it.  This may not be a quick process, but I will make the request now and set the process in motion so that Intel can decide whether or not to implement it if it is not already supported.

AMArostegui, 2020-09-30:
Yeah. Not available in 2.37.0 yet. Would be nice to have this feature

Mad-Thanos, 2020-10-12:
hi @ankittecholution, @MartyG-RealSense  and @AMArostegui 
I guess the problem is: the librs handle in java code can *NOT* be directly `reinterpret_cast` to c++ object. Instead, it can only be casted to the underlying c struct pointer of the c++ object. It means you have to convert the `PipelineProfile.getHandle()` jlong to `rs2_pipeline_profile *` and convert the `Device.getHandle()` jlong to `rs2_device*`. Then you use the c-style functions to process those c struct pointers instead of using c++ methods. Well, of course you could convert the rs2_device to rs2::device but be careful the ownership and the reference count of the underlying rs2_device object. As far as I know, `rs2::device` is a `std::shared_ptr` wrapper of `rs2_device *` and the java layer code usually also retains and owns an reference to the handle(i.e. the underlying c struct instance). For more info, please read the source code of rs2::device defintion. The same goes for rs2::pipeline_profile, etc.

Now I think what you need to change is:

```
extern "C"
JNIEXPORT void JNICALL
test_JNI_METHOD(playback)(JNIEnv *env, jclass clazz, jlong device) {
    //rs2::pipeline_profile *pipelineProfile = reinterpret_cast<rs2::pipeline_profile *>(device);
    //pipelineProfile->get_device().as<rs2::playback>().set_real_time(false);
    rs2_pipeline_profile *pipelineProfile = reinterpret_cast<rs2_pipeline_profile *>(device);
    rs2_error *err=nullptr;
    rs2_device * device=rs2_pipeline_profile_get_device(pipelineProfile, &err);
    rs2::error::handle(err);
    bool isPb=rs2_is_device_extendable_to(device,RS2_EXTENSION_PLAYBACK,&err);
    rs2::error::handle(err);
    if(isPb){
	  rs2_playback_device_set_real_time(device,0,&err);
	  rs2::error::handle(err);
    }
}

extern "C"
JNIEXPORT jint JNICALL
test_JNI_METHOD(playbackPosition)(JNIEnv *env, jclass clazz, jlong device) {
    //rs2::device *device1 = reinterpret_cast<rs2::device *>(device);
    //return static_cast<jint>(device1->as<rs2::playback>().get_position());

    rs2_device *device1 = reinterpret_cast<rs2_device *>(device);
    rs2_error* e = nullptr;
    uint64_t pos = rs2_playback_get_position(device1, &e);
    rs2::error::handle(e);
    return (jint)pos;
}
```

MartyG-RealSense, 2020-10-12:
Thanks so much @Mad-Thanos for your very detailed assistance!   :)

closed 2024-08-07 by MartyG-RealSense