realsenseai/librealsense #6757: How to improve L515 depth estimation on black object?

cdb0y511, 2020-07-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (1.04.01) | 
| Operating System & Version |   (Ubuntu 18.04)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.35.2 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I just get the L515.  I know it is based on TOF tech, the ambient light and reflection rate of object will affect the depth estimation.  However, it cant estimate a black chair completely, I don't expect that. Any advice to improve the on black object?
[image]
[image]
[image]

The black chair become invisible at 1 meter.

MartyG-RealSense, 2020-07-05:
Hi @cdb0y511   Depth cameras in general (not just RealSense) have difficulty reading depth from black or dark grey objects because physics dictates that these colors absorb light.  So black / dark grey objects may appear to be rendered on the image but are actually empty areas without depth detail that are shaped like the object.

For example, scanning a black cable may result in a shape on the image that looks like a cable but is actually a cable-shaped area of empty detail with no texture.

In some cases, the visibility of objects may be increased by applying a fine spray-on powder such as baby powder or foot powder to the object, or applying a professional 3D scanning spray.  In situations such as this though (scanning a chair), those methods will not be practical.

Comparing the depth images to the RGB image, it looks as though the areas of the chair that are colored are those where the strong light source nearby is casting light onto those parts of the chair.

cdb0y511, 2020-07-05:
@MartyG-RealSense thanks for quick response.
I wonder if it has a chance to increase the laser power. Ultimately, there are two way to solve this problem. 1. more sensitive  receiver (better photodiode,may be next generation of L5xx). 2. larger laser power.
I think you mean black object is much  more sensitive to ambient light cause the lower SNR.

MartyG-RealSense, 2020-07-05:
Black objects absorb all wavelengths of light and reflect very little or none.  The darker the shade of black, the less light that is reflected.  Googling for **black objects absorb light** provides a lot of references about the subject.  

I would speculate that a more sensitive photodiode and laser laser power would raise the cost of the camera and increase its casing size, reducing two of the positive selling-points of the L515 (small package size and low power usage).

cdb0y511, 2020-07-05:
> Black objects absorb all wavelengths of light and reflect very little or none. The darker the shade of black, the less light that is reflected. Googling for **black objects absorb light** provides a lot of references about the subject.
> 
> I would speculate that a more sensitive photodiode and higher laser power would raise the cost of the camera and increase its casing size, reducing three of the positive selling-points of the L515 (small package size and low power usage at an affordable price).

I see, however black objects are very common. It is  inconvenience and dangerous in many scenario if they are invisible to sensors. I think Intel will figure out eventually, maybe next generation of L5xx. 
I think the best solution is combine D435i (IR projection) with L515 for now.
I will look into the combination of  them, it will be another story. 
Thank you

closed 2020-07-05 by cdb0y511