realsenseai/librealsense #6759: L515 align question , align to color, which resolution is better for reprojection precision?

cdb0y511, 2020-07-05:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (1.04.01) | 
| Operating System & Version |   (Ubuntu 18.04)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.35.2 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
This is an old problem bothers me a long time.
For reprojection precision, which resolution is better ,1920 or 1280?
It is easy to decide on D435i. 
It seems the closest resolution to depth is the suitable one.
However the L515 seems much different from D435i cause one is TOF and another is stereo match.
I never figure out how depth align to RGB in D435i or L515, could any one explain it for me?
And which resolution has the better reprojection precision in slam or reconstruction?
1920 seems to have the same reprojection precision with 1280 at first look.
But how 1024 depth align to 1920 RGB frame? linear interpolation？How to solve the different FOV?
I am really confused. 
Thanks for any advice.

MartyG-RealSense, 2020-07-05:
Hi @cdb0y511   I would recommend trying the **Align** example program, and aligning 1024x768 depth to 1280x720 color.  Let the align process take care of making adjustments to FOV.

https://github.com/IntelRealSense/librealsense/tree/master/examples/align

dnjfk2003, 2020-07-12:
hi @cdb0y511 
I tried to make color_raw and depth_raw aligned each other once with D400 series camera(depth -> color)
Depth FOV is wider than Color's so I croped some area in the depth array and resized it to color resolution and checked color array. After some times of try, I made well aligned arrays.

with python code like bellow
depth_array_crop = depth_array[y_min:ymax, x_min:x_max]
depth_array_aligned = cv2.resize(depth_array_crop, (1280, 720)) 

I don't know how Intel make two frames aligned but just for your information...
Thank you

cdb0y511, 2020-07-12:
> hi @cdb0y511
> I tried to make color_raw and depth_raw aligned each other once with D400 series camera(depth -> color)
> Depth FOV is wider than Color's so I croped some area in the depth array and [...]

Thanks, for sharing. 
I think Intel has done something similar.
Now I get it , to deal the different FOV, crop and resize. Since the depth is interpolated.  It is bad for accuracy in many usages, like slam or reconstruction.

dnjfk2003, 2020-07-12:
@cdb0y511
I got the idea from this image from example of align
[image]
As you can see, the accuracy can be worse because we resize cropped array size bigger

cdb0y511, 2020-07-12:
@dnjfk2003 
The infrared cameras of D435i  have much larger FOV than its RGB camera.
So it seems worse to align rgb to depth.
However, the FOV of l515 or D415 are different from D435i. The infrared camera are not wide angle camera. 
My point is to avoid to resize depth map in any circumstances cause depth is important in my application. 
You can try to align to infrared.  I notice if there is not depth estimation , the color will be missed.

RealSenseSupport, 2020-09-15:
Hi @cdb0y511 

Is there anything else needed on this thread?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-09-22 by RealSenseSupport