realsenseai/librealsense #6760: Potential leftovers from debugging in align-gl.cpp

bmegli, 2020-07-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| SDK Version                            |  2.35.2                              | 

In current master and development branches, in [align-gl.cpp](https://github.com/IntelRealSense/librealsense/blob/master/src/gl/align-gl.cpp#L87), in `align_gl::align_z_to_other`:

```C++
    //...
    render(p, depth, intr, extr, aligned_tex);

    //aligned.get_data();
    aligned = _upload->process(aligned);
    aligned = _upload->process(aligned);
```

Are the last 3 lines intentional?

-------------------------------------------------------------------------------------

By the way - when using `rs2::gl::align`, does user have to do anything specific to get texture data back to CPU (like `glGetTexImage`)? 

Librealsense [gl example](https://github.com/IntelRealSense/librealsense/tree/master/examples/gl) mentions:

> Once get_frame_data() on a GPU-frame gets called frame content will be copied to main memory.

but from what I understand this is not librealsense user facing API.

dorodnic, 2020-07-06:
Hi @bmegli 
This is a work-around for an issue I investigated some time ago. On specific model of GPUs, performing alignment once was not producing valid texture. This is almost certainly due to a bug in our implementation, but I wasn't able to find the root-cause. Assuming this method is performance prohibitive we can re-open it as a bug and look more closely into it.

bmegli, 2020-07-07:
Ok, thanks!

closed 2020-07-07 by bmegli