realsenseai/librealsense #6761: Portability

xiebei, 2020-07-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Operating System & Version |   {Win10}  | 

### Issue Description
I've been testing cameras for some time, We think D415 is available to us, But there's another problem that bothers me.
I used six cameras attached to my desktop's PCIE card, It seems to be working well
Now we want them to be portable，How can I connect them to my laptop？Thunderbolt 3 ？or something other?
by the way, I just want them to run at 6fps.

MartyG-RealSense, 2020-07-06:
A USB hub with 6 or 7 ports that has a short hub-to-PC cable (to help keep cabling tidy) and is unpowered (not connected to mains electricity) is likely the best option for multicam portability on a laptop.

I would be concerned about running 6 cams off a laptop's internal power supply compared to the strong power supply units on a desktop PC.  It may be doable if you are only using 6 FPS though.

xiebei, 2020-07-09:
ok, I used hub, It looks good now.

closed 2020-07-09 by xiebei

MartyG-RealSense, 2020-07-09:
Great news - thanks for the update!