realsenseai/librealsense #6762: How to controll the SR305 shutter?

weseu, 2020-07-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR305 | 
| Firmware Version                   | librealsense 2.32.1 | 
| Operating System & Version |    Linux (Ubuntu 18.04)   | 
| Kernel Version (Linux Only)    |  TX2 Jetpack 4.2.1 | 
| Platform                                 | NVIDIA TX2  |
| SDK Version                            |    | 
| Language                            |  C++  | 

### Issue Description
Because I want to controll the SR305 shutter with synchroflash, but I don't know how to use it....
Please tell me how to do...

by the way...I have found the liberealsense example,but I didn,t find how to controll the shutter.

MartyG-RealSense, 2020-07-06:
Because RealSense camera shutters can not be physically controlled, the usual method of simulating shutter timing control is by manipulating **exposure time**.  

My understanding of 'synchroflash' is that it activates a flash lamp in sync with the opening of a camera's shutter.  Syncing an external flash would likely be easier on a RealSense 400 Series camera (as described in the link below) than with an SR305 that has no external device sync mechanism.

https://github.com/IntelRealSense/librealsense/issues/4574

There was a demonstration in 2018 where Intel did things the other way around, and controlled camera capture by initiating a camera flash and using the spike in camera data caused by the flash as a mechanism to control the start and end of capture.

https://www.intelrealsense.com/intel-realsense-volumetric-capture

weseu, 2020-07-08:
OK~~thank you.
I can't find D435 connector define for strobe synced with the camera.
Is any document can let me referㄡ

MartyG-RealSense, 2020-07-08:
There is a mode that only works on cameras with a global shutter (such as D430, D435 and D435i) where the laser emitter is very rapidly alternated on / off.   The link below has an animated image with a fast-strobing light, so beware if you have epileptic sensitivity.

https://github.com/IntelRealSense/librealsense/pull/3066

weseu, 2020-07-15:
I didn't need to controll laser emitter...
I want to controll RGB shutter and my external flash for RGB image...and I get a response that SR305 & D435i can not let me controll shutter, but I can use the hardware synchronization pins to controll external flash.

Is any hardware synchronization pins define can let me know which pin is shutter sync signal ?

by the way....If SR305 have the shutter sync signal pin, please let me know...

MartyG-RealSense, 2020-07-15:
The SR300 and SR305 do not have pins for sync with external devices.

There was a case I handled 5 years ago where a RealSense user cut open the USB cable of a RealSense F200 (the very similar predecessor of the SR300) to gain access to the electrical pulses passing through the cable.  Doing so requires some electrical skills though.

https://community.intel.com/t5/Software-Archive/Solved-F200-Crashing-When-Opening-Depth-Stream/m-p/1023918/highlight/true#M39327

dorodnic, 2020-07-29:
I'd like to close the issue, since direct control over camera shutter is not available with the SR30X

closed 2020-07-29 by dorodnic