realsenseai/librealsense #6764: Transfer image coordinates of D435i color camera image to real-world coordinate system.

ahasan016, 2020-07-06:
I am using realsense **D435i** camera which is attached in a robot. I somehow want to use any kind of transformation matrix to transform the image coordinate system to real-world coordinate system. For example in the image 
[image]  
I know the four corners of the green boxes and I also know the coordinates in the real world. Now, taking those 8 points as references, I want to calculate all the pixels coordinate in the real world. I have used [solvePnp](https://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html#bool%20solvePnP(InputArray%20objectPoints,%20InputArray%20imagePoints,%20InputArray%20cameraMatrix,%20InputArray%20distCoeffs,%20OutputArray%20rvec,%20OutputArray%20tvec,%20bool%20useExtrinsicGuess,%20int%20flags)) with the 8 pixel points and their known real-world values but the result isn't correct even if i try on those reference points. I need something really close to precise and stable. I will have to calculate the matrices with every frame because my camera is moving with the robot. 

        
```ruby
        pipeline pipe;
	auto config = pipe.start();
	auto profile = config.get_stream(RS2_STREAM_COLOR)
		.as<video_stream_profile>();

	

	Size cropSize;
	if (profile.width() / (float)profile.height() > WHRatio)
	{
		cropSize = Size(static_cast<int>(profile.height() * WHRatio),
			profile.height());
	}
	else
	{
		cropSize = Size(profile.width(),
			static_cast<int>(profile.width() / WHRatio));
	}

	Rect crop(Point((profile.width() - 640) / 2,
		(profile.height() - 480) / 2),
		cropSize);

	rs2_intrinsics intrinsics = profile.get_intrinsics();
	auto principal_point = std::make_pair(intrinsics.ppx, intrinsics.ppy);
	auto focal_length = std::make_pair(intrinsics.fx, intrinsics.fy);
	rs2_distortion model = intrinsics.model;
	cout << profile.width() << endl;
	cout << cropSize.width << endl;

	principle_point_x = intrinsics.ppx * 640.0/profile.width() ;
	principle_point_y = intrinsics.ppy * 480.0/ profile.height()  ;
	focal_length_x = intrinsics.fx;
	focal_length_y = intrinsics.fy;
       
        read_visualize_realsense();
```
```ruby
       void read_visualize_realsense()
	{
	VideoCapture cap("C:/Users/ahasa/OneDrive/Desktop/aruco_test_color_vid/Color-948122072459-2020-7-1_12-30-39.avi");
	cap >> color_mat;

	
	while (!color_mat.empty())
	{

		std::vector<int> ids;
		std::vector<std::vector<cv::Point2f> > corners;
		vector<cv::Point2f> corners_;
		vector<cv::Point2f> corners_1;
		corners_.push_back(cv::Point2f(475, 93)); // corners of 1st square in images
		corners_.push_back(cv::Point2f(506, 93));
		corners_.push_back(cv::Point2f(506, 124));
		corners_.push_back(cv::Point2f(475, 124));

		corners_.push_back(cv::Point2f(33, 100)); // corners of 2nd square in images
		corners_.push_back(cv::Point2f(69, 100));
		corners_.push_back(cv::Point2f(69, 135));
		corners_.push_back(cv::Point2f(33, 135));

		std::vector<cv::Point3f> Real_corners;
		Real_corners.push_back(cv::Point3f(-141, 702, 0)); // corners of 1st square in real world
		Real_corners.push_back(cv::Point3f(29, 702, 0));
		Real_corners.push_back(cv::Point3f(29, 872, 0));
		Real_corners.push_back(cv::Point3f(-141, 872, 0));
		

		Real_corners.push_back(cv::Point3f(2032, 824, 365));// corners of 2nd square in real world
		Real_corners.push_back(cv::Point3f(2202, 824, 365));
		Real_corners.push_back(cv::Point3f(2202, 994, 365));
		Real_corners.push_back(cv::Point3f(2032, 994, 365));

		
		cv::Mat camera_matrix = (cv::Mat_<double>(3, 3) << focal_length_x, 0, principle_point_x, 0, focal_length_y, principle_point_y, 0, 0, 1);
		cv::Mat distCoeffs(4, 1, cv::DataType<double>::type);
		distCoeffs.at<double>(0) = 0;
		distCoeffs.at<double>(1) = 0;
		distCoeffs.at<double>(2) = 0;
		distCoeffs.at<double>(3) = 0;
		cv::Mat rvec(3, 1, cv::DataType<double>::type);
		cv::Mat tvec(3, 1, cv::DataType<double>::type);
		
		cv::solvePnP(Real_corners, corners_, camera_matrix, distCoeffs, rvec, tvec);
		/////cv::solvePnPRansac(Real_corners, corners_, camera_matrix, distCoeffs, rvec, tvec);
		
		Mat rotationMatrix(3, 3, cv::DataType<double>::type);
		Rodrigues(rvec, rotationMatrix);
		
		
		cout << "CM = " << endl << " " << camera_matrix << endl << endl;
		cv::Rodrigues(rvec, rotationMatrix);
		/////cout << "R = " << endl << " " << rotationMatrix << endl << endl;

		cout << "T = " << endl << " " << tvec << endl << endl;

		cout << "CM = " << endl << " " << camera_matrix << endl << endl;

		cv::Mat uvPoint = (cv::Mat_<double>(3, 1) << 33, 100, 1); // u =33, v = 100, ##corners_.push_back(cv::Point2f(475, 93));

		cv::Mat leftSideMat = rotationMatrix.inv() * camera_matrix.inv() * uvPoint;
		cout << "leftSideMat = " << endl << " " << leftSideMat << endl << endl;
		cv::Mat rightSideMat = rotationMatrix.inv() * tvec;
		cout << "rightSideMat = " << endl << " " << rightSideMat << endl << endl;
		double s_ = (365 + rightSideMat.at<double>(2, 0)) / (leftSideMat.at<double>(2, 0)); //365 represents the height Zconst
		
		std::cout << "P = " << rotationMatrix.inv()* (s_* camera_matrix.inv()* uvPoint - tvec) << std::endl;
               
```

 Above are some code snippets of my current approach. Anyone who has already done it or any sample from realsense will be highly appreciated.

MartyG-RealSense, 2020-07-07:
An alternative approach to OpenCV solvepnp for getting the "pose" of objects when using RealSense with a robot arm is to use a ROS technique called **inference**.  An example of this is the **Deep Object Pose Estimation** project.

https://github.com/mintar/Deep_Object_Pose/tree/hybrit

ahasan016, 2020-07-10:
> An alternative approach to OpenCV solvepnp for getting the "pose" of objects when using RealSense with a robot arm is to use a ROS technique called **inference**. An example of this is the **Deep Object Pose Estimation** project.
> 
> https://github.com/mintar/Deep_Object_Pose/tree/hybrit

thanks for your suggestion but I don't want to use ROS or any other deep neural network in my case because I am running another segmentation network. 
Anyway, I guess I have somehow managed to solve my problem with the help of your measurement example.  Haven't really tested in the real-time processing though but will give it a try soon. hope it will work. 

I

MartyG-RealSense, 2020-07-10:
Thanks for the update - please let us know the results of your tests.  Good luck!

ahasan016, 2020-07-10:
> Thanks for the update - please let us know the results of your tests. Good luck!

Thanks. Could you please have a look at my new issue? #6800

MartyG-RealSense, 2020-08-02:
Hi @ahasan016   Do you wish to provide further test results please, or can this case be closed?  Thanks!

closed 2020-08-08 by MartyG-RealSense