realsenseai/librealsense #6765: two-dimensional pixel coordinates to solve the three-dimensional camera coordinates usingmatlab

Jinhui-L, 2020-07-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC.  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {matlab }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
where Is  similar to rs_deproject_pixel_point() in maltab?
Known 2D pixel coordinates，need to know camera  coordinates,but my matlab api have not the function

MartyG-RealSense, 2020-07-06:
Would the MATLAB function **pointsToWorld** meet your needs for projecting 2D image points to 3D world coordinates , please?

https://uk.mathworks.com/matlabcentral/answers/324361-how-to-do-a-manual-calculation-of-worldpoints-from-image-points-with-matlab-computer-vision-data

MartyG-RealSense, 2020-07-12:
If you aligning depth to color to get xyz 3D world coordinates, I wonder if you could instead use the  built-in MATLAB function **pointsToWorld** to project 2D image points to 3D world coordinates?  Would that meet your needs, please?

https://uk.mathworks.com/matlabcentral/answers/324361-how-to-do-a-manual-calculation-of-worldpoints-from-image-points-with-matlab-computer-vision-data

MartyG-RealSense, 2020-07-18:
Hi @Jinhui-L   Do you still require assistance with this case please, or can it be closed?   Thanks!

closed 2020-07-18 by Jinhui-L

MartyG-RealSense, 2020-07-18:
Thanks so much for the update