realsenseai/librealsense #6767: Modifying depth data before using RS functions

Jeff-AB, 2020-07-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |  Ubuntu 20  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.33.1  | 
| Language                            |  python                        | 
| Segment			|  computer vision | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I would like to perform some calculations on the disparity values with opencv and push the changed data back into the frame in python. Is there any way to do this currently?

Something like
```
depth_frame = frames.get_depth_frame()
depth_data = np.asarray(depth_frame.get_data())
depth_data = cv2.erode(depth_data, np.ones((5,5))) # or another opencv function
depth_frame.data = depth_data ## does not work
## Continue processing frame with RS filters 
```

MartyG-RealSense, 2020-07-07:
The link below has a tutorial for generating a disparity map within OpenCV, outside of the RealSense SDK.  Is it any help to you, please?

https://opencv-python-tutroals.readthedocs.io/en/latest/py_tutorials/py_calib3d/py_depthmap/py_depthmap.html

Jeff-AB, 2020-07-07:
I think I may have been unclear. Sorry for the misunderstanding. 

The issue I am having is that any changes I make to the depth data from a frame is not persistent in python. For example, in the snippet above, I use OpenCV's erode function on the depth data, but when I further process the frame, the changes I made with OpenCV disappear. I have tried just reassigning the modified data to the frame, but that raises an error saying that the attribute is not writable.

MartyG-RealSense, 2020-07-07:
I wonder if you need to put the OpenCV data in a format that can be transported back to RealSense, live a cv::mat.  The link below discusses converting a cv::mat to rs::frame.  The discussion centers around C++ rather than Python, but the principles discussed may provide some ideas for a Python version.

https://github.com/IntelRealSense/librealsense/issues/2634

Jeff-AB, 2020-07-07:
How would I allocate a new frame and set all the necessary parameters? In python, it seems like the only way to get frames is to pull them from a device.

MartyG-RealSense, 2020-07-07:
The **frame_source** documentation for pyrealsense2 may be a useful reference for creating and re-generating frames, if you have not seen it already.

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.frame_source.html#pyrealsense2.frame_source

Jeff-AB, 2020-07-07:
It turns out you can modify the frame data directly with numpy without having to generate new frames.

```
data = np.asarray(frame.get_data())
#perform transformations in opencv
data = cv2.dilate(data, np.ones((5,5,)))
# copy back to frame 
np.asarray(frame.get_data())[:,:]  = data
# continue using frame normally
```

closed 2020-07-07 by Jeff-AB

MartyG-RealSense, 2020-07-07:
Great news that you found a solution - thanks very much for sharing it!