realsenseai/librealsense #6769: how to use rs.stream.pose in t265_stereo.py sample

masterchop, 2020-07-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265} | 
| Firmware Version                   | (0.2.0.926) | 
| Operating System & Version |   {Win (10) | 
| Platform                                 | PC |
| SDK Version                            |  { l2.33.1 }                          | 
| Language                            |  {python}                          | 

### Issue Description
As soon i add the following line in the t265_stereo.py sample:

cfg.enable_stream(rs.stream.pose)

the script fails right away.

```

<pyrealsense2.pyrealsense2.pipeline_profile object at 0x000001B54EC45730>
Traceback (most recent call last):
  File "mod_t265_stereo.py", line 147, in <module>
    streams = {"left"  : profiles.get_stream(rs.stream.fisheye, 1).as_video_stream_profile(),
RuntimeError: Profile does not contain the requested stream

C:\Scripts>python mod_t265_stereo.py
Traceback (most recent call last):
  File "mod_t265_stereo.py", line 116, in <module>
    pipe.start(cfg, callback)
RuntimeError: No device connected
```

ev-mp, 2020-07-07:
@masterchop ,
The demo is already configured to generate all the available streams (including `pose`) as elaborated in the [source code](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_stereo.py#L109-L113):
```python
# Build config object and stream everything
cfg = rs.config()
# Start streaming with our callback
pipe.start(cfg, callback) 
```

by injecting that line
>cfg.enable_stream(rs.stream.pose)

you _discard the original settings and instruct the demo to stream `pose` data only_, thus causing the demo to crash when it tries to access the anticipated left/right IR data.
If you want to extract and use the pose data in that demo then you probably need to look at the `callback` function
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_stereo.py#L91-L104

masterchop, 2020-07-09:
Thanks for your update, i tested the following:

```
def callback(frame):
    global frame_data
    if frame.is_frameset():
        frameset = frame.as_frameset()
        f1 = frameset.get_fisheye_frame(1).as_video_frame()
        f2 = frameset.get_fisheye_frame(2).as_video_frame()
        **pose = frame.get_pose_frame()**
        left_data = np.asanyarray(f1.get_data())
        right_data = np.asanyarray(f2.get_data())
        ts = frameset.get_timestamp()
        frame_mutex.acquire()
        frame_data["left"] = left_data
        frame_data["right"] = right_data
        frame_data["timestamp_ms"] = ts
        frame_mutex.release()
```

Just adding that line, froze the code and the windows dont show any color or depth data, it just hangs.

ev-mp, 2020-07-09:
From the context it seem to be a syntax error:
>f2 = frameset.get_fisheye_frame(2).as_video_frame()
        pose = ~~frame~~**frameset**.get_pose_frame()  # <-- added this line to the code only

masterchop, 2020-07-09:
Thanks, that works but in the other sample it says i should query the information like this:

data = pose.get_pose_data()

but that doesnt work how to access this object.

ev-mp, 2020-07-09:
That's correct, these two are separate yet complementary concepts:
1. Extracting the `pose` frame from frameset - the call `pipe.wait_for_frames()` returns a frameset container which needs to iterated over to extract the required frame, as in the code above.
2. Extracting the actual data from the 'pose' frame  - can be achieved by the mentioned `data = pose.get_pose_data()` call.

masterchop, 2020-07-15:
OK my callback has this:

`pose = frameset.get_pose_frame()`

and when i Loop i try to get it like this:

`data = pose.get_pose_data()`

which fails but if do help(pose)
i do get this in retrun:

```
Method resolution order:
 |      pose_frame
 |      frame
 |      pybind11_builtins.pybind11_object
 |      builtins.object
 |
 |  Methods defined here:
 |
 |  __init__(...)
 |      __init__(self: pyrealsense2.pyrealsense2.pose_frame, arg0: pyrealsense2.pyrealsense2.frame) -> None
 |
 |  get_pose_data(...)
 |      get_pose_data(self: pyrealsense2.pyrealsense2.pose_frame) -> pyrealsense2.pyrealsense2.pose
 |
 |      Retrieve the pose data from T2xx position tracking sensor.
 |
 |  ----------------------------------------------------------------------
 |  Data descriptors defined here:
 |
 |  pose_data
 |      Pose data from T2xx position tracking sensor. Identical to calling get_pose_data.
 |
```
I keep getting the error:

RuntimeError: null pointer passed for argument "frame"

07hokage, 2021-01-10:
> Thanks, that works but in the other sample it says i should query the information like this:
> 
> data = pose.get_pose_data()
> 
> but that doesnt work how to access this object.

hey @masterchop , did this work for you??? im not getting error, but im getting pose as None.

masterchop, 2021-05-19:
nop, it doesnt make any sense. i cant believe that its been years and there is not a single example in python.
I am starting to think the wrapper don't work as it should::

For example this is my call back:

`def callback(frame):
    global frame_data
    if frame.is_frameset():
        frameset = frame.as_frameset()
        f1 = frameset.get_fisheye_frame(1).as_video_frame()
        f2 = frameset.get_fisheye_frame(2).as_video_frame()
        left_data = np.asanyarray(f1.get_data())
        right_data = np.asanyarray(f2.get_data())
        pose = frameset.get_pose_frame()
        if pose:            
            print(pose) 
            data = pose.get_pose_data()  

        ts = frameset.get_timestamp()
        frame_mutex.acquire()
        frame_data["left"] = left_data
        frame_data["right"] = right_data
        frame_data["timestamp_ms"] = ts
        frame_mutex.release()`

  If you look at the code is exacly what was requested from me to try but that doesnt  work the code just froze when you do the pose.get_pose_data() I even run a print command into the obj for pose, and this is what i get:
  
<pyrealsense2.pyrealsense2.pose_frame object at 0x000001A15102EDF0>

I open the other sample call t265_example.py which only runs loop of 50  iterations which literary the exact same code:

`        pose = frames.get_pose_frame()
        print(pose)
        if pose:
            # Print some of the pose data to the terminal
            data = pose.get_pose_data()`

ANd if you do a print for the pose object is the exact same object:

<pyrealsense2.pyrealsense2.pose_frame object at 0x00000249C50B8F70>

  So it doesnt make any sense, the python wrapper has to be broken other wise why there isnt a single sample??

masterchop, 2021-05-19:
resolve

closed 2021-05-19 by masterchop