realsenseai/librealsense #6770: DynamicCalibrator - Out of calibration

nik518, 2020-07-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version | Win (8.1/10 | 
| Kernel Version (Linux Only)    | - | 
| Platform                                 | PC |
| SDK Version                            | 2.33.10 | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

Hi, I'm using DynamicCalibrator (v2.11.0.0) in order to calibrate my camera. Every time after RGB scale phase completes I receive following message: "Out of calibration". Does this mean that camera was out of calibration or it is now out of calibration? 
[image]

MartyG-RealSense, 2020-07-07:
My interpretation of the messages on screen, both in the main window and the message log, is that calibration completed successfully and the results were written to the camera.  You can re-run the calibration process as many times as you want to if you feel that the depth quality after calibration is not satisfactory.

nik518, 2020-07-07:
> My interpretation of the messages on screen, both in the main window and the message log, is that calibration completed successfully and the results were written to the camera. You can re-run the calibration process as many times as you want to if you feel that the depth quality after calibration is not satisfactory.
> 
> There is also the additional option of using the newer **On-Chip Calibration** system in the RealSense Viewer, accessible from the **More** option at the top of the Viewer's options side-panel.
> 
> https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras

Thx for a quick response.
Correction on my statement in the original post. 
"Out of calibration" log message appears **occasionally** after RGB calibration finishes.

MartyG-RealSense, 2020-07-07:
I don't think it is a matter for concern, but you can certainly repeat the calibration for your own reassurance.

MartyG-RealSense, 2020-07-18:
Hi @nik518  Do you still require assistance with this case, please?   Thanks!

nik518, 2020-07-20:
Yes, because I would like to know why this is happening..

> Correction on my statement in the original post.
"Out of calibration" log message appears occasionally after RGB calibration finishes.

MartyG-RealSense, 2020-07-20:
I read through the programmer's guide for the calibration tool (the advanced guide for those who want to create their own custom calibrator tool).  I located the following information about the circumstances in which the out of calibration message may be generated:

[image]

The guidance is not clear about whether it is declaring that the calibration overall is a success, or whether it is declaring that it was successfully able to confirm that the device is out of alignment.  My previous advice remains though: if you are not confident in the calibration results, feel free to repeat the process until you are satisfied.

nik518, 2020-07-20:
Yes, I also saw this in documentation.. But as you mention it is not clear about overall calibration status..

MartyG-RealSense, 2020-07-20:
The Dynamic Calibrator program is not open-source and there are no credits for who authored the software or the user guides.  So there is nowhere else to take the enquiry, unfortunately.

closed 2020-08-02 by MartyG-RealSense