realsenseai/librealsense #6771: OpenCV can`t process ROI got from pointcloud.calculate

Kosta404, 2020-07-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  SR300 | 
| Firmware Version                   | 3.15.0.0 | 
| Operating System & Version |   Linux (Ubuntu 18.04)  
| Kernel Version  5.3.0-62-generic 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  python | 
| Segment			|  others   | 

### Issue Description
I cut the ROI from a depth frame, using pointcloud and then process it with OpenCV. But I get the output that sequences, which get ROI as an input are empty.
I`ve already checked that "verts", which are a numpyarray, contain negative values. Is there a some way to process such a ROI? Thank you.

ev-mp, 2020-07-07:
@Kosta404 , the description is not clear, can you elaborate and provide some screeshots/data snippets?
>I cut the ROI from a depth frame, using pointcloud and then process it with OpenCV. 

Do you crop ROI from the original depth or first generate a pointcloud and then apply custom ROI?

>But I get the output that sequences, which get ROI as an input are empty.

It not clear from the context what do you mean by "sequences" and how those are generated.

>I`ve already checked that "verts", which are a numpyarray, contain negative values

The vertices can have negative values for X and Y components  (but not Z) as the 3D origin is the location of the Depth sensor

Kosta404, 2020-07-07:
@ev-mp 
here is the code of cutting the ROI from a depth frame 

[image]

here is the output of the array, which OpenCV can`t process 

[image]

Sorry if the description is not clear, I`m new to such forums.

ev-mp, 2020-07-07:
@Kosta404 , thank you for the additional info,
The SDK code and the generated vertices data (I presume those are the "sequences") are absolutely valid:
- You have several vertices with no depth and consequently no vertices data; denoted as 0,0,0.
- Then you have vertices that point to an object in front of the camera  - If I don't mistake for the axis order it is 13.4cm forward, ~4cm to the right and ~7cm down.

I'm not familiar with `cv2.selecROI`, but from its [documentation](https://kite.com/python/docs/cv2.cv2.selectROI) it is not clear that it is intended to handle 3-dimensional data as an input. 
ROI is well-defined for 2D images. For 3D scene a more frequent term is a rendering/processing frustrum which requires a camera's FOW and near/far clipping ranges to take the camera's perspective into account.
Maybe you want to apply ROI on original Depth frame and then use the cropped data to generate a pointcloud?

Kosta404, 2020-07-07:
> pointcloud

@ev-mp 
I want to cut a ROI on depth frame and do tracking on it. But the only way to cut a ROI on depth frame is a method with a pointcloud. If there is the other way to do it can you please tell? 
And I also want to see object only on a particular range f.e. (from 1.5 m to 3 m.) But the max/min distance option isn`t available on SR300. Can you suggest something.
Thank you.

ev-mp, 2020-07-07:
@Kosta404 , there is no direct analogy for `cv2.SelectROI` as it represents a human-interactive call with mouse picking while the SDK is primarily interacts with HW layer.
You can, however, break it down to a sequence of sub-routines and obtain  same functionality with Librealsense quite intuitively:
1. Present the picture using the available renderer. There are examples how to visualize RealSense depth data with OpenCv in the [wrappers chapter](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv).
2. Select the ROI in the 2D view. Technically you'll get a bounded rectangle (left,right,up,down) .
3. Generate the Pointcloud from the original depth frame (using the code you already have)
4. Extract a sub-buffer from `frame.get_vertices()` bounded by the ROI parameters collected in (2).
5. Run the tracking algorithm...

Kosta404, 2020-07-08:
> @Kosta404 , there is no direct analogy for `cv2.SelectROI` as it represents a human-interactive call with mouse picking while the SDK is primarily interacts with HW layer.
> You can, however, break it [...]

@ev-mp 
I found the way to do what I wanted. Thank you.
But still how can I display objects on a particular distance if preprocessing options "min_distance" and "max_distance" are disabled on SR300?

Kosta404, 2020-07-08:
> > @Kosta404 , there is no direct analogy for `cv2.SelectROI` as it represents a human-interactive call with mouse picking while the SDK is primarily interacts with HW layer.
> > You can, however, break it down to a sequence of sub-routines and obtain same functionality with Librealsense quite intuitively:
> > ```
> > 1. Present the picture using the available renderer. There are examples how to visualize RealSense depth data with OpenCv in the [wrappers chapter](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv).
> > 
> > 2. Select the ROI in the 2D view. Technically you'll get a bounded rectangle (left,right,up,down) .
> > 
> > 3. Generate the Pointcloud from the original depth frame (using the code you already have)
> > 
> > 4. Extract a sub-buffer from `frame.get_vertices()` bounded by the ROI parameters collected in (2).
> > 
> > 5. Run the tracking algorithm...
> > ```
> 
> @ev-mp
> I found the way to do what I wanted. Thank you.
> But still how can I display objects on a particular depth if preprocessing options "min_distance" and "max_distance" are disabled on SR300?

ev-mp, 2020-07-08:
@Kosta404 ,
>And I also want to see object only on a particular range f.e. (from 1.5 m to 3 m.) But the max/min distance option isn`t available on SR300. Can you suggest something?

 **SR**300 is a **S**hort **R**ange camera with [maximal range up to 2 meters](https://ark.intel.com/content/www/us/en/ark/products/92329/intel-realsense-camera-sr300.html) and best results achieved for depth under 1 meter , so it may not fit for that particular use-case. There are other RealSense Depth cameras, such as D415, D435(i), L515 and the recently announced D455 that you can consider.

Kosta404, 2020-07-08:
> @Kosta404 ,
>
>
> **SR**300 is a **S**hort **R**ange camera with [maximal range up to 2 meters](https://ark.intel.com/content/www/us/en/ark/products/92329/intel-realsense-camera-sr300.html) and best [...]

@ev-mp Ok, I`ve got what you mean, but how can I filter objects within this range by deth?

Kosta404, 2020-07-08:
> @Kosta404 ,
>
>
> **SR**300 is a **S**hort **R**ange camera with [maximal range up to 2 meters](https://ark.intel.com/content/www/us/en/ark/products/92329/intel-realsense-camera-sr300.html) and best [...]

thank you for being helpful

closed 2020-07-08 by Kosta404