realsenseai/librealsense #6772: Dynamic Calibration Tool v2-11-0-0 (LINUX)

nik518, 2020-07-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version | Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    | 4.15.0-45-generic | 
| Platform                                 | PC |
| SDK Version                            | 2.33.10 | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

Hi, is there Dynamic Calibration Tool v2-11-0-0 available for Linux?
On official website there is only Windows version available for download and if you follow instructions
for Linux installation from provided documentation it will install Dynamic Calibration Tool v2-6-8-0.

https://downloadcenter.intel.com/download/29618/Intel-RealSense-D400-Series-Dynamic-Calibration-Tool

https://dev.intelrealsense.com/docs/intel-realsensetm-d400-series-calibration-tools-user-guide

dorodnic, 2020-07-09:
Hi @nik518 
I was told version 2.11 was uploaded to the Debian repository. We will update relevant documentation shortly.

nik518, 2020-07-09:
> Hi @nik518
> I was told version 2.11 was uploaded to the Debian repository. We will update relevant documentation shortly.

Ok, thank you.

RealSenseCustomerSupport, 2020-07-09:
----------------------------------------------

Hi

The 2.11.0.0 for Linux is already on AWS, same location as before. The documentation will be uploaded soon.

Here are the simplified steps:

echo 'deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main' | sudo tee /etc/apt/sources.list.d/realsense-public.list
sudo apt-key adv --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE
sudo apt-get update
sudo apt-get install librscalibrationtool
sudo dpkg -L librscalibrationtool

Hope it's helpful.

Thanks

**Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.

nik518, 2020-07-10:
> Hi
> The 2.11.0.0 for Linux is already on AWS, same location as before. The documentation will be uploaded soon.
> Here are the simplified steps:
> echo 'deb [...]

Ok, thank you.

closed 2020-07-15 by dorodnic

JiangweiNEU, 2021-07-12:
> Hi
>
> The 2.11.0.0 for Linux is already on AWS, same location as before. The documentation will be uploaded soon.
>
> Here are the simplified steps:
>
> echo 'deb [...]

hello,
This key "C8B3A55A6F3EFCDE"  cannot be used. We cannot get the calibration tool from the aws.Is there any other way we can get the  Dynamic Calibration Tool v2-11-0-0 available for Linux?
Thanks!
(Executing: /tmp/apt-key-gpghome.2nRpwLBFRl/gpg.1.sh --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE
gpg: Failed to receive from the public key server：No name)

l33tl4bs, 2022-01-18:
replace with:
`sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key C8B3A55A6F3EFCDE`