realsenseai/librealsense #6774: How to downgrade T265 camera firmware?

vitorhirozawa, 2020-07-07:
Hi, I know that t265 firmware is updated using librealsense SDK 2.0.
I updated t265 firmware to version 0.2.0.951 using librealsense v2.35.2.
Now I want to downgrade the t265 firmware because initial orientation is not zero when ROS t265 node is launched.
Before I update the t265 fw, the initial orientation was zero.
So I installed the librealsense v2.34.0 that downloads T265 firmware version 0.2.0.926.
But after run realsense-viewer the fw version of t265 still 0.2.0.951.

Can anyone help me to downgrade t265 firmware? 
Thanks

ev-mp, 2020-07-08:
@vitorhirozawa hello,
The Firmware image for T265 is embedded into the Librealsense SDK dll during compilation and distributed along with it.
To reconfigure Librealsense for a specific FW version one needs to rebuild cmake+make with the following modifications in the source tree:
_./common/fw/firmware-version.h_  - Specify the T265 FW version required
_./common/fw/CMakeLists.txt_  - Adjust the `T26X_FW_SHA1` parameter. (Run cmake once to find out the expected SHA1)

Tip: as T265' FW is persistent till power cutoff it can be used as a temporal w/a to run any supported FW version with an arbitrary Librealsense (useful in testing or one-time demos):
1. Power cycle T265 to discard the loaded FW image.
2. Activate the Librealsense Viewer built with the targeted FW version. For instance, if you want to have FW 0.2.0.926 loaded then use Realsense-Vewer v2.34.0. 
When activated, the Viewer will automatically scan T265 devices and upload the predefined FW.
3.  Close Realsense-Viewer only (don't power off T265) and then run Librealsense version of your choice - the original FW verison will be preserved.

vitorhirozawa, 2020-07-08:
**It looks that the librealsense version can update and downgrade the t265 firmware!** The problem was a missing unplug and replug t265 usb cable on pc or/and a reboot. 
Now t265 has 0.2.0.926 fw version using librealsense v2.34.0.

closed 2020-07-08 by vitorhirozawa