realsenseai/librealsense #6775: Strange dot noisy on L515 depth. Need help

cdb0y511, 2020-07-08:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (1.04.01) | 
| Operating System & Version |   (Ubuntu 18.04)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.35.2 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

I notice there are some strange noisy on L515 depth. Any advice? thanks.
I use low ambient presets. And capture these depth indoor at night with led light on.
 Most of them are good, but some of them have strange dot noisy on it. 
Like these:
[image]
[image]
[image]
And some good ones:
[image]
[image]

blakely3dnm, 2020-07-08:
Does Ragnaros (I think that's who that is) have any shiney optically reflective surfaces? I have noticed that in some cases when I had a car reflector in frame, noisey points with inaccurace depth readings would sometimes occur. Some what I see in your images (1-3) appears similar to what I see when I have a very low reflectivity surface, but in those cases the surface just shows up poorly, rather than introducing much error into the depth measurements (this is probably due to confidence or other filters omitting bad data from those surfaces).

If you are scanning a static object, try using temporal filtering.

cdb0y511, 2020-07-09:
> Does Ragnaros (I think that's who that is) have any shiney optically reflective surfaces? I have noticed that in some cases when I had a car reflector in frame, noisey points with inaccurace depth [...]

Thanks,I don't think there is any reflective object like mirror on Ragnaros. But there are low reflective object behind it , black chairs. I think the latter one is the reason. Any suggestions to revmove these low reflective object? Maybe by confidence adjust?

RealSenseSupport, 2020-07-16:
Hi @cdb0y511 
You can definitely have a look at the controls in the Viewer and make adjustments to them to see what works for your environment and scenario.

[image]

cdb0y511, 2020-07-17:
> Hi @cdb0y511 
> You can definitely have a look at the controls in the Viewer and make adjustments to them to see what works for your environment and scenario.
> 
> 
> [image]
> 

Thanks, I think this is advanced mode of L515. I don't find any examples about how to enable advanced mode of L515 yet. How to manipulate the control settings of L515 in C++?    I think it may need enable advanced mode first. But how?

dorodnic, 2020-07-19:
Hi @cdb0y511 
No need to explicitly enable Advanced Mode. 
You can adjust the relevant settings by setting options:
```
auto res = pipe.start();
res.get_device().first<rs2::depth_sensor>().set_option(RS2_OPTION_CONFIDENCE_THRESHOLD, 1.0f);
```

cdb0y511, 2020-07-19:
> Hi @cdb0y511 
> No need to explicitly enable Advanced Mode. 
> You can adjust the relevant settings by setting options:
> ```
> auto res = pipe.start();
> res.get_device().first<rs2::depth_sensor>().set_option(RS2_OPTION_CONFIDENCE_THRESHOLD, 1.0f);
> ```

Thanks, I find the options in the rs_option.h.
But which is for l515? Some I can guess by name. Is there a document or wiki to describe the l515 options?

cdb0y511, 2020-07-19:
For example, I find Min distance option in l515 control settings.  And RS2_OPTION_MIN_DISTANCE in rs_options.h. Together with RS2_OPTION_MAX_DISTANCE, but Max distance option is not in control settings. So I am confused.

RealSenseSupport, 2020-07-23:
@cdb0y511 

Nothing specific to the L515 options other than some comments in code.  There are some information in the documentation section that may help - https://github.com/IntelRealSense/librealsense/tree/master/doc, https://github.com/IntelRealSense/librealsense/wiki/API-Changes

Also trying to match up options that show up in the Viewer depending on the camera that is plugged into the system.

Hope this helps.

RealSenseSupport, 2020-08-20:
Hi @cdb0y511 

Is there anything else you need on this thread?

L515 documentation links:
https://support.intelrealsense.com/hc/en-us/articles/360051646094-Intel-RealSense-LiDAR-Camera-L515-User-Guide
https://www.intelrealsense.com/optimizing-the-lidar-camera-l515-range/

*If we don’t hear from you in 7 days, this issue will be closed.

cdb0y511, 2020-08-21:
Thanks, there is still an odd phenomenon. I notice there is a white dot moving in the depth frame occasionally. It looks like white snow moving from left to right when I scan a static object. I suspect if it is the mems part goes wrong, or just merely dust moving in front of the sensor? It seems to happen more often when the sensor temperature goes up.

cdb0y511, 2020-09-01:
Still, there is some strange fly dot noise in the depth frame. It is hard to capture. I wonder if it is related to my optical measurement platform cause it is a reflective metal.
Any advice will be grateful.
Thanks

RealSenseSupport, 2020-09-15:
Hi @cdb0y511 

Do you by chance have the ability to capture without the reflective metal platform?  The reflective metal could influence your depth capture.

cdb0y511, 2020-09-16:
@RealSenseSupport 
Hi, this time I cover every reflective thing. The Ghost Nosie is still coming. 
It random appears and disappears, moves frome one side to another.
It influences the infrared,depth and confidence map. 
It looks like a black hole in the left side. 
I guess it may relate to the APD casue the infrared (intensity) does not have value on that part.
confidence:

[image]

infrared:

[image]

depth:

[image]

rgb:

[image]

cdb0y511, 2020-09-16:
> Hi @cdb0y511
> 
> Do you by chance have the ability to capture without the reflective metal platform? The reflective metal could influence your depth capture.

It appears less than 1s, so I guess 10-20 frames, moves from left to right.
And it seems to be more often when the temperature increase.

cdb0y511, 2020-09-17:
> Thanks for the information @cdb0y511 
> 
> Looking at the infrared image and the dot or black hole appears, that means that either the laser is blocked at the point or the APD is blocked at that point.  Have you tried wiping the L515 cover glass and see if the dots show up again?  Or is there dust or something on the platform that needs to be wiped off?

It moves, then disapears in few frames. Maybe dust in the air?

RealSenseSupport, 2020-09-23:
@cdb0y511 

Could be dust in the air but it is also something that seems to be inherent in the camera that you have since you say you are seeing it on a regular basis.

Did you by chance order the camera through our webstore on intelrealsense.com or through another channel?

cdb0y511, 2020-09-24:
> @cdb0y511
> 
> Could be dust in the air but it is also something that seems to be inherent in the camera that you have since you say you are seeing it on a regular basis.
> 
> Did you by chance order the camera through our webstore on intelrealsense.com or through another channel?

Hi, I buy it from the Chinese distributor of Intel. https://www.intelrealsense.com/zh-hans/buy/
Should I contact them?

closed 2020-09-28 by cdb0y511