realsenseai/librealsense #6778: Json file stream parameter

younussenn, 2020-07-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | v2.33.1 | 
| Operating System & Version |  Win (10)  | 
| Platform                                 | PC |
| SDK Version                            |  { 2 }                          | 
| Language                            |  {C# }                          | 
| Segment			|  {others }                          | 

### Issue Description

Hi,

Stream parameters in json file are not assigned. How can i set the depth frame information?

var pipeline = new Pipeline();
var profile = pipeline.Start();
var adv = AdvancedDevice.FromDevice(profile.Device);
adv.JsonConfiguration = File.ReadAllText("1.json");

" adv.JsonConfiguration "  doesn't contain stream information.

MartyG-RealSense, 2020-07-08:
Hi @younussenn   You can define streams with **cfg** (config) statements placed before the pipeline start instruction.  I hope that the example script in the link below for defining a stream configuration in C# will be helpful.

https://github.com/IntelRealSense/librealsense/issues/1747

younussenn, 2020-07-08:
Thank you for your quick reply. Previously, I started the pipeline with cfg, but when I read from the json file, I removed it. Thanks again.

closed 2020-07-08 by younussenn