realsenseai/librealsense #6780: "Resource temporarily unavailable" on NVIDIA Jetson TX2

rcroset, 2020-07-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Operating System & Version |  Ubuntu 18.04.4  | 
| Kernel Version (Linux Only)    | 4.9.140-tegra         | 
| Platform                                 | NVIDIA Jetson TX2  |
| SDK Version                            |3.35.2 }                          | 
| Language                            |  python| 
| Segment			| other| 

### Issue Description
I'm currently working on a system involving 3 D435 cameras connected to a NVIDIA Jetson TX2 platform via a USB3 hub. Quite often when trying to access the cameras, I get a "Resource temporarily unavailable" from one of the camera and I can't access the device that throws this error, even in the RealSense Viewer. It happens quite randomly, on any of the cameras. To get things back on track, I have to shutdown the TX2 platform and reconnect the USB hub manually. Is there any way to do this (programmatically or other) without having to manually reconnect things ? Since this system will soon go to production, it won't be easy to manually fix things up. Any help or suggestion on how to get rid of this issue ? Many thanks :)

MartyG-RealSense, 2020-07-08:
Hi @rcroset   You can add a **hardware_reset()** routine to perform an automated hardware reset of the camera that has a similar effect to physically unplugging and re-plugging the camera in the USB port.  The link below has example scripting for doing so with Python and multiple cameras using the camera serial numbers:

https://github.com/IntelRealSense/librealsense/issues/5428

rcroset, 2020-07-09:
I unfortunately can't perform the hardware_reset() routine because I cannot access the device. When I want to query a device from the context, I get the error (see below):
```>>> import pyrealsense2 as rs2
>>> ctx = rs2.context()
>>> list = ctx.query_devices()
>>> list[0]
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: set_xu(ctrl=1) failed! Last Error: Resource temporarily unavailable
```
 The CPU usage of the TX2 is quite high (but still not near 100%) when running my project, but even when running the snippet above the error occurs.

MartyG-RealSense, 2020-07-09:
@tispratik had a similar situation with this error when using Python, and doing a hardware_reset() did not correct the problem for them either.

https://github.com/IntelRealSense/librealsense/issues/6132

rcroset, 2020-07-09:
Thank you for pointing this link. Unfortunately, it doesn't seem to provide a solution. Rebooting the system is not enough in our case.

MartyG-RealSense, 2020-07-09:
In past cases where doing a hardware_reset() on the camera has not been practical due to detection problems and you cannot reset the computer, there has been the possibility of achieving the same effect by using a USB port reset script with Ubuntu    Please google for **ubuntu usb port reset script** for more details.

rcroset, 2020-07-09:
I've already tried that kind of script, unsuccessfully. It appears that only a manual reconnection works.

MartyG-RealSense, 2020-07-09:
Thanks for your patience.  I have been considering further possibilities.  To aid that analysis, could you tell me please if the project is being used outdoors, with the cameras exposed to sunlight?

rcroset, 2020-07-09:
The project is used indoor. There are a small window providing sunlight but the cameras are not directly exposed (i.e. they are on the same wall as the window).

MartyG-RealSense, 2020-07-09:
Okay, thank you very much.  My reading of your situation is that 3 cameras are attached to one TX2 board.  And when one camera stops responding, the other two remain accessible.  Is that correct, please?

What FPS speed are the cameras running at, please?

rcroset, 2020-07-09:
The 3 cameras are actually connected to a USB3 hub connected to the TX2 board. When a camera stops responding, it also shuts down the cameras plugged below on the hub (e.g. when the camera plugged on the USB slot 2 of the hub stops responding, it also affect the one plugged in slot 3). So when the first cameras stops responding, it blocks all cameras.

The cameras are running at 6 FPS with maximum resolution for depth and color stream.

MartyG-RealSense, 2020-07-09:
6 FPS is a speed that is prone to errors that disappear at 15 FPS.  Would it be possible to try 15 FPS and see if you continue to experience the problems, please?

rcroset, 2020-07-10:
Ok I'll try this and I'll get back to you in a few days to inform you if the problem comes back.

How is it possible that low FPS can introduce such errors that disappear at higher framerate ?

MartyG-RealSense, 2020-07-10:
I believe it is because when the FPS is updating slowly, it can lead to timeouts while waiting for frames to arrive.

rcroset, 2020-07-16:
We've been running at 15 FPS for almost one week and this problem didn't occur (yet). But a new problem appeared. The depth frames seemed not to update anymore. They update every minute or so. I've tried using `wait_for_frames` and `poll_for_frames` and nothing changes. When I switch back to 6 FPS, everything is back to normal. How is this possible ?

MartyG-RealSense, 2020-07-16:
As you are using Python, how you are storing the frames could be a factor.  Like in the case below:

https://github.com/IntelRealSense/librealsense/issues/946

rcroset, 2020-07-16:
Thanks for this link. However, I am not storing the frames in an array. All post-processing is done on a copy of the frame, and the variable containing the original frame is overridden at each iteration of the loop.

MartyG-RealSense, 2020-07-16:
My understanding is that when a frame is modified (for example, by a post-processing filter), the original is not destroyed but instead a copy of the frame is **automatically** created.  Every frame has a counter stating how many copies of the same frame are being held in the frameset (which is a collection of frame objects).   

Old frames get pushed out when a new frame enters the frame queue, and waitforframes() pulls a frame from the queue.

rcroset, 2020-07-16:
Indeed, but the original is destroyed and overridden by the return value of `wait_for_frames`. How to access this copy counter?

Is it possible to have an example on how to properly use `wait_for_frames` or `poll_for_frames` for multicam in Python? I can't find any.

MartyG-RealSense, 2020-07-16:
Remember from earlier in this case that for multicam projects, it is recommended that **poll_for_frames()** is used instead of **wait_for_frames()**.  This includes programs that do not use multicam hardware sync, such as **rs-multicam**.

https://github.com/IntelRealSense/librealsense/issues/6780#issuecomment-655517695

I could not find a short and neat example of multicam use of poll_for_frames() in Python, though a device manager script in the multi-camera **box_dimensioner_multicam** example program makes use of it:

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/realsense_device_manager.py#L208

You can read the **frame counter** from metadata with Python, though I'm not sure that is what you were asking for.

https://github.com/IntelRealSense/librealsense/issues/3179#issuecomment-462093873

rcroset, 2020-07-16:
I remember, thanks :)
I tried to do the same thing as in the box_dimensioner_multicam example, but still, the depth frames do not update. The color frames are fine. And when switching back to 6 FPS, the problem disappears.

MartyG-RealSense, 2020-07-16:
I apologise for the limitations of my Python programming knowledge, which may be slowing this diagnostic process.

Some Python users who have multiple streams active have found that it can help to separate the stream types into different pipelines.  The Python script in the link below separates RG / depth in one pipeline and IMU in the other.  Whilst you are not using IMU, conceivably you could put RGB in one pipeline and depth in the other pipeline.

https://github.com/IntelRealSense/librealsense/issues/6773#issuecomment-655364659

rcroset, 2020-07-16:
Thanks for the link and the idea but I'm not sure this will suit our project. We need to gather color frames and the corresponding depth frames to be able to look at them at the same time. Will those two pipelines be perfectly synchronous with each other?

MartyG-RealSense, 2020-07-16:
I would think so.  Librealsense can freely pass data between threads for purposes such as having a different processing pipeline for each stream type.

https://dev.intelrealsense.com/docs/frame-management#section-frames-and-threads

rcroset, 2020-07-16:
Thanks, I will try this. Do you recommend continuing to use pipelines or switching to something else like a syncer ?

MartyG-RealSense, 2020-07-16:
My understanding is that **syncer** is useful for synchronizing between different streams (e.g all sensors across all cameras).  So it may be appropriate for your three-camera project.

rcroset, 2020-07-17:
I'm trying to use a syncer, but I have a strange behavior. Depth frames are always empty (i.e. I get the following RuntimeError: `null pointer passed for argument "frame_ref"`) on multicam. If I'm working with only one camera, it works fine. Is there any good practices or things to do when doing multicam with a syncer ?

MartyG-RealSense, 2020-07-17:
This case sounds similar to what you are experiencing with this error in multicam:

https://github.com/IntelRealSense/librealsense/issues/2656

rcroset, 2020-07-17:
Thanks for the link. However, it doesn't solve my problem. If I want to ensure that the size of the frameset coming from the syncer is 2 (i.e. color frame + depth frame), it takes approximately 50 seconds per syncer, which is way too long. Even if I use one syncer per  stream instead of one syncer per camera, it still takes way too long for gathering the two frames. From what I saw, the `poll_for_frames` returns a frameset containing only one frame (either the color or the depth one) but rarely the two frames in one frameset.

This problem occurs only at a framerate of 15 FPS or more. At 6 FPS everything works fine, but the initial error from this thread occurs again (I got one `Resource temporarily unavailable` error this morning while testing at 6 FPS and I had to reboot everything and reconnect the cameras to make it disappear)

MartyG-RealSense, 2020-07-17:
Is the any depth to color alignment being done in your project?

rcroset, 2020-07-17:
Yes, depth to color alignment. But in the tests I've done for using the syncers, I didn't do the alignment

MartyG-RealSense, 2020-07-17:
Ok, the reason I asked is that if the CMake build flag **BUILD_WITH_OPENMP** is set to True when building Librealsense then multiple processor cores can be utilized during depth to color spatial alignment.  This reduces latency, at the expense of higher processor usage.

rcroset, 2020-07-17:
So you recommend setting this flag to True ?

MartyG-RealSense, 2020-07-17:
If you are going to be using depth to color alignment and have a powerful computer board like the TX2 then it may certainly be worth trying in a future version of the program.  If the benefits of the flag are limited to depth to color (and YUY to RGB conversion, which the flag can also assist) though then it may not be worth trying right now with your current test program.

rcroset, 2020-07-17:
Alright, I'll try it in a few weeks, I cannot do it right now.
In the mean time, any suggestions on how to make those syncers (or pipelines) work at 15 FPS (recall that everything works fine at 6 FPS) ? Or on how to remove this `Resource temporarily unavailable` error without reconnecting the cameras manually ?

MartyG-RealSense, 2020-07-17:
Some multicam problems may occur simply because of the particular model of USB hub being used.  Though Intel recently introduced improvements in SDK **2.35.2** for how multicam is handled, these improvements do not help fix problems if the issue is related to a specific USB hub model.

Intel has successfully tested RealSense 400 Series cameras with the **AmazonBasics** mains-powered USB 3 hub with up to 4 cameras attached to the hub.  I have one myself that I bought based on this recommendation and have had no problems with it.  The link below references the subject of the hub.

https://github.com/IntelRealSense/realsense-ros/issues/1184#issuecomment-620466170

Checking through the case from the start, I notice that the subject of USB cable lengths has not been addressed yet.   Are you using the official short cables supplied with the camera or a longer cable of your own choice, please?

rcroset, 2020-07-17:
Thanks, I'll see if we can update the SDK soon.

I don't use the official cables because they are too short for the purpose of our project. I'm using longer cables. When I tested them (not multicam, one camera at the time) everything worked fine.

A weird thing is that in realsense-viewer, with the multicam, color and depth frames are coming correctly, at any framerate with all cameras streaming. The problem occur only in code

MartyG-RealSense, 2020-07-17:
Because RealSense can transmit a lot of data through the USB cable, using a high quality cable is important with cable lengths of 2 meters or more, otherwise performance deterioration or disconnections can occur.  Intel's data sheet document for the 400 Series cameras officially recommends the high quality USB equipment supplier **Newnex** as a purchasing source for USB cables.

https://www.newnex.com/realsense-3d-camera-connectivity.php

Having said that, your success with your own choice of cable when using the camera on its own would suggest that your cable is okay.

When creating your own program, differences in the running of that program compared to the results of the RealSense Viewer may differ because the RealSense Viewer may have post-processing filters enabled.  In self-created programs, filters have to be deliberately programmed in.

You can check whether post-processing filters are being applied by default in your Viewer if you expand open the **Stereo Module** side-panel options, scroll down to the **Post Processing** option and see whether it has a blue colored icon (Enabled) beside it.

[image]

rcroset, 2020-07-17:
Post Processing is enabled in RealSense Viewer (as well as all filters) and everything works fine. In my program it doesn't work even when not doing any post processing at all. Just gathering the frames with `poll_for_frames` takes ages to get both depth and color frames together.

MartyG-RealSense, 2020-07-17:
There is a detailed discussion here about how the RealSense Viewer processes streams:

https://github.com/IntelRealSense/librealsense/issues/3836

As the Viewer program is open-source, RealSense users are welcome to use it as a basis to adapt into a custom viewer to build their own application around.

https://github.com/IntelRealSense/librealsense/tree/master/tools/realsense-viewer

Having got as far as you have with your own program, that works fine with single cameras at least, I can understand that you would want to continue with your own solution.

rcroset, 2020-07-17:
Thanks for those informations.

The frustrating part is that my solution works fins at 6 FPS (besides the resource unavailable error) and not at all at 15 FPS

MartyG-RealSense, 2020-07-17:
An approach that Intel have tested with multicam at 6 FPS is to connect the multiple cameras to a central computer with ethernet networking, using a tool called **rs-server**.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

rcroset, 2020-07-17:
Thanks, I'll have a look at it. But ideally, for this project we'd like to stick to USB only and not using another device as a central computer and avoid networking

MartyG-RealSense, 2020-07-17:
I am starting to wonder if it is a Linux kernel related issue.  If it is, building librealsense from source using the **RSUSB backend** installation method may resolve it.  It has a tendency to clear up weird little issues.  For example, one Linux user this week had a green line on the corner of their images that wasn't present on the Windows version of the SDK, and building with backend fixed it.

I also saw a case where the user was testing streaming at 6 and 15 FPS.  they could get frames at 15, but could barely get frames at 6 FPS at all, and only if they used 1280x720 resolution.

https://github.com/IntelRealSense/librealsense/issues/2771

rcroset, 2020-07-17:
Fine, will try it probably next week. Thanks!

This issue with the Xavier seems to be the opposite of mine. He has troubles with 6 FPS and I have issues with 15 ;) Unfortunately, this thread doesn't seem to give any solutions

MartyG-RealSense, 2020-07-17:
Okay, do you want to take a break from it for the weekend and come back to it next week after you have had the opportunity to test some of the things discussed?  Thanks!

rcroset, 2020-07-17:
Sure, I won't be able to try anything during the weekend and I have to check with my supervisor before trying to rebuild the library with those flags. I will come back as soon as possible when some things have been tested. Thanks for your help!

MartyG-RealSense, 2020-08-02:
Hi @rcroset   Do you require futher assistance please, or can this case be closed now?  Thanks!

rcroset, 2020-08-06:
Hi! Yes I'll need further assistance soon. I still didn't had the opportunity to try those things we talked about. I'll come back to you as soon as possible in the coming weeks to discuss the results of those tests.

MartyG-RealSense, 2020-08-06:
Okay, thanks very much for the update.  I will keep the case open for the next month.   Please do keep us updated about your progress in the coming weeks.   Thanks!

rcroset, 2020-08-18:
Hi! So we've tried building librealsense from source using the **RSUSB backend** as you suggested but the problems are still here, i.e. at 6 FPS the camera crash (resource temporarily unavailable) and at 15 FPS the depth frames don't update at the correct rate (once every minute or so). Any suggestions?

MartyG-RealSense, 2020-08-18:
We have already established that using **poll_for_frames()** is not a fix for the problems, and that getting the color and depth frames together takes "ages".  I don't think you said whether you were also putting the CPU to sleep though.

With poll_for_frames, you have to manually control when the CPU is put to sleep, otherwise you end up "maxing out" the CPU.  You said that !he CPU usage of the TX2 is quite high (but still not near 100%) when running my project".

I wonder if poll_for_frames() is still the solution and it is the CPU sleep period value that needs adjusting for it to work properly.

MartyG-RealSense, 2020-08-27:
Hi @rcroset   Do you still require assistance with this case, please?   Thanks!

rcroset, 2020-08-31:
Hi @MartyG-RealSense ! We are still trying some things in order to make everything working. We still have some issues and things to tests. I'll come back to you when we're done to tell you if we solved this case or not.

MartyG-RealSense, 2020-08-31:
Okay, no problem.  Thanks for the update!

rcroset, 2020-09-02:
Hi @MartyG-RealSense ! Here are some news about those issues.

We've rebuilt the whole library using OpenMP and RSUSB backend and our problems evolved.
At 15 FPS, we had issues to get synchronized color and depth frames (the color frames were up to date but the depth frames updated once every minute or so). Now we have synchronized frames and we're very glad about this (using both `wait_for_frames` or `poll_for_frames`).
With `wait_for_frames` the "Resource temporarily unavailable" still occur and requires a manual intervention to reboot/restart everything as before.
With `poll_for_frames` and putting the CPU to sleep as you suggested, this error disappears BUT sometimes it fails to get both color and depth frames, so the resulting frameset contains only one frame, which is unusable for our project. When this situation occurs, it lasts forever until we kill the program and restart it, which is not very convenient. We've tried to detect this situation and restart the faulty pipelines programmatically but it raises a "Failed to set power state" error.
Fun fact, this problem seems to affect all cameras at the same time, so it might be related to USB. 

Any suggestions ?

MartyG-RealSense, 2020-09-02:
If **wait_for_frames()** is the ideal but is prone to error, I wonder if you could use **try_wait_for_frames()** instead.  This blocks until a frame is available like wait_for_frames() does, but returns a False state if timeout occurs instead of throwing an error.

https://github.com/IntelRealSense/librealsense/issues/2422#issuecomment-423254709

rcroset, 2020-09-02:
Isn't there a risk of still having the Resource temporarily unavailable error with `try_wait_for_frames()`, like as with `wait_for_frames()` ?

MartyG-RealSense, 2020-09-02:
I am not personally aware of whether it blocks 'resource temporarily available' being thrown and returns a 'False' logic state instead.  It's probably a quick tweak to find out one way or the other though.

rcroset, 2020-09-02:
Alright, I'll try with this function. I'll come back to you soon to tell you if it worked well !

rcroset, 2020-09-02:
So I've just tried using `try_poll_for_frames` the results are not as good as expected. Fortunately, no "Resource temporarily unavailable" error, which is a good point, but after 20 minutes this method was unable to gather frames anymore (i.e. it continuously returns 'False'). This lead to a USB disconnection as I mentioned before.
What should I do to avoid this and keep my pipelines open and healthy ?

[22 more replies]

closed 2021-01-12 by MartyG-RealSense