realsenseai/librealsense #6781: How can i apply post-processing filter in c#?

younussenn, 2020-07-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | v2.33.1 | 
| Operating System & Version |  Win (10)  | 
| Platform                                 | PC |
| SDK Version                            |  { 2 }                          | 
| Language                            |  {C# }                          | 
| Segment			|  {others }                          | 

### Issue Description

Hi,

There is too much noise in the depth view at a long distance. So I want to use post-processing filters. Is there an example for this? Also, is there a json configuration where we can get accurate results in long distance?

MartyG-RealSense, 2020-07-08:
There is a C# example program in the SDK that can use processing blocks to do post-processing and align.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/cs-tutorial-3-processing

Also, a RealSense user on the Intel Support RealSense forum wrote a script for the **threshold filter** in C#.  This is a filter that lets you set a minimum and maximum range for depth sensing, with data outside of that range being excluded from the image.

https://support.intelrealsense.com/hc/en-us/community/posts/360039445773/comments/360010898533

I do not know of a json configuration for improving accuracy over long range.  Part of the difficulty with this is that on the 400 Series cameras, the error rate increases linearly over distance, becoming greater as the distance of the observed object from the camera increases.  This is called **RMS Error**.

Intel have addressed this problem in the newest **RealSense D455** model, which is available to pre-order now.  It has better accuracy over distance than the D435 models by a factor of two.  This means that it has an optimal depth sensing distance of 6 meters, whereas on the D435 models, accuracy starts drifting noticeably after 3 meters.

https://github.com/IntelRealSense/librealsense/issues/6610

Another way to improve results over long distance may be to purchase a physical optical filter and apply it over the lens of the camera.  Intel have published a white paper on the subject of optical filters.

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400

For example, **Section 4.2.1** of the paper discusses using a **long-pass filter** for making a plain white wall in the distance more visible by increasing the relative brightness of the IR projector.

MartyG-RealSense, 2020-07-18:
Hi @younussenn   Do you still require assistance with this case, please?   Thanks!

younussenn, 2020-07-20:
Thank you so much for helping.

closed 2020-07-20 by younussenn