realsenseai/librealsense #6784: Can't compile C++ examples code

milan-r-shah, 2020-07-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |   Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    | 5.3.0-62-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.0                          | 
| Language                            |  C++                         | 
| Segment			|  others                           | 

### Issue Description
I just cloned the repo and tried to run examples/hello-realsense/rs-hello-realsense.cpp file by using following command:
```
cd ~/librealsense/examples/hello-realsense
mkdir build
cd build
cmake ..
```
but it's giving the following error:
```
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:14 (install):
  install TARGETS given no RUNTIME DESTINATION for executable target
  "rs-hello-realsense".

-- Configuring incomplete, errors occurred!
See also "/home/milan/librealsense/examples/hello-realsense/build/CMakeFiles/CMakeOutput.log".
```

What should I do to resolve this issue?

Thanks!

MartyG-RealSense, 2020-07-08:
Please navigate to the build directory and try running this instruction:

cmake .. -DBUILD_EXAMPLES=true

milan-r-shah, 2020-07-08:
> Please navigate to the librealsense > builld directory with **cd build** after having created the build directory with **mkdir build** and try running this instruction:
> 
> **cmake .. -DBUILD_EXAMPLES=true**

It's still giving the same error!

MartyG-RealSense, 2020-07-08:
You can the instruction from **librealsense > build** ?

milan-r-shah, 2020-07-08:
Oh, no. Nope, I had run from **librealsense > examples > hello-realsense > build** !!

When I ran **cmake .. -DBUILD_EXAMPLES=true** in **librealsense > build**, it worked but then it's giving me following errors when I run **make** : 
```
.
.
.
Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    ANDROID_ABI
    ANDROID_STL
    CMAKE_CXX_STANDARD_LIBRARIES

-- Build files have been written to: /home/milan/librealsense/build/libusb-prefix/src/libusb-build
[  2%] Performing build step for 'libusb'
Scanning dependencies of target usb
[  9%] Building C object CMakeFiles/usb.dir/libusb/core.c.o
/home/milan/librealsense/build/third-party/libusb/libusb/core.c:23:10: fatal error:config.h: No such file or directory
 #include "config.h"
          ^~~~~~~~~~
compilation terminated.
CMakeFiles/usb.dir/build.make:62: recipe for target 'CMakeFiles/usb.dir/libusb/core.c.o' failed
make[5]: *** [CMakeFiles/usb.dir/libusb/core.c.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/usb.dir/all' failed
make[4]: *** [CMakeFiles/usb.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/libusb.dir/build.make:111: recipe for target 'libusb-prefix/src/libusb-stamp/libusb-build' failed
make[2]: *** [libusb-prefix/src/libusb-stamp/libusb-build] Error 2
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/libusb.dir/all' failed
make[1]: *** [CMakeFiles/libusb.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

Also, is there any way to compile only 1 program individually?

Thanks again!

MartyG-RealSense, 2020-07-08:
Advice about modifying individual programs is given here:

https://github.com/IntelRealSense/librealsense/issues/2946#issuecomment-449290237

MartyG-RealSense, 2020-07-08:
I found a case where the same **Makefile:129: recipe for target 'all' failed** error had occurred when trying to build the examples. This was the advice I gave:

https://community.intel.com/t5/Items-with-no-label/How-to-compile-librealsense-example/m-p/645304/highlight/true#M14685

milan-r-shah, 2020-07-08:
> I found another case where the same **Makefile:129: recipe for target 'all' failed** error had occurred when trying to build the examples. This was the advice I gave:
> 
> https://community.intel.com/t5/Items-with-no-label/How-to-compile-librealsense-example/m-p/645304/highlight/true#M14685

Thanks @MartyG-RealSense for sharing this link and I also tried ```cmake .. -DBUILD_EXAMPLES=true && make && sudo make install```

However, it didn't work, too. Getting the same error!

MartyG-RealSense, 2020-07-08:
If all else fails, it can be easiest to use the **backend** method of installation from source, which requires an internet connection but is not dependent on Linux versions or kernel versions and does not require patching.  It also automatically includes support for metadata so that you do not have to patch that in either.

https://github.com/IntelRealSense/librealsense/issues/6710#issuecomment-650611442

ev-mp, 2020-07-09:
@milan-r-shah , this error -  
>/home/milan/librealsense/build/third-party/libusb/libusb/core.c:23:10: fatal error:config.h: No such file or directory
 #include "config.h"

was previously found indicative of a missing dependency from the installation guide  (libusb-1.0-0-dev) - https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites

See #5518

MartyG-RealSense, 2020-07-18:
Hi @milan-r-shah   Do you still require assistance with this case, please?   Thanks!

milan-r-shah, 2020-07-20:
Hi @MartyG-RealSense ,

Sincere apology for the late reply. Yes, I'm able to compile the example codes now. 

However, first I had tried installing the SDK using package files ([distribution_linux.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md)) and as it didn't work, then I tried installing the SDK from source code ([installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md)). Therefore I think in doing so, now **realsense-viewer** is giving the following error/warning, upon launching:

> Multiple realsense udev-rules were found! :
> 1:/etc/udev/rules.d/99-realsense-libusb.rules
> 2: /lib/udev/rules.d/60-librealsense2-udev-rules.rules

Could you please let me know how can I fix this? 
Thank you!

MartyG-RealSense, 2020-07-20:
The advice in this situation is that it is okay to remove the rule in the **/etc** folder but not the other one.under the **lib/udev** folder.

https://github.com/IntelRealSense/librealsense/issues/6153#issuecomment-605687547

milan-r-shah, 2020-07-20:
Thanks @MartyG-RealSense for such a quick response. Yup, after deleting `/etc/udev/rules.d/99-realsense-libusb.rules`, that error/warning disappeared.

However, at the same time, there was a firmware for D435i. So, I tried that but it showed: Update failed (automatically)! On the other hand, after restarting everything, I can see that it has changed from **05.12.05.00** to **05.12.06.00**
Now, when I launch realsense-viewer, it's giving me multiple different errors. In fact, I can't see the camera output; it says: "No Frames Received!"

Is this because I deleted that rules file from `/etc/...` or should I create a new issue?
Thanks!

milan-r-shah, 2020-07-20:
Also, on the top right corner, I keep getting errors and their count keep increasing as shown in the screenshot below:

[image]

MartyG-RealSense, 2020-07-21:
I suggest first trying a firmware diwngrade.  Please download the firmware file for version 5.12.5 from the firmware releases page.

https://dev.intelrealsense.com/docs/firmware-releases

Extract the content of the zip file.  Then go to the Viewer and from the menu of the **More** option at the top of the Viewer's options side panel, select the option to update firmware (not the one to update to recmmended firmware).  This option will ask you to select the firmware .bin file that you extracted from the zip.   It should then change the firmware to 5.12.5 (the previous version) and hopefully fix your problems, including the USB 2.1 state.

milan-r-shah, 2020-07-21:
Thanks @MartyG-RealSense for your reply. 

I followed the steps you mentioned above to downgrade the firmware from 05.12.06.00 to 05.12.05.00. So, in *realsense-viewer*, for **Firmware Version**, it's showing **05.12.05.00**. However, for **Min FW Version**, it's still showing **05.12.06.00**. So, is this normal or do I have to follow some other steps as well?

Also, as of now, in the application window of *realsense-viewer*, everything seems OK but in the terminal, I'm getting the following errors & warnings:
[image]

MartyG-RealSense, 2020-07-21:
The minimum firmware recommended changes over time as new SDK versions and new firmwares are released.  Assuming that your earlier firmware problems were due to the firmware update failure, you could try to re-upgrade to firmware 5.12.6 by selecting from the **More** option's menu the option to update to **recommended** firmware.

milan-r-shah, 2020-07-21:
Thanks @MartyG-RealSense for such a quick response.

I tried upgrading the firmware to 05.12.06.00, as you said. It also worked fine for some time. However, afterward, it again started giving me the same errors that I was getting before (I don't know why)! In fact, I unplugged the camera and plugged again and since then, it started showing USB 2.1!!

So, I had to downgrade it to 05.12.05.00. During that, in the *realsense-viewer* application window, it showed "Update completed" and didn't give any error/warning there. However, in the terminal, I got the following errors/warnings:
[image]

To confirm, I closed that terminal and ran realsense-viewer again. Now, in the application window, everything else seems OK but still, in the terminal, I'm getting this:
[image]

Please let me know how can I resolve this? Thanks again!

ev-mp, 2020-07-21:
@milan-r-shah hello, the warnings above are given for the **Embedded/Web camera** (see the highlighted label) that you have attached to the PC/Laptop in addition to D435i.

[image]

Can you share the results of running `rs-capture` and `rs-enumerate-devices` with D435i ?
In addition please post `dmesg` log when connecting camera to PC.

milan-r-shah, 2020-07-21:
Thanks a lot @ev-mp for such a quick response. Apart from D435i, I haven't attached any external camera. There is one in-built camera in my laptop (Dell XPS 15).

Here is the output of `rs-capture`:
```bash

#### RS Frame Rate Printer ####
Depth[0]: 30.0/30 [FPS] || Color[0]: 30.0/30 [FPS] || Accel[0]: 30.0/63 [FPS] || Gyro[0]: 30.0/200 [FPS] || 
```
[image]

---

Here is the output of `rs-enumerate-devices`:
```bash
Device info: 
    Name                          : 	Intel RealSense D435I
    Serial Number                 : 	833612071459
    Firmware Version              : 	05.12.05.00
    Recommended Firmware Version  : 	05.12.06.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/video4linux/video2
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B3A
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	850123051659
    Firmware Update Id            : 	850123051659

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format   
    Infrared 1	  1280x800	@ 30Hz	   Y8
    Infrared 2	  1280x800	@ 30Hz	   Y8
    Infrared 1	  1280x800	@ 25Hz	   Y16
    Infrared 2	  1280x800	@ 25Hz	   Y16
    Infrared 2	  1280x800	@ 15Hz	   Y16
    Infrared 1	  1280x800	@ 15Hz	   Y16
    Infrared 2	  1280x800	@ 15Hz	   Y8
    Infrared 1	  1280x800	@ 15Hz	   Y8
    Infrared 2	  1280x720	@ 30Hz	   Y8
    Infrared 1	  1280x720	@ 30Hz	   Y8
    Infrared 1	  1280x720	@ 15Hz	   Y8
    Infrared 2	  1280x720	@ 15Hz	   Y8
    Infrared 2	  1280x720	@ 6Hz	   Y8
    Infrared 1	  1280x720	@ 6Hz	   Y8
    Infrared 2	  848x480	@ 90Hz	   Y8
    Infrared 1	  848x480	@ 90Hz	   Y8
    Infrared 2	  848x480	@ 60Hz	   Y8
    Infrared 1	  848x480	@ 60Hz	   Y8
    Infrared 2	  848x480	@ 30Hz	   Y8
    Infrared 1	  848x480	@ 30Hz	   Y8
    Infrared 1	  848x480	@ 15Hz	   Y8
    Infrared 2	  848x480	@ 15Hz	   Y8
    Infrared 1	  848x480	@ 6Hz	   Y8
    Infrared 2	  848x480	@ 6Hz	   Y8
    Infrared 2	  848x100	@ 300Hz	   Y8
    Infrared 1	  848x100	@ 300Hz	   Y8
    Infrared 2	  848x100	@ 100Hz	   Y8
    Infrared 1	  848x100	@ 100Hz	   Y8
    Infrared 2	  640x480	@ 90Hz	   Y8
    Infrared 1	  640x480	@ 90Hz	   Y8
    Infrared 1	  640x480	@ 60Hz	   Y8
    Infrared 2	  640x480	@ 60Hz	   Y8
    Infrared 1	  640x480	@ 30Hz	   Y8
    Infrared 2	  640x480	@ 30Hz	   Y8
    Infrared 1	  640x480	@ 15Hz	   Y8
    Infrared 2	  640x480	@ 15Hz	   Y8
    Infrared 2	  640x480	@ 6Hz	   Y8
    Infrared 1	  640x480	@ 6Hz	   Y8
[... 759 lines cut ...]
[ 7802.439298] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 7802.590456] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 7802.942011] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 7803.092683] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 7829.840131] hid-sensor-hub 0003:8086:0B3A.0016: No report with id 0xffffffff found
[ 7829.840393] hid-sensor-hub 0003:8086:0B3A.0016: No report with id 0xffffffff found
[ 7858.316647] uvcvideo: Non-zero status (-71) in video completion handler.
[ 7858.343383] uvcvideo: Non-zero status (-71) in video completion handler.
[ 7858.412909] hid-sensor-hub 0003:8086:0B3A.0016: No report with id 0xffffffff found
[ 7858.413133] hid-sensor-hub 0003:8086:0B3A.0016: No report with id 0xffffffff found
[ 7858.601371] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 7858.808200] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 7858.959401] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 7859.110184] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 7859.261048] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 7859.311415] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 7859.663310] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 7859.713685] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 7861.166422] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 7880.708981] hid-sensor-hub 0003:8086:0B3A.0016: No report with id 0xffffffff found
[ 7881.020348] hid-sensor-hub 0003:8086:0B3A.0016: No report with id 0xffffffff found
```
**Note:** The output for `dmesg` shown above is not the complete output because I couldn't scroll up after certain point!

Please let me know if you need any other info/details from my end.

milan-r-shah, 2020-07-22:
Another issue is that I'm getting **Motion Module failure** error upon selecting *Motion Module* in the side panel of *realsense-viewer*:
[image]

Here is the terminal output when that error occurs in *realsense-viewer*
[image]

Weirdly enough this error doesn't occur every time! For example, after the above-mentioned incident, I closed *realsense-viewer*, opened it again and at that time I didn't see any error in *realsense-viewer*. However, at that time, the terminal output was something like this:
[image]

Any suggestions? Thank you!

MartyG-RealSense, 2020-07-22:
Usually when the **Motion Module failure** error box pops up in the RealSense Viewer, it does not stop the IMU functions from working correctly, so it is more of an annoyance than a serious problem.

milan-r-shah, 2020-07-22:
I see. Thanks @MartyG-RealSense  for the confirmation.

Could you please also take a look into my previous comment related to the following error/warning upon launching *realsense-viewer*:
```bash
$ realsense-viewer 
 22/07 11:59:10,886 ERROR [140605082564352] (types.h:308) xioctl(VIDIOC_S_CTRL) failed Last Error: Connection timed out
 22/07 11:59:10,893 WARNING [140605082564352] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Enable / disable backlight compensation
.
.
.
```
or should/can I ignore that as well?

MartyG-RealSense, 2020-07-22:
If the Viewer appears to function fine after the launch error then you can likely dismiss it as a "hiccup" during launch in which there was a problem for a second but it recovered.

milan-r-shah, 2020-07-22:
Got it. Thanks again @MartyG-RealSense for the clarification/confirmation.

MartyG-RealSense, 2020-08-08:
Case closed due to no further comments received.

closed 2020-08-08 by MartyG-RealSense

vuhuulan, 2021-08-25:
> Please navigate to the librealsense > builld directory with **cd build** after having gone to the librealsense root folder and creating the build directory with **mkdir build**, and then try running this instruction:
> 
> **cmake .. -DBUILD_EXAMPLES=true**

Hi, I'm not sure if it's appropriate to ask here but after this what should I do to run ./rs-hello-realsense ? Many thanks.

monson-li, 2024-08-22:
> install TARGETS given no RUNTIME DESTINATION for executable target
>   "rs-hello-realsense"

Though it is much long time for this issue. But I found one useful method to solve this problem.  
The error is:
" install TARGETS given no RUNTIME DESTINATION for executable target
  "rs-hello-realsense"". You can note  "install(TARGETS rs-save-to-disk RUNTIME DESTINATION ${CMAKE_INSTALL_BINDIR})" in the CMakeLists.txt.