realsenseai/librealsense #6785: Android Application "No RealSense devices were found"

YichaoW, 2020-07-08:
I am trying to connect realsense D415 to my pixel 3a XL (Android 10). I followed the documentation thoroughly (https://dev.intelrealsense.com/docs/build-an-android-application-for-intel-realsense-sdk) but it shows "No RealSense devices were found" when I open the app. It works when I connected D415 to my computer.

MartyG-RealSense, 2020-07-09:
Hi @YichaoW   The RealSense Android wrapper can experience conflicts with Android 10 devices.  The fix that works in most cases is to set **targetSdkVersion** to 27.

YichaoW, 2020-07-09:
Hi @MartyG-RealSense I have changed targetSdkVersion to 27 but it still doesn't work. Is there any other way to fix it?

MartyG-RealSense, 2020-07-09:
The discussion below looks at USB permissions on Pixel 2 and 3:

https://github.com/IntelRealSense/librealsense/issues/5388

YichaoW, 2020-07-13:
@MartyG-RealSense I have tried RS Camera app in Google Play and it works but it still can't find realsense device use my own app which developed follow the documentation.

MartyG-RealSense, 2020-07-14:
It looks as though the Android setup guide that you linked to ( https://dev.intelrealsense.com/docs/build-an-android-application-for-intel-realsense-sdk ) is an old one for using the camera with **rooted** Android devices.  A newer version was later released for **unrooted** Android devices.

https://dev.intelrealsense.com/docs/android-build-of-the-intel-realsense-sdk-20

YichaoW, 2020-07-16:
@MartyG-RealSense Thanks a lot! It works now. Do you have Android (Java) API for it?

MartyG-RealSense, 2020-07-16:
Great news that it works for you now!

There is a RealSense Android example for building a Java application at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/java_example/readme.md

closed 2020-08-02 by MartyG-RealSense