realsenseai/librealsense #6787: About T265 Capability in Agricultural Field Robot

5730289021-NN, 2020-07-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  T265  | 
| Firmware Version                   |  | 
| Operating System & Version |     | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 |   |
| SDK Version                            |                           | 
| Language                            |                             | 
| Segment			|  Robot                           | 

### Issue Description
I'm really interesting about the T265 Camera, but I'd like to ask for the information about it first before buying it.
1. My area of interest is about agriculture, the robot will have to perform complete coverage task in the area of 50x100 meter sq. There is nothing within the area except mud which is their roughness going to be changed overtime(because the robot treads over it), however, are noticeable 4 edges in the land which supposed to be the only landmark, I'm little worry when robot is around performing task in this situation. From your experience, will the VSLAM going to localize within 1 meter accuracy?
 
2. Does this camera alone, could it be able to provide the map? The robot should be able to detect the boundary of the land to prevent the robot from traveling across the edge.

3. Does this camera able to work in the dark with light from flashlight ?

Thank you
Norawit

mrantonwinter, 2020-07-09:
following,  im interested for an autonomous mower in a backyard

dorodnic, 2020-07-09:
Hi @5730289021-NN  

> My area of interest is about agriculture, the robot will have to perform complete coverage task in the area of 50x100 meter sq. There is nothing within the area except mud which is their roughness going to be changed overtime(because the robot treads over it), however, are noticeable 4 edges in the land which supposed to be the only landmark, I'm little worry when robot is around performing task in this situation. From your experience, will the VSLAM going to localise within 1 meter accuracy?

We never tested the device in these conditions. You concern seems justified

> Does this camera alone, could it be able to provide the map? The robot should be able to detect the boundary of the land to prevent the robot from traveling across the edge.

The device maintains internal map, but this map is not exposed to upper layers. You should be able to check that the robot is within boundaries but there is no way to get the map.

> Does this camera able to work in the dark with light from flashlight ?

Again, its a valid concern. We cannot guaranty that it does since moving flashlight may mess with appearance of key points.

5730289021-NN, 2020-07-22:
Thank you, now I realize that RTK might be more suitable for my application

closed 2020-07-22 by 5730289021-NN