realsenseai/librealsense #6788: T265 viewer features in UNITY

masterchop, 2020-07-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265} | 
| Firmware Version                   | (0.2.0.951) | 
| Operating System & Version |   {Win (10) | 
| Platform                                 | PC |
| SDK Version                            |  { l2.35.2 }                          | 
| Language                            |  {UNITY}                          | 

### Issue Description

When i test the camera in the intel tool: Realsense Viewer i have the following set of options:

Enable Mapping
Enable Relocalization
Enable Pose Jumping
Enable Dynamic Calibration
Enable Map Preservation

I dont think this features exist in the unity integration as it seems to provide a lower quality of tracking.
Also i dont see any way to load a map or to enable Map Preservation which seems to give better results.

Thanks

dorodnic, 2020-07-09:
Hi @masterchop 
These options are exposed in C# API so you can use them from Unity.
We are not planning to add this feature to Unity wrapper.

masterchop, 2020-07-10:
Sorry to ask but do you have any pointers i dont seem to find any documentation or sample of the C# api for those options.

Thanks

RealSenseSupport, 2021-03-18:
We don't have direct examples to point to, only the current C# examples/documentation there within our wrappers section:  https://github.com/IntelRealSense/librealsense/tree/master/wrappers/csharp

closed 2021-04-06 by RealSenseSupport