realsenseai/librealsense #6792: why TheD435i becomes d430i

bjthemost, 2020-07-09:
Hi，hello everyone，

After updating my d435i to the latest firmware version with realsensor viewer, it becomes d430i. No matter which version of firmware is updated, it will not be displayed as d435i in realsensor viewer. Under Linux, windows displays d430i, and its serial number is 845112070872.

thank you very much

MartyG-RealSense, 2020-07-09:
The model that the camera is detected as depends on the sensor components that are detected by the firmware driver.  If a D435's RGB sensor cannot be detected then the firmware identifies the camera as a D430, as the D430 Depth Module (the camera circuit board inside the D435) does not have an integrated RGB sensor.  In the D435, the RGB sensor is attached separately to the D430 camera board via a cable.

https://github.com/IntelRealSense/librealsense/issues/3124#issuecomment-500233636

There have been a couple of extremely rare cases where the internal cable has become disconnected, so that the D435 is recognised as D430.  It is not something that you could check without opening the camera's casing and invalidating any remaining warranty on it though.

The most common fix is to downgrade the firmware and then re-upgrade to the latest firmware (i.e do not downgrade to an old one and then an older one).  

If a firmware downgrade-upgrade does not fix the problem and you do not want to check inside the casing, you should have the option of returning the camera for a replacement if you purchased it less than 30 days ago.

MartyG-RealSense, 2020-07-18:
Hi @bjthemost   Do you still require assistance with this case, please?   Thanks!

bjthemost, 2020-07-18:
It's been a long time since the warranty

And in China

so。。。

I'm going to throw him back,

thank you @MartyG-RealSense

MartyG-RealSense, 2020-07-18:
Thanks very much for the update.  I appreciate it.

closed 2020-07-18 by MartyG-RealSense