realsenseai/librealsense #6794: T265 error booting when power on

Huibean, 2020-07-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Raspberry pi OS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Raspberry Pi/ |
| SDK Version                            |  { legacy }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
T265 error booting when power on have to replug
```
pi@raspberrypi:~ $  ~/librealsense/build/tools/enumerate-devices/rs-enumerate-devices
 09/07 10:35:00,607 ERROR [3069579280] (tm-boot.h:39) Error booting T265
```

csr-kick, 2020-07-09:
The workaround for this is has been to powercycle the USB chip when the rpi 4 boots up.

install uhubctl then I put in /etc/rc.local to power cycle usb hub:
uhubctl -a cycle -l 1 -p 1-4
uhubctl -a cycle -l 2 -p 1-4

Huibean, 2020-07-11:
@csr-kick not sure that can work yet, but it did broke my ssh channel 😢

Huibean, 2020-07-16:
just confirm that the command is not working on rpi4

Huibean, 2020-07-16:
OK, it have to compile the latest uhubctl then work

closed 2020-07-16 by Huibean