realsenseai/librealsense #6796: What exactly is the difference between librealsense and RealSense SDK 2.0?

milan-r-shah, 2020-07-09:
**Note:** *This one is more of like a question but I couldn't find the proper answer from anywhere else, so creating an issue here.*

I have been trying to understand the difference between **librealsense** and **RealSense SDK 2.0** but so far haven't found the clear answer.
e.g. in the **docs** there are two installation guides: [installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) and another one is [distribution_linux.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md). What is the purpose of each of them?

MartyG-RealSense, 2020-07-09:
The current version of librealsense is known as **librealsense2**, because there was an earlier incarnation of librealsense for older RealSense cameras (F200, R200, ZR300 and SR300) that is now known as Legacy Librealsense.

So librealsense2 (the current librealsense SDK) and RealSense SDK 2.0 are just different names for the same SDK.

Regarding the two different installation guides: **installation.md** is for building librealsense from source code, and **distribution_linux.md** is for building librealsense with package files.

milan-r-shah, 2020-07-13:
Thank you so much @MartyG-RealSense for the clarification :)

closed 2020-07-15 by dorodnic