realsenseai/librealsense #6797: How exactly the build procedure for librealsense for C++ work? How to use it while working on an independent code?

milan-r-shah, 2020-07-09:
| Required Info                 |                                                    |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i                    | 
| Firmware Version                   | 05.12.05.00          | 
| Operating System & Version |   Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)   | 5.3.0-62-generic     | 
| Platform                                 | PC                           |
| SDK Version                          |  2.0                          | 
| Language                              |  C++                         | 
| Segment			        |  others                      | 

### Issue Description
How exactly the build procedure for **librealsense** for C++ work?

Initially, I was facing the problems with building example code snippets. So, I created #6784 
Then to build them without any errors, I had to follow [installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) file for the proper installation of the library.

So, after running required scripts, for running an example code e.g. **rs-hello-realsense.cpp** (in Ubuntu 18.04 LTS), I did this:
```
cd librealsense
mkdir build && cd build
cmake ../ -DBUILD_EXAMPLES=true
sudo make uninstall && make clean && make && sudo make install
cd ./examples/hello-realsense/
./rs-hello-realsense
```
and it worked!!

Afterwords, I also followed #2946 to understand what to do if I modify any example code. So, e.g. if I modify **rs-hello-realsense.cpp** then I have to do simply:
```
cd librealsense/build/examples/hello-realsense/
make
./rs-hello-realsense
```

But, this procedure is only for example codes, right? The main confusion is how can I build my own independent code which is not residing in **librealsense** directory (but somewhere else)? How to use this library just like **OpenCV** or **PCL** (i.e. once installed globally, can be used from anywhere)?

Thanks in advance :)

MartyG-RealSense, 2020-07-10:
Hi @milan-r-shah   I hope that the link below will be helpful to you.

https://github.com/IntelRealSense/librealsense/issues/4277

dorodnic, 2020-07-13:
CMake files provided with the examples are for building them inside the SDK. We also provide [`cmake`](https://github.com/IntelRealSense/librealsense/tree/master/examples/cmake) example showing how you can develop an application on top of the SDK and link to librealsense using CMake. Or if you are on Windows, the link above may be useful

milan-r-shah, 2020-07-13:
> CMake files provided with the examples are for building them inside the SDK. We also provide [`cmake`](https://github.com/IntelRealSense/librealsense/tree/master/examples/cmake) example showing how [...]

Thanks a lot @MartyG-RealSense & @dorodnic . As I'm working on Linux, the quoted response worked. 

I'm not sure whether you guys have already planned to update [code-samples page](https://dev.intelrealsense.com/docs/code-samples) or not but it would be really helpful, if you can include this `cmake` example over there as well. Or may be on [C/C++ wrapper page](https://dev.intelrealsense.com/docs/c-wrapper) when it becomes available.

Thanks again!

MartyG-RealSense, 2020-08-08:
Hi @milan-r-shah    I think that the best way to progress with your request above is to mark this case with a Documentation label and leave it open so that your request about the examples can be looked at by Intel at a later date.   Thanks!

SuperElectron, 2022-04-29:
This is a duplicate of https://github.com/IntelRealSense/librealsense/issues/10436.

Has the IntelRealSense project provided an example of how to create an independent project?

Request:
An independent cmake project that creates a binary executable in C++
`
cd example-project && mkdir build && cd build
cmake -S ../ -B .
./example-project
`

Problem:
It is very hard to understand how to create an independent project.  The reason it is hard is because the examples repo builds 20+ examples in one script (I'd like to do it just for one project).  thanks in advance

MartyG-RealSense, 2022-04-29:
Hi @SuperElectron   There is a C++ example of creating a new project with CMake at the link below.

https://github.com/IntelRealSense/librealsense/tree/master/examples/cmake

There is also a **Getting Started** tutorial project in the link below, which does not have dependencies and so is a good learning experience before doing projects with dependencies.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

Also, a RealSense user created a guide to setting up a Visual Studio project on Windows with the three props files provided by the RealSense SDK.

https://github.com/EduardoWang/visual-studio-real-sense-record-and-playback-project

The prop files are located in the root directory of the RealSense SDK folder at this Windows location after the full SDK has been installed:

**C: > Program Files (x86) > Intel RealSense SDK 2.0**

[image]

MartyG-RealSense, 2022-05-10:
Hi @SuperElectron  Was the information in the comment above helpful to you, please?  Thanks!

SuperElectron, 2022-05-10:
Hey, thank you for the detailed reply.  
I am familiar with Nvidia products and a bit new to intel and its SDK.  Another way of saying this is it takes a while to get used to the realsense SDK (and that costs my employer lots of money).  The cmake example is good; however, it doesn't show how to build a project using the SDK, it tells you how you should do it.  I think I need to do a bit of explaining to justify myself, and I am not saying the documentation and examples are not good, it is more just that the structure is someone difficult to deconstruct.  So, I'll elaborate on the structure for project building.

1. Let's look at this demo: https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openvino/rs-vino

- I can only build it with a shell script as directed from realsense docs from the top directory
- when I take the project and put it into its own folder, the inference libraries don't build
- the build structure for everything in the wrappers directory (librealsense/wrappers/openvino/rs-vino) relies on a 'nested' structure of cmake files which is hard to deconstruct and create my own project.

I hope this explains what I was trying to say with my previous comment above: "It is very hard to understand how to create an independent project. The reason it is hard is because the examples repo builds 20+ examples in one script (I'd like to do it just for one project)."

2. Here is my attempt at creating a standalone project: https://github.com/SuperElectron/openvino-rs-face-demo/tree/master
- View the build instructions at the top of CMakeLists.txt
- I have added a folder called 'cmake-modules' to include dependencies.  This is my attempt at "deconstructing"

3. This example below is not a stand alone example.
- https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openvino/face
- You cannot build it in wrappers/openvino/face
- If you take the project and move it to somewhere else (i.e. ~/code/face), the code won't build
- moving to another directory (~/code/face) cannot build because of the nested cmake build structure. 
- this is what I meant by it's hard to deconstruct an example and build my own project

----------------------------------------------------------------------

So, my challenge is this:
- Can I build my own project from https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openvino/face?

Any thoughts, pointers, or hints is greatly appreciated.  And thank you very much for the links @MartyG-RealSense , I will be reading them and trying to get this repo working.

MartyG-RealSense, 2022-05-11:
Thanks again for your patience.  A RealSense support team member provides advice at the link below for setting up an independent OpenVINO project in Visual Studio instead of CMake.  

https://support.intelrealsense.com/hc/en-us/community/posts/4404515238163-Intel-Realsense-SDK-OpenVINO-wrapper-for-the-D435-depth-camera

Are you able to use Visual Studio on Windows?

Shreyas-NR, 2022-07-27:
Hi all,
Here's how I got my project working starting from scratch. 

Steps:-
1. Open Visual studio, create an empty project
2. Go to View -> other windows -> property manager
    In property manager, right-click your project, add the existing property sheet
    Browse to where you installed Intel RealSense SDK. In my case, it is C:\Program Files (x86)\Intel RealSense SDK 2.0
    Add the file "intel.realsense.props" Project Property File.

Create a new Source file"FileName.cpp"
Paste the source code from any example project and save.

3. In Project Properties
1. VC++ Directories 
	Update the Library Directories with the following path
	`C:\Program Files %28x86%29\Intel RealSense SDK 2.0\lib\x64;$(LibraryPath)`
	
2. 	C/C++
	General-> Update the Additional Include Directories with the following path
	`C:\Program Files %28x86%29\Intel RealSense SDK 2.0\third-party\glfw-imgui\include\GLFW;C:\Program Files %28x86%29\Intel RealSense SDK 2.0\include;%(AdditionalIncludeDirectories)`
	
3. Linker
	General-> Update the Additional Library Directories with the following path
        `$(librealsenseSDK)\lib\$(PlatformShortName);%(AdditionalLibraryDirectories)`
		
	Input-> Update the Additional Dependencies with the following path
        `C:\Program Files (x86)\Intel RealSense SDK 2.0\lib\x64\realsense2.lib;%(AdditionalDependencies)`	
	System->
	`SubSystem Type = Console (/SUBSYSTEM:CONSOLE)`
	
4. Build Events
	Post-Build Event->
	Update the Command Line with the following command
	`xcopy /y "C:\Program Files (x86)\Intel RealSense SDK 2.0\bin\x64\realsense2.dll"  "$(OutDir)"`
	
That's it! you have now successfully created your custom project with Intel RealSense SDK.

MartyG-RealSense, 2022-07-27:
Thanks so much @Shreyas-NR for sharing your detailed Windows and Visual Studio project setup guide with the RealSense community!

MartyG-RealSense, 2026-10-23:
This open GitHub issue is being closed due to its significant age.

closed 2026-10-23 by MartyG-RealSense