realsenseai/librealsense #6798: where I can find L515 driver for windows 7. L515 can't work on windows 7.  D435 work well

donaldlee2008, 2020-07-10:
where I can find L515 driver for windows 7 ?. L515 can't work on windows 7.  D435 work well

Intel.RealSense.SDK-WIN7-2.36.0.2034.exe
Depth Quality Tool for Intel RealSense Cameras (Win7)
Intel RealSense Viewer (Win7)
[image]
[image]
[image]

RealSenseSupport, 2020-07-13:
Hi @donaldlee2008 

You are correct, we do not have Windows 7 support for the L515 camera.  At this time it has been discussed but no schedule or timeframe has been decided.  We will take note of your request for Win 7 support.

cja-515, 2020-09-29:
Hi@dorodnic 
I have added the product ID according to “Updating Win7 INF with L515 product ID”, and I saw the corresponding driver in the device manager, but the device cannot be automatically recognized when I open the Depth Quality Tool for Intel RealSense Cameras (Win7). Is it my fault in which step?

The following is what I modified based on your solution
-------------------------------------------------------
%DeviceNameDepthL515% =UVC_Install, USB\VID_8086&PID_0B64&MI_00
%DeviceNameColorL515% =UVC_Install, USB\VID_8086&PID_0B64&MI_01
-------------------------------------------------------------------------
DeviceNameDepthL515="Intel(R) RealSense(TM) L515 Depth"
DeviceNameColorL515="Intel(R) RealSense(TM) L515 RGB"

cja-515, 2020-09-29:
Hi @dorodnic
 I have modified the product ID based on ‘’Updating Win7 INF with L515 product ID‘’, and I also saw the driver in the device manager, but when I opened Depth Quality Tool for Intel RealSense Cameras (Win7)（Intel.RealSense.SDK-WIN7-2.36.0.2034.exe）, it did not automatically connect to the device and  showed "please connect device".Did I miss anything?

The following is the part I modified on your solution

%DeviceNameDepthL515% =UVC_Install, USB\VID_8086&PID_0B64&MI_00
%DeviceNameColorL515% =UVC_Install, USB\VID_8086&PID_0B64&MI_01

DeviceNameDepthL515="Intel(R) RealSense(TM) L515 Depth"
DeviceNameColorL515="Intel(R) RealSense(TM) L515 RGB"

RealSenseSupport, 2020-10-09:
@cja-515 As PR https://github.com/IntelRealSense/librealsense/pull/6873 not merged yet, win7 support for L515 is still not working yet. Thanks!

cja-515, 2020-10-09:
Does the Realsense Viewer or Deep Quality Tool not support L515 in WIN 7?

cja-515, 2020-10-09:
Does the Realsense Viewer or Deep Quality Tool not support L515 in WIN 7?  When will WIN7 be compatible with L515?？

RealSenseSupport, 2020-10-09:
@cja-515  Yes, RealSensViewer and Depth Quality Tool don't support L515 in Win7 yet. The feature is still under working. Any update will let you know. Thanks!

RealSenseSupport, 2021-08-24:
@cja-515 Sorry that this feature won't do. Close this accordingly.

closed 2021-08-24 by RealSenseSupport