realsenseai/librealsense #6799: Is it possible to convert pointcloud back from pcl to rs2::frame/rs2::points/rs2::pointcloud to perform post-processing?

96daja11, 2020-07-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |   Linux Ubuntu 16.04  | 
| Kernel Version (Linux Only)    | 4.15.0-106-generic                          | 
| Platform                                 | PC  |
| SDK Version                            |  2.35.2     | 
| Language                            |  C++        | 
| Segment			|  Robotics        | 

Hi all,

I am working on a project where I first use the post-processing filters on the input from the camera. Then I convert the output to pcl pointcloud for using its functionality for segmenting out planes in the cloud. But since the output from the pcl algorithm is a bit noisy, I would like to once again perform some filtering with e.g. the spatial-edge and temporal filters. I therefore wonder if it is possible to convert the pcl pointcloud back to some kind of rs2 data type to again perform post-processing by the filters? Thankful for help!

MartyG-RealSense, 2020-07-10:
Hi @96daja11  I considered this problem very carefully but could not find an easy solution.  It may be better to first try filtering the noise using PCL's own noise filters.  The PCL documentation has a couple of tutorials for this.

**Removing sparse outliers from noisy data**
http://pointclouds.org/documentation/tutorials/statistical_outlier.html#statistical-outlier-removal

**Removing outliers from noisy data using a Conditional or RadiusOutlier removal**
http://pointclouds.org/documentation/tutorials/remove_outliers.html#remove-outliers

96daja11, 2020-07-15:
Yes that was my other option, I just wanted to check if the post-processing filters were an option and if converting from pcl to rs2 was possible. Thanks anyway!

closed 2020-07-15 by 96daja11