realsenseai/librealsense #6800: Aligned image size is half of the depth image size.

ahasan016, 2020-07-10:
Hello, I need to align depth image with rgb and I shouldn't reduce the dimension of the aligned image less than (640,480) . I have using c++ to write my program and has tried your alignment example. Whenever I try to align them, it always make the aligned image half of the size of depth image. Can anyone help? I  am pretty sure that I am doing something wrong here. Also i need precise depth information. So, any suggestion about these would be highly appreciated. 

```
	rs2::spatial_filter spat;
	spat.set_option(RS2_OPTION_HOLES_FILL, 5); // 5 = fill all the zero pixels
        rs2::align align_to(RS2_STREAM_DEPTH); //align to the depth (if i align to color then the depth info is really worst +/-25cm)
        rs2::pipeline pipe;
         
        rs2::config cfg;
	cfg.enable_stream(RS2_STREAM_DEPTH,848,480); // Enable depth camera
	cfg.enable_stream(RS2_STREAM_COLOR,1920,1080); //no matter what size i put the result is same size after alignment
	
        auto profile = pipe.start(cfg);
	auto sensor = profile.get_device().first<rs2::depth_sensor>();
	// Set the device to High Accuracy preset of the D400 stereoscopic cameras
	if (sensor && sensor.is<rs2::depth_stereo_sensor>())
	{
		sensor.set_option(RS2_OPTION_VISUAL_PRESET, RS2_RS400_VISUAL_PRESET_HIGH_ACCURACY);
	}

	auto stream = profile.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>();
        
        const auto window_name = "Display Image";
	namedWindow(window_name, WINDOW_AUTOSIZE);
	//rs2::frameset data;
	while (true)
	{
		auto data = pipe.wait_for_frames();
		
		// First make the frames spatially aligned
		//data = data.apply_filter(align_to);
		data = data.apply_filter(dec);
		data = data.apply_filter(spat);
		data = data.apply_filter(align_to);
		
                /* 
		   data = align_to.process(data);*/  /// I have also tried aligned_to.process()
	
		
		auto color = data.get_color_frame();
		auto depth = data.get_depth_frame();
		auto color_mat = frame_to_mat(color);
		
		
		cout << "depth width"<<depth.get_width() << " - " <<depth.get_height()<<endl;
		cout << "color width" << color.get_width() << " - " << color.get_height() << endl;

		imshow(window_name, color_mat);
		waitKey(1);
          }
  ```
   The resulted picture looks like this (please ignore the opencv window color frame ): 
  
[image]

The resulted depth is wrong (originally about 33 cm) and the height and width of the aligned images are (424,240). 
I somehow want the images of size (640,480) without black portion.  Btw, I couldn't exactly follow the aligned example or measure example as i dont want to use OpenGL.

kafan1986, 2020-07-10:
`data = data.apply_filter(dec);`

@ahasan016 I believe you are using a decimation filter here. Can you post your decimation factor value? You are aligning post the decimation step. If the decimation factor is 2, then it will reduce the depth frame size by factor of 2 and the behaviour that you are seeing is expected.

MartyG-RealSense, 2020-07-10:
Thanks again for your help @kafan1986    :)

ahasan016, 2020-07-10:
> `data = data.apply_filter(dec);`
>
> @ahasan016 I believe you are using a decimation filter here. Can you post your decimation factor value? You are aligning post the decimation step. If the decimation [...]

@kafan1986  and @MartyG-RealSense yes. This was it. Thanks for your quick reply. Btw, I still am facing issue with depth precision. Any idea about it. I get depth value really good if I use these settings 
```
dec.set_option(RS2_OPTION_FILTER_MAGNITUDE, 2);
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_DEPTH,640,480); // Enable depth
// For the color stream, set format to RGBA
// To allow blending of the color frame on top of the depth frame
cfg.enable_stream(RS2_STREAM_COLOR,640,480);
```
[image]

but its always +/-(6or7) for my current case if I use anything else.

MartyG-RealSense, 2020-07-10:
@ahasan  The optimal depth accuracy resolutions are 848x480 for D435 models and 1280x720 for D415.  I would recommend matching depth and color resolutions where possible.  If you have a choice between 1280x720 and 1920x1080 when aligning depth to color, I would recommend 1280x720 for color.

ahasan016, 2020-07-10:
> @Ahasan The optimal depth accuracy resolutions are 848x480 for D435 models and 1280x720 for D415. I would recommend matching depth and color resolutions where possible. If you have a choice between 1280x720 and 1920x1080 when aligning depth to color, I would recommend 1280x720 for color.

I am using D435i camera.. So what is your recommendation?

MartyG-RealSense, 2020-07-10:
Try 848x480 for depth and 1280x720 for color, please/

ahasan016, 2020-07-10:
> Try 848x480 for depth and 1280x720 for color, please.

I have tried but the result isn't changed. It's the same. Only works with 

```
dec.set_option(RS2_OPTION_FILTER_MAGNITUDE, 2);
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_DEPTH,640,480); // Enable depth
// For the color stream, set format to RGBA
// To allow blending of the color frame on top of the depth frame
cfg.enable_stream(RS2_STREAM_COLOR,640,480);
```
However, things are different in your Intel RealSense Viewer. I always find a good depth value in the viewer app (regardless of the depth and color image size). Currently I have the following settings in the viewer:
```
{
    "aux-param-autoexposure-setpoint": "1536",
    "aux-param-colorcorrection1": "0.298828",
    "aux-param-colorcorrection10": "-0",
    "aux-param-colorcorrection11": "-0",
    "aux-param-colorcorrection12": "-0",
    "aux-param-colorcorrection2": "0.293945",
    "aux-param-colorcorrection3": "0.293945",
    "aux-param-colorcorrection4": "0.114258",
    "aux-param-colorcorrection5": "-0",
    "aux-param-colorcorrection6": "-0",
    "aux-param-colorcorrection7": "-0",
    "aux-param-colorcorrection8": "-0",
    "aux-param-colorcorrection9": "-0",
    "aux-param-depthclampmax": "65536",
    "aux-param-depthclampmin": "0",
    "aux-param-disparityshift": "0",
    "controls-autoexposure-auto": "True",
    "controls-autoexposure-manual": "8500",
    "controls-color-autoexposure-auto": "True",
    "controls-color-autoexposure-manual": "156",
    "controls-color-backlight-compensation": "0",
    "controls-color-brightness": "0",
    "controls-color-contrast": "50",
    "controls-color-gain": "64",
    "controls-color-gamma": "300",
    "controls-color-hue": "0",
    "controls-color-power-line-frequency": "3",
    "controls-color-saturation": "64",
    "controls-color-sharpness": "50",
    "controls-color-white-balance-auto": "True",
    "controls-color-white-balance-manual": "4600",
    "controls-depth-gain": "16",
    "controls-laserpower": "150",
    "controls-laserstate": "on",
    "ignoreSAD": "0",
    "param-amplitude-factor": "0",
    "param-autoexposure-setpoint": "1536",
    "param-censusenablereg-udiameter": "6",
    "param-censusenablereg-vdiameter": "7",
    "param-censususize": "6",
    "param-censusvsize": "7",
    "param-depthclampmax": "65536",
    "param-depthclampmin": "0",
    "param-depthunits": "1000",
    "param-disableraucolor": "0",
    "param-disablesadcolor": "0",
    "param-disablesadnormalize": "0",
    "param-disablesloleftcolor": "0",
    "param-disableslorightcolor": "0",
    "param-disparitymode": "0",
    "param-disparityshift": "0",
    "param-lambdaad": "630",
    "param-lambdacensus": "26",
    "param-leftrightthreshold": "18",
    "param-maxscorethreshb": "2047",
    "param-medianthreshold": "347",
    "param-minscorethresha": "80",
    "param-neighborthresh": "82",
    "param-raumine": "3",
    "param-rauminn": "1",
    "param-rauminnssum": "6",
    "param-raumins": "2",
    "param-rauminw": "2",
    "param-rauminwesum": "3",
    "param-regioncolorthresholdb": "0.0254403",
    "param-regioncolorthresholdg": "0.954012",
    "param-regioncolorthresholdr": "0.0283757",
    "param-regionshrinku": "4",
    "param-regionshrinkv": "1",
    "param-robbinsmonrodecrement": "15",
    "param-robbinsmonroincrement": "5",
    "param-rsmdiffthreshold": "3.34375",
    "param-rsmrauslodiffthreshold": "0.96875",
    "param-rsmremovethreshold": "0.666667",
    "param-scanlineedgetaub": "442",
    "param-scanlineedgetaug": "16",
    "param-scanlineedgetaur": "770",
    "param-scanlinep1": "21",
    "param-scanlinep1onediscon": "3",
    "param-scanlinep1twodiscon": "27",
    "param-scanlinep2": "182",
    "param-scanlinep2onediscon": "41",
    "param-scanlinep2twodiscon": "63",
    "param-secondpeakdelta": "205",
    "param-texturecountthresh": "0",
    "param-texturedifferencethresh": "3474",
    "param-usersm": "1",
    "param-zunits": "1000",
    "stream-depth-format": "Z16",
    "stream-fps": "30",
    "stream-height": "480",
    "stream-width": "848"
}
```

I wanted to apply the exported JSON file as the setting but I don't know how to do that. 
```
       //// extra_added
	auto prof = pipe.start();
	rs2::device dev = prof.get_device();
	auto advanced = dev.as<advanced_mode>();
	advanced.load_json("D:/Mask_Robot_neural_stick2/effective_settings.json");
	//////////////////////
     while (true)
	{
		//// Fetch frames from the pipeline and send them for processing
		//if (pipe.poll_for_frames(&data))
		//{
		auto data = pipe.wait_for_frames();
		auto color1 = data.get_color_frame();
		auto color_mat1 = frame_to_mat(color1);
		imshow(window_name1, color_mat1);
		waitKey(1);
        }
```
but ended up with an error "Unhandled exception at 0x00007FFB0934A799 in roboter_mask_try.exe: Microsoft C++ exception: rs2::error at memory location 0x00000003EE8FEEA0. occurred"

MartyG-RealSense, 2020-07-10:
Regarding loading the json, could you try moving the json loading code to directly beneath the  **rs2::pipeline pipe;** line please.

` rs2::pipeline pipe;
auto prof = pipe.start();
rs2::device dev = prof.get_device();
auto advanced = dev.as<advanced_mode>();
advanced.load_json("D:/Mask_Robot_neural_stick2/effective_settings.json");`

ahasan016, 2020-07-10:
> Regarding loading the json, could you try moving the json loading code to directly beneath the **rs2::pipeline pipe;** line please.
>
> **rs2::pipeline pipe; auto prof = pipe.start(); rs2::device dev = [...]

actually I am doing it.  please check my main function:
```
int main()
{
	bool bExit = false;

	

	rs2::align align_to(RS2_STREAM_DEPTH);*/
	rs2::colorizer color_map;
	//color_map.set_option(RS2_OPTION_COLOR_SCHEME, 2.f);
	// Decimation filter reduces the amount of data (while preserving best samples)
	rs2::decimation_filter dec;
	// If the demo is too slow, make sure you run in Release (-DCMAKE_BUILD_TYPE=Release)
	// but you can also increase the following parameter to decimate depth more (reducing quality)
	dec.set_option(RS2_OPTION_FILTER_MAGNITUDE, 1);
	// Define transformations from and to Disparity domain
	rs2::disparity_transform depth2disparity;
	rs2::disparity_transform disparity2depth(false);
	// Define spatial filter (edge-preserving)
	rs2::spatial_filter spat;
	// Enable hole-filling
	// Hole filling is an agressive heuristic and it gets the depth wrong many times
	// However, this demo is not built to handle holes
	// (the shortest-path will always prefer to "cut" through the holes since they have zero 3D distance)
	spat.set_option(RS2_OPTION_HOLES_FILL, 5); // 5 = fill all the zero pixels
	// Define temporal filter
	rs2::temporal_filter temp;
	// Spatially align all streams to depth viewport
	// We do this because:
	//   a. Usually depth has wider FOV, and we only really need depth for this demo
	//   b. We don't want to introduce new holes
	rs2::align align_to(RS2_STREAM_DEPTH);

	// Declare RealSense pipeline, encapsulating the actual device and sensors
	rs2::pipeline pipe;

	//// extra_added
	auto prof = pipe.start();
	rs2::device dev = prof.get_device();
	auto advanced = dev.as<advanced_mode>();
	advanced.load_json("effective_settings.json");
	
	//////////////////////
	const auto window_name = "Display Image";
	namedWindow(window_name, WINDOW_AUTOSIZE);
	const auto window_name1 = "Display Image_before_align";
	namedWindow(window_name1, WINDOW_AUTOSIZE);
	//rs2::frameset data;
	while (true)
	{
		//// Fetch frames from the pipeline and send them for processing
		//if (pipe.poll_for_frames(&data))
		//{
		auto data = pipe.wait_for_frames();
		auto color1 = data.get_color_frame();
		auto color_mat1 = frame_to_mat(color1);
		imshow(window_name1, color_mat1);
		waitKey(1);
		

		data = data.apply_filter(align_to);
		

		
		
		auto color = data.get_color_frame();
		
		auto depth = data.get_depth_frame();
		
		auto color_mat = frame_to_mat(color);
		
		cout << "depth width"<<depth.get_width() << " - " <<depth.get_height()<<endl;
		cout << "color width" << color.get_width() << " - " << color.get_height() << endl;
		imshow(window_name, color_mat);
		waitKey(1);
		//aruco_try(color_mat);
		

		//render_simple_distance(depth, corners[0], corners[1]);
			

		
	}

}
```

MartyG-RealSense, 2020-07-10:
Okay, thank you.  It was not clear from your provided listings that you were already doing this, as the **rs2::pipeline pipe;** line was not listed the first time that you showed your  **//// extra_added** code snippet above.

[image]

ahasan016, 2020-07-10:
I am sorry that I forgot to put that line in my code snippet. Anyway, I really need to make the depth information right. The depth also changes according to the position (x,y) of the object. Isn't there any way to make it right? I can tolerate a deviation of max (+/- 1 cm). what is the max depth accuracy for the D435i can I expect? I will have to deal with the changing height as well. Because the camera will be installed in a robotic arm and the arm can move along all the axis. Any particular settings, post-processing or any kind of suggestion I can follow.

MartyG-RealSense, 2020-07-10:
The depth accuracy of the 400 Series cameras is less than one percent of the distance from the object.  So if the camera is 1 meter from the object, the expected accuracy is between 2.5 mm to 5 mm.

The new RealSense D455 model has improved accuracy over distance by a factor of 2x compared to the D435 models (giving accuracy at 6 meters comparable to that of 3 meters distance on D435).

To be honest, the ever helpful @kafan1986 Can answer this subject better than I can.  Until he is back, you may find useful information in a recent discussion about getting xyz coordinates that @kafan1986 also helped with.

https://github.com/IntelRealSense/librealsense/issues/6749

ahasan016, 2020-07-10:
ok. I have checked the issue. But I guess my scenario is a bit different as I have multiple objects to focus on and my camera will also move. However, I have noticed from the above issue that the depth changes over the area on the image is also an issue. you can check the following picture. Although I have pointed the camera on the flat surface but the depth values are different in different directions. Shouldn't it be the same at all pixels? 
[image]

I think I also have my camera placed somewhat parallel to the floor. Say for it is not perfectly parallel, is the depth value affected by the camera placement angle? if so, then can I use the imu data to fix it? any solution or example can I follow till @kafan1986 comes back?

kafan1986, 2020-07-10:
@ahasan016 Frankly I have never worked with D435 camera and don't know much about IMU data. But given my limited knowledge, I do believe you can deduce the angle of the camera from IMU. If the angle is known, then depth values can be adjusted as per requirement using simple trigonometry.

MartyG-RealSense, 2020-07-10:
Welcome back @kafan1986 

@ahasan016   I would think that having the camera oriented at an angle instead of pointing straight ahead (parallel to the floor) would have an effect on depth analysis, because some parts of the observed surface (the floor in this case) would appear to the camera to be nearer to the camera lens than other parts.

ahasan016, 2020-07-10:
@kafan1986 , @MartyG-RealSense  I have checked the parallelity o my camera and I have these pictures for your reference. Do you think this is normal to have this much deviation in depth like in the first picture for the installment of the camera like the second picture?
[image]
[image]

@kafan1986 could you please direct me towards the mathematical equations for correcting the depth regardless of the area using the Euler angles from the accelerometer and gyroscope?

kafan1986, 2020-07-10:
There will be 3 different angles - Yaw, Pitch and Roll. Roll won't impact the depth, so can be ignored. Rest two will impact the depth. In my case, I had only Pitch to worry about, so calculations only had to take care of this factor. I currently, don't have a notepad with me to calculate the formulae to be used.

MartyG-RealSense, 2020-07-10:
I work the difference between the values out to be about 4 cm.  Given that the upper object is further away from the center of the camera lens, this seems a reasonable value.  Even though it is flat on the floor, the upper object is further from the camera.  It would be like putting a book on the floor at your feet and then pushing it away from you with your foot: you would then have to reach forward over a greater distance to touch it.

kafan1986, 2020-07-10:
> I work the difference between the values out to be about 4 cm. Given that the upper object is further away from the center of the camera lens, this seems a reasonable value. Even though it is flat [...]

If the floor and camera plane is parallel, then the depth should be same. The depth calculated through stereo disparity always gives perpendicular distance to camera plane. So it won't matter if the pixel is at the center or at the edge of the image. The euclidean distance will off course change as described by your floor and feet example.

ahasan016, 2020-07-10:
@kafan1986 exactly as you have written, I was thinking like that and it seemed to me that the camera was exactly parallel to the floor so it shouldn't affect the depth so much. at least not by 4-5 cm.  However, if you can write me the equation or give me a suitable link to follow then it will be really helpful. I want to give it a try even with considering the angles.

ahasan016, 2020-07-13:
@MartyG-RealSense , @kafan1986  sorry to bother you guys. I have actually found a pattern for the depth changes on a flat surface. Please check the images I have plotted for depths.
[image]
[image]

Only changes in the images are the actual heights from the camera. Any idea how to flatten them to a certain point? let say I want to have a certain height for all the pixels as the mid pixel of the image? I can ignore the spike on the second image.

MartyG-RealSense, 2020-07-13:
@ahasan016   I recalled that I had seen something like this graph before and managed to find the case on the Intel Support RealSense forum.

https://support.intelrealsense.com/hc/en-us/community/posts/360037100954-Deviation-in-distance-to-a-flat-surface-Realsense-415-Python-

The floor in their image looked like the image below, with the upper image demonstrating the variations in detected depth as a histogram.

[image]

During that case I noted that on the RGB image there was heavy shadow in some places and lighter areas in others and speculated whether that was affecting the depth values being read from the floor.

ahasan016, 2020-07-13:
@MartyG-RealSense.. Sorry for being late. 
I guess the i have the similar kind of floor where i am trying right now. I don't know whether i have the shadow or not as i haven't seen anything like that with my eyes. I have a fixed light above me. 
Anyway, if it is the case of shadow then what am i required to do? Any idea?

MartyG-RealSense, 2020-07-13:
@ahasan016   If your carpet is also a dark grey shade then that can be problematic in itself for depth analysis, as dark grey / black shades absorb light and make it difficult for depth cameras (not just RealSense) to read depth detail from them as they reflect little light back.  The darker the shade, the less light that is reflected and so the harder it is for the camera to depth-sense the surface.  

A recent case involving a black office chair demonstrated that depth detail can be perceived if the dark surface is illuminated by a strong light source, lightening the surface.  

https://github.com/IntelRealSense/librealsense/issues/6757

I wonder if purchasing an **IR-pass filter** and applying it over the camera lens would make a dark carpet easier to depth-analyse by making the projector pattern appear brighter.  Section 4.2.1 of Intel's white paper document on optical filters describes this.

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400

ahasan016, 2020-07-13:
@MartyG-RealSense Thanks for your information. 
This isn't my original environment of operation. I have a few recorded depth videos from the actual robot working environment. I will check those depth frames. And will let you know whether the problem occurs there too. The environment basically looks like the following images: 
[image]
[image]
[image]

MartyG-RealSense, 2020-07-13:
Thanks very much for the above images.  Regarding the floor here, there is the potential for the camera to be confused by a different phenomenon related to surfaces that form repetitive patterns, like a row of fence posts.  I hope that the link below will provide useful information about addressing this if it does occur.

https://github.com/IntelRealSense/librealsense/issues/6713#issuecomment-651114720

ahasan016, 2020-07-13:
@MartyG-RealSense Thanks for the early alert. I will also check on that. will get back to you after I test the camera on the real environment.

MartyG-RealSense, 2020-08-02:
Hi @ahasan016   Do you wish to provide further test results please, or can this case be closed?  Thanks!

ahasan016, 2020-08-02:
I have actually done test on the real scene where the distance of the camera from the object is about 1.5 meter. I need some final suggestions to improve the accuracy of depth to approx. ~4mm. I am going to try anything which is necessary. Recently, I have also tried the Lidr L515 and the accuracy was even a lot worse than the D435i. I can't do any changes to the scenario. But can try any kind of calibration or can add light sources or filters. So, some specific advise to achieve the accuracy on my environment will be highly appreciated.

MartyG-RealSense, 2020-08-02:
The depth accuracy of the D400 series cameras is less than one percent of the distance from the object.  If the camera is 1m from the object, the expected accuracy is between 2.5mm to 5mm.

So at 1.5 meters, 4 mm would likely be considered a good result and there may not be much more that can be achieved without the aid of additional means such as physical optical filters or by turning on as many post-processing filters as possible

If you are making use of the camera's projector dot pattern then you may be able to reduce error further by using an external LED-based pattern projector.  Using an LED projector instead of the built-in projector helps to minimise a type of noise on the image called "laser speckle" that the laser-based projector in the camera can generate.  A reduction in RMS error (error over distance) of around 30% is expected when the built-in projector is turned off.

ahasan016, 2020-08-03:
> The depth accuracy of the D400 series cameras is less than one percent of the distance from the object. If the camera is 1 meter from the object, the expected accuracy is between 2.5mm to 5mm.
> 
> So at 1.5 meters, 4 mm would likely be considered a good result and there may not be much more that can be achieved without the aid of additional means such as physical optical filters or by turning on as many post-processing filters as possible
> 
> If you are making use of the camera's projector dot pattern then you may be able to reduce error further by using an external LED-based pattern projector. Using an LED projector instead of the built-in projector helps to minimise a type of noise on the image called "laser speckle" that the laser-based projector in the camera can generate. A reduction in RMS error (error over distance) of around 30% is expected when the built-in projector is turned off.
> 
> The subject of external projectors is discussed in the Intel paper in the link below.
> 
> https://dev.intelrealsense.com/docs/projectors

 Any particular suggestions for buying the external projectors? I mean do you recommend any specific projectors which should be good enough for us? We have also tested the Lidar L515 sensor and have recently bought D415. L515 has really poor performance compared to the D435i but the D415 seems promising. Still would like to get the best out of D415 and it would be great if you could suggest the specific add ons like projectors or filters. I can also upload some more RGBs of our environment if you want.

MartyG-RealSense, 2020-08-03:
As far as I know Intel haven't recommended specific pattern projector models for RealSense, though have provided guidance for choosing one in the camera tuning guide for the 400 Series cameras.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-use-external-projector

Make sure that it is a pattern projector that you are choosing if you need a dot-pattern projector, and not just a projector that casts illumination but not a pattern.  The link below illustrates this point.

https://github.com/IntelRealSense/librealsense/issues/2000#issuecomment-480795410

MartyG-RealSense, 2020-08-16:
Case closed due to no further comments received.

closed 2020-08-16 by MartyG-RealSense

ahasan016, 2020-08-24:
So, I have now D415 camera and this time I have tried it on a white wall. I have the results as follows.
Camera to wall distance :0576mm :
[image]
Camera to wall distance :0917mm :
[image]
Camera to wall distance :1282mm :
[image]
Camera to wall distance :1615mm :
[image]
Camera to wall distance :1952mm :
[image]

The parameters are fixed as follows :
```
{
    "aux-param-autoexposure-setpoint": "1536",
    "aux-param-colorcorrection1": "0.461914",
    "aux-param-colorcorrection10": "-0.553711",
    "aux-param-colorcorrection11": "-0.553711",
    "aux-param-colorcorrection12": "0.0458984",
    "aux-param-colorcorrection2": "0.540039",
    "aux-param-colorcorrection3": "0.540039",
    "aux-param-colorcorrection4": "0.208008",
    "aux-param-colorcorrection5": "-0.332031",
    "aux-param-colorcorrection6": "-0.212891",
    "aux-param-colorcorrection7": "-0.212891",
    "aux-param-colorcorrection8": "0.68457",
    "aux-param-colorcorrection9": "0.930664",
    "aux-param-depthclampmax": "65536",
    "aux-param-depthclampmin": "0",
    "aux-param-disparityshift": "0",
    "controls-autoexposure-auto": "True",
    "controls-autoexposure-manual": "75714",
    "controls-color-autoexposure-auto": "True",
    "controls-color-autoexposure-manual": "312",
    "controls-color-backlight-compensation": "0",
    "controls-color-brightness": "0",
    "controls-color-contrast": "50",
    "controls-color-gain": "64",
    "controls-color-gamma": "300",
    "controls-color-hue": "0",
    "controls-color-power-line-frequency": "3",
    "controls-color-saturation": "64",
    "controls-color-sharpness": "50",
    "controls-color-white-balance-auto": "True",
    "controls-color-white-balance-manual": "4600",
    "controls-depth-gain": "16",
    "controls-depth-white-balance-auto": "False",
    "controls-laserpower": "360",
    "controls-laserstate": "on",
    "ignoreSAD": "0",
    "param-amplitude-factor": "0",
    "param-autoexposure-setpoint": "1536",
    "param-censusenablereg-udiameter": "9",
    "param-censusenablereg-vdiameter": "9",
    "param-censususize": "9",
    "param-censusvsize": "9",
    "param-depthclampmax": "65536",
    "param-depthclampmin": "0",
    "param-depthunits": "1000",
    "param-disableraucolor": "0",
    "param-disablesadcolor": "0",
    "param-disablesadnormalize": "0",
    "param-disablesloleftcolor": "0",
    "param-disableslorightcolor": "0",
    "param-disparitymode": "0",
    "param-disparityshift": "0",
    "param-lambdaad": "800",
    "param-lambdacensus": "26",
    "param-leftrightthreshold": "24",
    "param-maxscorethreshb": "2047",
    "param-medianthreshold": "500",
    "param-minscorethresha": "1",
    "param-neighborthresh": "7",
    "param-raumine": "1",
    "param-rauminn": "1",
    "param-rauminnssum": "3",
    "param-raumins": "1",
    "param-rauminw": "1",
    "param-rauminwesum": "3",
    "param-regioncolorthresholdb": "0.0499022",
    "param-regioncolorthresholdg": "0.0499022",
    "param-regioncolorthresholdr": "0.0499022",
    "param-regionshrinku": "3",
    "param-regionshrinkv": "1",
    "param-robbinsmonrodecrement": "10",
    "param-robbinsmonroincrement": "10",
    "param-rsmdiffthreshold": "4",
    "param-rsmrauslodiffthreshold": "1",
    "param-rsmremovethreshold": "0.375",
    "param-scanlineedgetaub": "72",
    "param-scanlineedgetaug": "72",
    "param-scanlineedgetaur": "72",
    "param-scanlinep1": "60",
    "param-scanlinep1onediscon": "105",
    "param-scanlinep1twodiscon": "70",
    "param-scanlinep2": "342",
    "param-scanlinep2onediscon": "190",
    "param-scanlinep2twodiscon": "130",
    "param-secondpeakdelta": "325",
    "param-texturecountthresh": "0",
    "param-texturedifferencethresh": "0",
    "param-usersm": "1",
    "param-zunits": "1000",
    "stream-depth-format": "Z16",
    "stream-fps": "15",
    "stream-height": "720",
    "stream-width": "1280"
} 
```
Any idea of how to get the same height for each pixel value?

[note: please ignore the straight lines in the graph. Those are pixels that didn't have any depth values. I have just fixed them to a certain height for good visualization of the graphs]

MartyG-RealSense, 2020-08-24:
Flat white walls with little or no texture can be difficult for the camera to analyze for depth detail.  If the IR Emitter component in the camera (a projector that casts a dot pattern onto the scene) is enabled then the camera can use the dots that are projected onto surfaces as a 'texture source' to read depth detail from.  

The link below has C++ code under the heading 'librealsense2' for enabling the IR Emitter using the instruction **RS2_OPTION_EMITTER_ENABLED**:

https://github.com/IntelRealSense/librealsense/wiki/API-How-To#controlling-the-laser